raspberrypi/linux #6288: INMP411 Microphone Module Not Recording Audio on Raspberry Pi 5

oyumusak, 2024-07-29:
### Describe the bug

Using the dtoverlay=googlevoicehat-soundcard with the INMP411 microphone module, audio recording works on Raspberry Pi 3 but not on Raspberry Pi 5. Despite identical hardware and software configurations, including following the Adafruit I2S MEMS Microphone Breakout guide, no audio is recorded on the Pi 5. The system and kernel are up to date. The issue seems specific to the Raspberry Pi 5, as the setup works perfectly on the Raspberry Pi 3.

### Steps to reproduce the behaviour

Add dtoverlay=googlevoicehat-soundcard,slave to /boot/config.txt.
Follow the Adafruit I2S MEMS Microphone Breakout guide for wiring the INMP411 microphone module to the Raspberry Pi 
Run the following command to attempt recording:

arecord -D plughw:0 -c1 -r 48000 -f S32_LE -t wav -V mono -v file.wav

Observe that no audio is recorded, while the same setup works on a Raspberry Pi 3.

### Device (s)

Raspberry Pi 5

### System

Raspberry Pi OS (Bookworm)
Kernel version: 6.6.42-v8-16k+
64-bit architecture
Latest firmware and kernel updates applied

pelwell, 2024-08-01:
I don't have that microphone, but looking at the signals on the bus it should be OK. Running the same arecord command I get a WAV file full of zeroes, which is what I would expect. And by periodically connecting SDI to 3V3 during the record I see a mixture of 0000s, ffffs, and a small number of intermediate values. This is with the current rpi-update kernel:
```
Linux raspberrypi 6.6.42-v8-16k+ #1787 SMP PREEMPT Thu Jul 25 17:45:51 BST 2024 aarch64 GNU/Linux
```

1. What behaviour are you seeing?
2. Are you sure that you are using the correct card number? I've seen it appear as card 2 in some configurations.

pelwell, 2024-09-20:
Closing due to a lack of response.

closed 2024-09-20 by pelwell