raspberrypi/linux #6289: Raspberry Pi 5 kernel upgrade breaks DWC2 USB-C OTG audio functionality over USB-C to USB-C cables

rithwikmakesthings, 2024-07-30:
### Describe the bug

A recent kernel upgrade (6.6.42-v8-16k+) appears to have broken the dwc2 gadget mode functionality of my Raspberry Pi 5 unit's USB C port, but only when using USB-C to USB-C cables (Still works as expected when I plug in an USB-A to USB-C OTG adaptor into a regular USB-A to USB-C cable though- same port on my Macbook and everything else constant too ofc).

I'm 100% sure this worked before the upgrade, and the config files remain untouched from before and after the upgrade (again, still works when I use a regular USB-C to A cable with an adaptor). The Pi 5 units themselves do not report any odd behaviour in dmesg, 

I've tested this with two different Pi5 units, many different cables, many different SD cards etc. Fresh installs of Raspberry Pi OS (from the Raspberry Pi Imager) seem to be affected too, so some update made between late 2023 and now appears to have caused this.

### Steps to reproduce the behaviour

run:
echo "dtoverlay=dwc2,dr_mode=peripheral" | sudo tee -a /boot/firmware/config.txt

followed by this script:
(https://github.com/brer-rabbit/zoxnoxious/blob/main/raspberry_pi/boot/create-usb-gadgets)

Plug in a USB C to USB C cable into a Macbook Pro (Unverified with other computers)

### Device (s)

Raspberry Pi 5

### System

Raspberry Pi reference 2023-10-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 962bf483c8f326405794827cce8c0313fd5880a8, stage2

2024/07/25 16:37:52 
Copyright (c) 2012 Broadcom
version 622dd3d8 (release) (embedded)

Linux norns 6.6.42-v8-16k+ #1 SMP PREEMPT Tue Jul 30 03:35:27 AEST 2024 aarch64 GNU/Linux

pelwell, 2024-07-30:
Can you describe the failure mode a bit?

popcornmix, 2024-07-30:
Can you identify the exact update which caused this. See:
https://github.com/raspberrypi/rpi-firmware/commits/master

If you click on each commit the end of the url contains a git hash. Run
```sudo rpi-update <hash>```
to revert back to that version. Report the first version with this error.