raspberrypi/linux #6291: Wrong parameter name in BQ32000 i2c-rtc overlay

rmolinete, 2024-07-31:
### Describe the bug

I've had some trouble trying to enable trickle charge in a bq32000 (the driver loaded but it didn't recognize the tricle-resistor-ohms parameter, so it won't charge the supercap) and I think it's an issue with the i2c-rtc.dts. 

I believe this change should be done:
[i2c-rtc-common.dtsi - line 343](https://github.com/raspberrypi/linux/blob/70636ad110715b5e1ec6b08e24f0ddaf5df7186d/arch/arm/boot/dts/overlays/i2c-rtc-common.dtsi#L357)

"trickle-resistor-ohms" instead of "abracon,tc-resistor"
```
trickle-resistor-ohms = <&ds1339>,"trickle-resistor-ohms:0",
					<&ds1340>,"trickle-resistor-ohms:0",
					<&abx80x>,"abracon,tc-resistor:0",
					<&rv3028>,"trickle-resistor-ohms:0",
					<&rv1805>,"abracon,tc-resistor:0",
					<&bq32000>,"trickle-resistor-ohms:0";
```

I've tested it recompiling and loading the .dtbo and seems to be fine now, but I'm still not sure enough to create a PR, so I leave the comment here in case you could check it out, thank you. I'm sorry if this is not the correct way to suggest changes but this is my first time I comment on an issue.

### Steps to reproduce the behaviour

Try to load i2c-rtc overlay for a bq32000 in config.txt.

```
dtoverlay=i2c0,pins_44_45
dtoverlay=i2c-rtc,bq32000,trickle-resistor-ohms=1120,i2c0
```

 Then, if you check via dmesg, the driver will load, but if you are using a supercap, the parameter "tricle-resistor-ohms" needed for the driver won't be passed to it, so the supercap won't charge.

### Device (s)

Raspberry Pi CM4

### System

Using jernel version  5.15.84-v7l+ 
Raspberry Pi OS lite based on Debian 11 bullseye

### Logs

The dmesg oytput is:

```

[    8.474935] bq32k 0-0068: invalid resistor value (0)
[    8.480327] bq32k 0-0068: registered as rtc0
[    8.482843] bq32k 0-0068: setting system clock to 2024-07-31T08:57:17 UTC (1722416237)

```

but it should be:

```
[    8.175632] bq32k 0-0068: Enabled trickle RTC battery charge.
[    8.178403] bq32k 0-0068: registered as rtc0
[    8.180881] bq32k 0-0068: setting system clock to 2024-07-31T08:46:06 UTC (1722415566)

```

6by9, 2024-07-31:
You are totally correct - the bq32k driver is looking for the generic trickle-resistor-ohms property, not the abracon vendor-specific one.

However the 5.15 tree is now unsupported by Raspberry Pi. We can fix up our current 6.6 tree and later supported branches (currently 6.10 and 6.11), but will be reluctant to backport.

pelwell, 2024-07-31:
You should just be able to copy the overlay from the 6.6 tree when it's ready, but you could also just use your fixed version - it's your choice.

pelwell, 2024-07-31:
Fixed in rpi-6.6.y by 05e3687.

rmolinete, 2024-07-31:
Thank you for your quick response. I'm glad that solved the problem.

closed 2024-07-31 by pelwell