raspberrypi/linux #6298: The edt_ft5x06 driver doesn't fire events for Waveshare 4.3" DSI touchscreen

kyngs, 2024-08-03:
### Describe the bug

Hi,
I'm having issues getting my Waveshare 4.3" touchscreen to work properly.
It appears that the correct driver (`edt_ft5x06`) does get loaded and even registers an input (in my case `/dev/input/event2`), however, when I attach `evtest` to the input and try to interact with the touchscreen, nothing happens. 
It almost behaves as if the touchscreen was broken, however, that is not the case as using the legacy `dtoverlay=rpi-ft5406` works flawlessly. Unfortunately, I cannot use `rpi-ft5406` as it doesn't work together with libcamera due to SPI conflicts

### Steps to reproduce the behaviour

- Get the latest octopi image with the new camera stack
- install a waveshare 4.3" LCD DSI display
- Add following overlays (as per the instructions in their documentation):
```
dtoverlay=vc4-kms-v3d
dtoverlay=vc4-kms-dsi-7inch
disable_touchscreen=0
```
- try to use `evtest` to verify whether the touchscreen works
- observethe 

### Device (s)

Raspberry Pi 4 Mod. B

### System

Raspberry Pi reference 2022-09-22
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 8a42abcd1dbd8c9c1fdfca4e0c3778255b2f9cc4, stage2

Mar 17 2023 10:51:33 
Copyright (c) 2012 Broadcom
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start_x)

Linux milada 6.1.21-v7l+ #1642 SMP Mon Apr  3 17:22:30 BST 2023 armv7l GNU/Linux (I can also reproduce this on the latest stable kernel)

### Logs

```
pi@milada:~ $ dmesg | grep "ft5x06"
[    8.385370] edt_ft5x06 10-0038: supply iovcc not found, using dummy regulator
[    8.710771] input: generic ft5x06 (00) as /devices/platform/soc/fe205000.i2c/i2c-22/i2c-10/10-0038/input/input2
```

```
pi@milada:~ $ evtest /dev/input/event2
Input driver version is 1.0.1
Input device ID: bus 0x18 vendor 0x0 product 0x0 version 0x0
Input device name: "generic ft5x06 (00)"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 330 (BTN_TOUCH)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value      0
      Min        0
      Max      799
    Event code 1 (ABS_Y)
      Value      0
      Min        0
      Max      479
    Event code 47 (ABS_MT_SLOT)
      Value      0
      Min        0
      Max        9
    Event code 53 (ABS_MT_POSITION_X)
      Value      0
      Min        0
      Max      799
    Event code 54 (ABS_MT_POSITION_Y)
      Value      0
      Min        0
      Max      479
    Event code 57 (ABS_MT_TRACKING_ID)
      Value      0
      Min        0
      Max    65535
Properties:
  Property type 1 (INPUT_PROP_DIRECT)
Testing ... (interrupt to exit)
```

### Additional context

Here's my whole /boot/config.txt in case you find it relevant:
```
arm_64bit=0
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=0

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
dtoverlay=vc4-kms-dsi-7inch
max_framebuffers=2

# Disable compensation for displays with overscan
disable_overscan=1

disable_touchscreen=0

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]

[pi4]
# Run as fast as firmware / board allows
arm_boost=1

[all]
# enable raspicam
start_x=1
gpu_mem=128
```

kyngs, 2024-08-04:
Now it seems like it started to complain about I2C issues:
```
[ 2656.824823] i2c-bcm2835 fe205000.i2c: i2c transfer timed out
[ 2656.824844] edt_ft5x06 10-0038: Unable to fetch data, error: -110
[ 2656.825214] edt_ft5x06 10-0038: Unable to fetch data, error: -5
[ 2659.886729] edt_ft5x06 10-0038: Unable to fetch data, error: -5
[ 2660.817678] edt_ft5x06 10-0038: Unable to fetch data, error: -5
[ 2661.944847] i2c-bcm2835 fe205000.i2c: i2c transfer timed out
[ 2661.944868] edt_ft5x06 10-0038: Unable to fetch data, error: -110
[ 2661.945238] edt_ft5x06 10-0038: Unable to fetch data, error: -5
```
However, (from my understanding) this shouldn't happen as it should work together with the imx477 driver, right?

6by9, 2024-08-04:
The vc4-kms-dsi-7inch overlay is written to work with the Raspberry Pi 7" panel. Whether it works with clones of that panel is down to the implementer of the clone, not with Raspberry Pi. The cloner is welcome to submit an overlay/driver for their panel if it does something different.

The edt_ft5x06 driver is written to work with EDT FT5x06 family of touchscreen controllers.

I suspect that Waveshare have done the same as DF Robot did in their panels and have a different touch controller on the panel with a microcontroller acting as a protocol translator. DF Robot emulated the bits of the FT5x06 protocol that the firmware happened to use. The kernel edt_ft5x06 driver uses some other commands within the FT5x06 protocol definition and they hadn't emulated them, hence it doesn't work.
There was discussion over this on one of the mainline Linux mailing lists over this. https://lore.kernel.org/all/CAJ+vNU3ZQy2SodE0sxiep-FaSUc4ugGK5BkZxowJF79tZ9iTpA@mail.gmail.com/T/ looks to be the link. It was followed up by https://www.spinics.net/lists/linux-input/msg75981.html being created as an RFC. I don't believe it was ever merged or taken further.

Support for third party products needs to come from the third party, not Raspberry Pi. Waveshare should be your first point of contact.

closed 2024-08-04 by pelwell