raspberrypi/linux #6303: Please add bcache as a module (CONFIG_BCACHE=m) (refs #455)

valord577, 2024-08-05:
### Describe the bug

drivers/md/bcache/Kconfig
config BCACHE

### Steps to reproduce the behaviour

as title

### Device (s)

Raspberry Pi 4 Mod. B

### System

* cat /etc/rpi-issue

Raspberry Pi reference 2024-07-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 48efb5fc5485fafdc9de8ad481eb5c09e1182656, stage2

* vcgencmd version

May 24 2024 15:30:04
Copyright (c) 2012 Broadcom
version 4942b7633c0ff1af1ee95a51a33b56a9dae47529 (clean) (release) (start)

* uname -a

Linux raspi-4b 6.6.31+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux

pelwell, 2024-08-06:
Interesting. Although that ancient issue resulted in `CONFIG_BCACHE=m` being added to rpi-3.10.y, it didn't survive the transitions to rpi-3.11.y and later. https://github.com/raspberrypi/linux/commit/a2f8ca289312a53da60bc0484a509bf790177982 adds it back.

pelwell, 2024-08-06:
Now that the autobuilds have completed, you can install a beta kernel including this patch by running:
```
$ sudo rpi-update rpi-6.6.y
```

valord577, 2024-08-07:
Thanks! It does work!

closed 2024-08-07 by valord577