raspberrypi/linux #6305: Breaking bluetooth hci changes in bookworm

DAbraham2, 2024-08-07:
### Describe the bug

hciattach will crash the kernel with null_ptr error if called like `sudo hciattach <device_tty> any <speed> <flow>`.

What I could trace this issue to is this commit: https://github.com/raspberrypi/linux/commit/090bebafc3eb0769a34e8fdf011ec868da3e17a1

### Steps to reproduce the behaviour

1. `sudo hciattach <device_tty> any 115200 noflow`

### Device (s)

Raspberry Pi 4 Mod. B

### System

Linux 6.6.31+rpt-rpi-v7l #1 SMP Raspbian 1:6.6.31-1+rpt1 (2024-05-29) armv7l GNU/Linux

This was consistent with 6.1.69 as well (whole bookworm range) and architecture independent (failing both 32 and 64 bit)

bluetoothctl: 5.66 

### Logs

```
Jul 31 12:01:21 host_name sudo[2460]:       pi : TTY=pts/0 ; PWD=/home/pi ; USER=root ; COMMAND=/usr/bin/hciattach /home/pi/pts_hci any 115200 noflow
Jul 31 12:01:21 host_name sudo[2460]: pam_unix(sudo:session): session opened for user root(uid=0) by pi(uid=1000)
Jul 31 12:01:21 host_name sudo[2460]: pam_unix(sudo:session): session closed for user root
Jul 31 12:01:21 host_name kernel: 8<--- cut here ---
Jul 31 12:01:21 host_name kernel: Unable to handle kernel NULL pointer dereference at virtual address 00000188 when read
Jul 31 12:01:21 host_name kernel: [00000188] *pgd=80000000004003, *pmd=00000000
Jul 31 12:01:21 host_name kernel: Internal error: Oops: 207 [#1] SMP ARM
Jul 31 12:01:21 host_name kernel: Modules linked in: cmac algif_hash aes_arm aes_generic aes_arm_bs crypto_simd cryptd algif_skcipher af_alg bnep xt_conntrack nft_chain_nat xt_MASQUERADE nf_nat nf_conntrack_netlink nf_conntrack n>
Jul 31 12:01:21 host_name kernel: CPU: 0 PID: 499 Comm: kworker/u13:3 Tainted: G         C         6.6.31+rpt-rpi-v7l #1  Raspbian 1:6.6.31-1+rpt1
Jul 31 12:01:21 host_name kernel: Hardware name: BCM2711
Jul 31 12:01:21 host_name kernel: Workqueue: hci1 hci_power_on [bluetooth]
Jul 31 12:01:21 host_name kernel: PC is at __dev_fwnode+0x8/0x1c
Jul 31 12:01:21 host_name kernel: LR is at hci_dev_open_sync+0xa8/0xae0 [bluetooth]
Jul 31 12:01:21 host_name kernel: pc : [<c0913660>]    lr : [<bf55684c>]    psr: 60000013
Jul 31 12:01:21 host_name kernel: sp : f0b21ea0  ip : 00000000  fp : 00000007
Jul 31 12:01:21 host_name kernel: r10: c2d66d6c  r9 : c2d67000  r8 : c2d6602c
Jul 31 12:01:21 host_name kernel: r7 : c2d668a8  r6 : c2fcbf00  r5 : c2d66000  r4 : c2d66000
Jul 31 12:01:21 host_name kernel: r3 : 00000002  r2 : 00000006  r1 : c2d6602c  r0 : 00000000
Jul 31 12:01:21 host_name kernel: Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
Jul 31 12:01:21 host_name kernel: Control: 30c5383d  Table: 02a1fa00  DAC: fffffffd
Jul 31 12:01:21 host_name kernel: Register r0 information: NULL pointer
Jul 31 12:01:21 host_name kernel: Register r1 information: slab kmalloc-8k start c2d66000 pointer offset 44 size 8192
Jul 31 12:01:21 host_name kernel: Register r2 information: non-paged memory
Jul 31 12:01:21 host_name kernel: Register r3 information: non-paged memory
Jul 31 12:01:21 host_name kernel: Register r4 information: slab kmalloc-8k start c2d66000 pointer offset 0 size 8192
Jul 31 12:01:21 host_name kernel: Register r5 information: slab kmalloc-8k start c2d66000 pointer offset 0 size 8192
Jul 31 12:01:21 host_name kernel: Register r6 information: slab maple_node start c2fcbf00 pointer offset 0 size 256
Jul 31 12:01:21 host_name kernel: Register r7 information: slab kmalloc-8k start c2d66000 pointer offset 2216 size 8192
Jul 31 12:01:21 host_name kernel: Register r8 information: slab kmalloc-8k start c2d66000 pointer offset 44 size 8192
Jul 31 12:01:21 host_name kernel: Register r9 information: slab kmalloc-8k start c2d66000 pointer offset 4096 size 8192
Jul 31 12:01:21 host_name kernel: Register r10 information: slab kmalloc-8k start c2d66000 pointer offset 3436 size 8192
Jul 31 12:01:21 host_name kernel: Register r11 information: non-paged memory
Jul 31 12:01:21 host_name kernel: Register r12 information: NULL pointer
Jul 31 12:01:21 host_name kernel: Process kworker/u13:3 (pid: 499, stack limit = 0x50ad6b6c)
Jul 31 12:01:21 host_name kernel: Stack: (0xf0b21ea0 to 0xf0b22000)
Jul 31 12:01:21 host_name kernel: 1ea0: c2ef4a00 c1404e60 c1562f18 f0b21f10 00000000 00000001 00000000 c3604a00
Jul 31 12:01:21 host_name kernel: 1ec0: c1408940 c1408940 efe924c0 c2ef4a00 c2ef4a00 7404bad2 c2ef50d0 c2d66688
Jul 31 12:01:21 host_name kernel: 1ee0: c2d66000 c2fcbf00 c2d668a8 c2d66008 c2d6602c c2fcbf05 00000007 bf509fd8
Jul 31 12:01:21 host_name kernel: 1f00: c129c4c0 c14055b4 c0cc5d04 7404bad2 00000000 c31f9100 c2d66688 c2fcbf00
Jul 31 12:01:21 host_name kernel: 1f20: c23ca200 c2ef4a00 000001a0 c023ca68 c23ca200 c1403d40 c23ca220 c31f9100
Jul 31 12:01:21 host_name kernel: 1f40: c23ca200 c1403d40 c23ca220 61c88647 c31f912c c2ef4a00 00000000 c023d3e4
Jul 31 12:01:21 host_name kernel: 1f60: c31f9100 c31d7ac0 f0aa1ec8 c35d6480 c2ef4a00 c023d14c c31f9100 c31d7ac0
Jul 31 12:01:21 host_name kernel: 1f80: f0aa1ec8 00000000 00000000 c0245f78 c35d6480 c0245e90 00000000 00000000
Jul 31 12:01:21 host_name kernel: 1fa0: 00000000 00000000 00000000 c020011c 00000000 00000000 00000000 00000000
Jul 31 12:01:21 host_name kernel: 1fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[... 23 lines cut ...]
Jul 31 12:01:21 host_name kernel: 1f80: f0aa1ec8 00000000 00000000 c0245f78 c35d6480 c0245e90 00000000 00000000
Jul 31 12:01:21 host_name kernel: 1fa0: 00000000 00000000 00000000 c020011c 00000000 00000000 00000000 00000000
Jul 31 12:01:21 host_name kernel: 1fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Jul 31 12:01:21 host_name kernel: 1fe0: 00000000 00000000 00000000 00000000 00000013 00000000 00000000 00000000
Jul 31 12:01:21 host_name kernel:  __dev_fwnode from hci_dev_open_sync+0xa8/0xae0 [bluetooth]
Jul 31 12:01:21 host_name kernel:  hci_dev_open_sync [bluetooth] from hci_power_on+0x5c/0x260 [bluetooth]
Jul 31 12:01:21 host_name kernel:  hci_power_on [bluetooth] from process_one_work+0x160/0x378
Jul 31 12:01:21 host_name kernel:  process_one_work from worker_thread+0x298/0x4f0
Jul 31 12:01:21 host_name kernel:  worker_thread from kthread+0xe8/0x104
Jul 31 12:01:21 host_name kernel:  kthread from ret_from_fork+0x14/0x38
Jul 31 12:01:21 host_name kernel: Exception stack(0xf0b21fb0 to 0xf0b21ff8)
Jul 31 12:01:21 host_name kernel: 1fa0:                                     00000000 00000000 00000000 00000000
Jul 31 12:01:21 host_name kernel: 1fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Jul 31 12:01:21 host_name kernel: 1fe0: 00000000 00000000 00000000 00000000 00000013 00000000
Jul 31 12:01:21 host_name kernel: Code: e1a00001 e8bd8010 e52de004 e28dd004 (e5903188)
Jul 31 12:01:21 host_name kernel: ---[ end trace 0000000000000000 ]---
Jul 31 12:01:21 host_name systemd[1]: Starting systemd-rfkill.service - Load/Save RF Kill Switch Status...
Jul 31 12:01:21 host_name systemd[2383]: Reached target bluetooth.target - Bluetooth.
Jul 31 12:01:21 host_name systemd[1]: Started systemd-rfkill.service - Load/Save RF Kill Switch Status.
Jul 31 12:01:26 host_name systemd[1]: systemd-rfkill.service: Deactivated successfully.
```

### Additional context

This behaviour was seen with Silicon Labs equipment. The above command runs fine on clean debian [images](https://raspi.debian.net) and I could attach my device to the bluetoothd.

pelwell, 2024-08-07:
Why are you using hciattach? The current Pi 4 .dtb file is configured for the kernel to bring up the HCI interface.

DAbraham2, 2024-08-07:
I am trying to attach an external ble device that I can control with bluez

pelwell, 2024-08-07:
It looks as though `hdev->dev.parent` must be NULL in your use case. I'll rustle up a patch to test.

pelwell, 2024-08-07:
#6306 is a potential fix for the crash. Wait about 45 minutes for the auto-builds to complete, then run `sudo rpi-update pulls/6306` to install a trial kernel with the patch. You should probably backup any important data, just in case.

DAbraham2, 2024-08-08:
This pr has solved the issue, many thanks!

Is this going in the main releases or what is the process?

pelwell, 2024-08-08:
I've hit the Merge button on the PR, so now that change will be in all future kernel releases.

DAbraham2, 2024-08-08:
Is there a timeframe for this or this is something that's going to auto-update with `apt-get upgrade`?

Sorry if this is a dumb question, I'm not a huge kernel enthusiastic

pelwell, 2024-08-08:
`sudo rpi-update` kernel releases are usually made at least once a week. Every few months one of those releases is pushed to the stable branch, from where it is added to the next apt kernel package (sort of - it's actually a separate build but starting from the same commit). Then roughly every 6 months we update the image downloads with the current version from apt.

DAbraham2, 2024-08-08:
okay, thanks!

closed 2024-08-08 by DAbraham2