raspberrypi/linux #6307: infinite loop when fstab points to non existent external drive

njamescouk, 2024-08-08:
### Describe the bug

having edited fstab to automatically mount an external hard drive, if the drive is removed or fails the machine boots to emergency mode, and asks for a root password, which apparently cannot even exist.

Pressing ctrl-d to continue gets you back to the same place.

### Steps to reproduce the behaviour

As above:
edit fstab to mount an external drive.
reboot, observe presemce of drive.
power off
remove external drive
power on
enter infinite loop

### Device (s)

Raspberry Pi 5

### System

Raspberry Pi reference 2023-10-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 962bf483c8f326405794827cce8c0313fd5880a8, stage5
$
2024/06/05 16:41:49 
Copyright (c) 2012 Broadcom
version 6fe0b091 (release) (embedded)

Linux raspberrypi 6.6.31+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux

pelwell, 2024-08-08:
What would you expect it to do? 

Have you tried putting `single` in `cmdline.txt`?

andrum993, 2024-08-08:
> having edited fstab to automatically mount an external hard drive, if the drive is removed or fails the machine boots to emergency mode...

Not a bug: you've misconfigured your system.

Add `nofail` to the list of options for that filesystem, and optionally `x-systemd.device-timeout=30` so it fails faster. See https://www.raspberrypi.com/documentation/computers/configuration.html#mount-a-storage-device and https://github.com/thagrol/Guides/blob/main/fstab.pdf.

andrum993, 2024-08-08:
And if you want to be able to log in as root, including in emergency mode, then boot normally and set a root password: `sudo passwd root`.

(To undo that, you would either need to use `passwd -l root` (that's a lowercase L), or use `vipw -s` to replace the root password in `/etc/shadow` with an asterisk, i.e. * (risky: type carefully).

pelwell, 2024-08-09:
Yes - it looks as though single user mode doesn't work without a root password:
```
Cannot open access to console, the root account is locked.
See sulogin(8) man page for more details.

Press Enter to continue.
```
Follow @andrum993's advice.

njamescouk, 2024-08-09:
ok thanks

I've just zapped the fstab entries

closed 2024-08-09 by popcornmix