raspberrypi/linux #6310: running cat proc/interrupts after removing my driver generates segafault

sagarramdev173, 2024-08-13:
### Describe the bug

I have created an interrupt driver with rpio gpio it loads perfectly fine but when it is loaded if I do cat on interrupts it works fine. The moment I unload my driver ADN do the same, then it generates segfault.

### Steps to reproduce the behaviour

I have created an interrupt driver with rpio gpio it loads perfectly fine but when it is loaded if I do cat on interrupts it works fine. The moment I unload my driver ADN do the same, then it generates segfault.

### Device (s)

Raspberry Pi 3 Mod. B+

### System

sagar@raspberrypi:~ $ cat /etc/rpi-issue
Raspberry Pi reference 2024-07-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 48efb5fc5485fafdc9de8ad481eb5c09e1182656, stage4
sagar@raspberrypi:~ $ uname -a
Linux raspberrypi 6.6.31+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux
sagar@raspberrypi:~ $ vcgencmd version
May 24 2024 15:31:28
Copyright (c) 2012 Broadcom
version 4942b7633c0ff1af1ee95a51a33b56a9dae47529 (clean) (release) (start)
sagar@raspberrypi:~ $

### Logs

[  264.878264] Unable to handle kernel paging request at virtual address ffffffe2944ba338
[  264.891135] Mem abort info:
[  264.895659]   ESR = 0x0000000096000007
[  264.901735]   EC = 0x25: DABT (current EL), IL = 32 bits
[  264.910349]   SET = 0, FnV = 0
[  264.915298]   EA = 0, S1PTW = 0
[  264.920388]   FSC = 0x07: level 3 translation fault
[  264.928296] Data abort info:
[  264.932959]   ISV = 0, ISS = 0x00000007, ISS2 = 0x00000000
[  264.941854]   CM = 0, WnR = 0, TnD = 0, TagAccess = 0
[  264.950045]   GCS = 0, Overlay = 0, DirtyBit = 0, Xs = 0
[  264.958661] swapper pgtable: 4k pages, 39-bit VAs, pgdp=0000000001358000
[  264.969532] [ffffffe2944ba338] pgd=100000003b3ff003, p4d=100000003b3ff003, pud=100000003b3ff003, pmd=1000000004814003, pte=0000000000000000
[  264.989882] Internal error: Oops: 0000000096000007 [#1] PREEMPT SMP
[  265.000051] Modules linked in: rfcomm snd_seq_dummy snd_hrtimer snd_seq snd_seq_device cmac algif_hash aes_arm64 aes_generic algif_skcipher af_alg bnep brcmfmac_wcc brcmfmac brcmutil
 cfg80211 binfmt_misc raspberrypi_hwmon bcm2835_codec(C) hci_uart bcm2835_v4l2(C) bcm2835_isp(C) btbcm bcm2835_mmal_vchiq(C) videobuf2_vmalloc v4l2_mem2mem bluetooth videobuf2_dma_conti
g videobuf2_memops videobuf2_v4l2 videodev videobuf2_common ecdh_generic ecc mc rfkill snd_bcm2835(C) libaes raspberrypi_gpiomem vc_sm_cma(C) uio_pdrv_genirq uio i2c_dev fuse dm_mod ip_
tables x_tables ipv6 vc4 snd_soc_hdmi_codec drm_display_helper cec drm_dma_helper drm_kms_helper drm drm_panel_orientation_quirks snd_soc_core snd_compress snd_pcm_dmaengine snd_pcm snd
_timer snd backlight i2c_bcm2835 [last unloaded: driver(O)]
[  265.113382] CPU: 2 PID: 1942 Comm: cat Tainted: G         C O       6.6.31+rpt-rpi-v8 #1  Debian 1:6.6.31-1+rpt1
[  265.129898] Hardware name: Raspberry Pi 3 Model B Plus Rev 1.3 (DT)
[  265.140063] pstate: 800000c5 (Nzcv daIF -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[  265.151362] pc : string+0x54/0x130
[  265.156892] lr : vsnprintf+0x238/0x6e0
[  265.162977] sp : ffffffc08151bb00
[  265.168346] x29: ffffffc08151bb00 x28: ffffffe2a7604f84 x27: 00000000ffffffd8
[  265.179941] x26: ffffffc08151bc40 x25: 0000000000000770 x24: ffffffc08151bc40
[  265.191535] x23: ffffff800293b890 x22: ffffff800293c000 x21: 0000000000000002
[  265.203127] x20: ffffff800293b892 x19: ffffffe2a7604f84 x18: 0000000000000003
[  265.214719] x17: 0000000000000000 x16: 0000000000000000 x15: ffffffc08151baf0
[  265.226311] x14: 0000000000000001 x13: ffffff800293b885 x12: 0000000000000000
[  265.237902] x11: 0000000000000000 x10: 0000000000000020 x9 : ffffffe2a671ab04
[  265.249495] x8 : 0000000000000020 x7 : 00000000ffffffff x6 : ffffff800293c000
[  265.261087] x5 : 0000000000000000 x4 : ffffffffffffffff x3 : ffff0a00ffffff04
[  265.272678] x2 : ffffffe2944ba338 x1 : 0000000000000000 x0 : ffffff800293b892
[  265.284271] Call trace:
[  265.288234]  string+0x54/0x130
[  265.293190]  vsnprintf+0x238/0x6e0
[  265.298711]  seq_printf+0xbc/0xe0
[  265.304093]  show_interrupts+0x2bc/0x400
[  265.310459]  seq_read_iter+0x34c/0x460
[  265.316542]  proc_reg_read_iter+0x8c/0xe8
[  265.323055]  vfs_read+0x214/0x2c0
[  265.328433]  ksys_read+0x78/0x118
[  265.333810]  __arm64_sys_read+0x24/0x38
[  265.340032]  invoke_syscall+0x50/0x128
[  265.346121]  el0_svc_common.constprop.0+0x48/0xf0
[  265.353759]  do_el0_svc+0x24/0x38
[  265.359140]  el0_svc+0x40/0xe8
[  265.364102]  el0t_64_sync_handler+0x100/0x130
[  265.371176]  el0t_64_sync+0x190/0x198
[  265.377127] Code: 91000400 eb07003f 54000560 aa0103e5 (38656844)
[  265.387017] ---[ end trace 0000000000000000 ]---
[  265.394503] note: cat[1942] exited with irqs disabled
[  265.402830] note: cat[1942] exited with preempt_count 1

pelwell, 2024-08-13:
Why do you think this is a bug in our kernel? It sounds like your driver is just not unregistering its interrupt when it is closed/released.

sagarramdev173, 2024-08-13:
Yeah i agree with that but it should not generate segfault , we should have handled it gracefully thats my thoughts

pelwell, 2024-08-14:
Drivers are inside the kernel and not subject to it's protection in the same way as user-space. If you wrote the driver, the ”we" is you.

closed 2024-08-14 by pelwell