raspberrypi/linux #6317: [rpi-02w] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)

qrp73, 2024-08-20:
### Describe the bug

When using Raspberry Pi Zero 2w with latest Bookworm aarch64, there is error in the log:
```
[   13.193367] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   13.433772] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
[   13.434558] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Jun 14 2023 07:27:45 version 7.45.96.s1 (gf031a129) FWID 01-70bd2af7 es7
```

I'm not sure what this error means?

Why first message shows ```using brcm/brcmfmac43430-sdio for chip BCM43430/1``` but actually loaded firmware is brcm/brcmfmac43436s-sdio? Is this also a bug?

wifi works in client and AP mode, but for some unknown reason don't works in p2p mode when configured through NetworkManager (I already [reported it here](https://github.com/raspberrypi/bookworm-feedback/issues/283)).  Is this error can be the root of cause for not working p2p?

I read that there is some kind of confusion with wifi firmware for rpi02w wifi module. 
It appears (correct me if I'm wrong) that rpi02w can use one of 2 module 43436 or 43436s. But actually they are hardware identical to 43430B0 and 43430A1 which are used for Raspberry Pi 3 and they have exactly the same id. I'm confused why there are two names 43436 and 43430B0 for two hardware identical modules with the same id?

And there are two firmwares for the same module - from Synaptics and from Infineon. So, if I understand correctly, in total there are 4 firmwares for 2 wifi modules. And it's not clear how the system selects which firmware to use - Synaptics or Infineon. Is there a way to choose which one will be used?

### Steps to reproduce the behaviour

Steps to reproduce:
1) Install latest Raspberry Pi Bookworm aarch64 on Raspberry Pi Zero 2w.
2) Run ```sudo apt update && sudo apt upgrade```
3) Run ```dmesg | grep "brcm"```

Expected result: no errors

Actual result: ```brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)```

### Device (s)

Raspberry Pi Zero 2 W

### System

```
$ cat /etc/rpi-issue
Raspberry Pi reference 2024-03-15
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, f19ee211ddafcae300827f953d143de92a5c6624, stage2

$ vcgencmd version
May 24 2024 15:31:28 
Copyright (c) 2012 Broadcom
version 4942b7633c0ff1af1ee95a51a33b56a9dae47529 (clean) (release) (start)

$ uname -a
Linux rpi02w 6.6.31+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux
```

### Logs

```
[   13.193367] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   13.433772] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
[   13.434558] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Jun 14 2023 07:27:45 version 7.45.96.s1 (gf031a129) FWID 01-70bd2af7 es7
```

frankozland, 2024-08-28:
I think it means you are missing the blob file for broadcom wifi chip
I've been fighting with broadcom wifi for a little while myself...

https://github.com/RaspAP/raspap-webgui/issues/1652

Make sure you do a backup before attempting.
Also you might want to consider downgrading to bullseye.

```
https://github.com/Infineon/ifx-linux-firmware/tree/master/firmware
# mkdir ~/firmware_backup
# cp -p /lib/firmware/brcm/brcmfmac43455* ~/firmware_backup
# wget --no-check-certificate -O /lib/firmware/brcm/brcmfmac43455-sdio.bin https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/cypress/cyfmac43455-sdio.bin
# wget --no-check-certificate -O /lib/firmware/brcm/brcmfmac43455-sdio.clm_blob https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/cypress/cyfmac43455-sdio.clm_blob
# wget --no-check-certificate -O /lib/firmware/brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.txt https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.txt

# sha256sum /lib/firmware/brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.txt ~/firmware_backup/brcmfmac43455-sdio.raspberrypi,4-model-b.txt 
edb6f4e4fb19e18940004124feb4ffe160d72fc607243a07a4480338a28b2748  /lib/firmware/brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.txt
ca709be81a78bdb6932936374f39943acbd7af07fae6151011127599a3ce9e3d  /root/firmware_backup/brcmfmac43455-sdio.raspberrypi,4-model-b.txt

# sha256sum /lib/firmware/brcm/brcmfmac43455-sdio.clm_blob ~/firmware_backup/brcmfmac43455-sdio.clm_blob
15f50a27020b263d1bea215c8f68d0550d912932d1d9ef19ffd59f18d82dd460  /lib/firmware/brcm/brcmfmac43455-sdio.clm_blob
843e14219c80aa3805a8b1255642c9d36e56471e9b7d5a5689b0e3ffbfa9b47f  /root/firmware_backup/brcmfmac43455-sdio.clm_blob

# sha256sum ~/firmware_backup/brcmfmac43455-sdio.bin /lib/firmware/brcm/brcmfmac43455-sdio.bin
d408faa9d0d5b1a2f9912dcea53ab0be48217288e398406d117f0edafe7c3edd  /root/firmware_backup/brcmfmac43455-sdio.bin
d408faa9d0d5b1a2f9912dcea53ab0be48217288e398406d117f0edafe7c3edd  /lib/firmware/brcm/brcmfmac43455-sdio.bin
```

learmj, 2024-08-29:
> I'm not sure what this error means?

The CLM blob is included in some of the firmware blobs (ie built-in). AFAIK there isn't a way to indicate to brcmfmac that a CLM blob isn't required for a particular platform. It complains if it isn't provided with one.
 
> Why first message shows `using brcm/brcmfmac43430-sdio for chip BCM43430/1` but actually loaded firmware is `brcm/brcmfmac43436s-sdio` (see version `7.45.96.s1`)?

This is firmware namespace vs filename. They don't have to be the same.
FYI the kernel in Bullseye does not have the ability to perform model specific firmware selection, so we introduced a mapping scheme using the DT. The newer kernel used in Bookworm does have the ability to do this which explains some subtle differences in the firmware loading for the same underlying hardware. For example:

Bullseye
```
brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43436s-sdio for chip BCM43430/1
brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Jun 14 2023 07:27:45 version 7.45.96.s1 (gf031a129) FWID 01-70bd2af7 es7
```

Bookworm:
```
brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Jun 14 2023 07:27:45 version 7.45.96.s1 (gf031a129) FWID 01-70bd2af7 es7
```

Take a look at the sym links in ```/lib/firmware/brcm/```.

> And why it logged as `Firmware: BCM43430/1`? Is this also a bug?

No