raspberrypi/linux #6346: Compiling 6.11 kernel

adminy, 2024-09-10:
### Describe the bug

gpio-pwm.c:138:19: error: initialization of 'void (*)(struct platform_device *)' from incompatible pointer type 'int (*)(struct platform_device *)' [-Werror=incompatible-pointer-types]

### Steps to reproduce the behaviour

build latest master kernel

### Device (s)

Raspberry Pi 5

### System

Linux cb6337052a15 6.10.0-linuxkit #1 SMP Wed Jul 17 10:51:09 UTC 2024 aarch64 GNU/Linux

popcornmix, 2024-09-12:
I've pushed a fixup commit. But I notice we don't actually have `CONFIG_GPIO_PWM` enabled in the default defconfig files
(which is why it is building okay for us).

Are you actually using this module, or were you just testing if all modules compile?

lategoodbye, 2024-10-11:
@popcornmix Please look at the [cover letter](https://lore.kernel.org/linux-gpio/20240604-pwm-gpio-v7-0-6b67cf60db92@linaro.org/) of this driver for the motivation. So it would be nice to integrate it into all relevant defconfigs.

6by9, 2024-10-11:
This issue was due to my downstream driver under CONFIG_GPIO_PWM which allowed a PWM device to be controlled via the GPIO subsystem. At the time I was wanting to drive the outputs of an I2C PWM device as simple GPIOs.
Being downstream it hadn't picked up the change for "remove" returning void instead of int.

The driver you've linked to is under CONFIG_PWM_GPIO which bitbashes PWM over the GPIO API.

Seeing as the downstream driver has been patched, I may as well close this.

closed 2024-10-11 by 6by9