raspberrypi/linux #6351: Strange OS freeze

qrp73, 2024-09-11:
### Describe the bug

I switched to the virtual terminal with Alt+Ctrl+F1 and played with text selection using the mouse and different combinations for copy/paste (I have installed gpm and clipman). Then I left it for 5-10 minutes, and when I came back, the system had frozen. It did not respond to keyboard input or SSH connections.

### Steps to reproduce the behaviour

I don't have steps for guaranteed reproduce of the issue.

### Device (s)

Raspberry Pi 4 Mod. B

### System

```
$ cat /etc/rpi-issue && vcgencmd version && uname -a
Raspberry Pi reference 2023-09-22
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 40f37458ae7cadea1aec913ae10b5e7008ebce0a, stage4
Aug 30 2024 19:17:39 
Copyright (c) 2012 Broadcom
version 2808975b80149bbfe86844655fe45c7de66fc078 (clean) (release) (start)
Linux raspi 6.6.47+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.47-1+rpt1 (2024-09-02) aarch64 GNU/Linux
```

### Logs

```
Sep 11 21:20:31 raspi rtkit-daemon[1013]: Supervising 8 threads of 5 processes of 1 users.
Sep 11 21:24:26 raspi rtkit-daemon[1013]: Supervising 8 threads of 5 processes of 1 users.
Sep 11 21:24:26 raspi rtkit-daemon[1013]: Supervising 8 threads of 5 processes of 1 users.
Sep 11 21:25:07 raspi rtkit-daemon[1013]: Supervising 8 threads of 5 processes of 1 users.
Sep 11 21:25:07 raspi rtkit-daemon[1013]: Supervising 8 threads of 5 processes of 1 users.
Sep 11 21:27:11 raspi rtkit-daemon[1013]: Supervising 8 threads of 5 processes of 1 users.
Sep 11 21:27:11 raspi rtkit-daemon[1013]: Supervising 8 threads of 5 processes of 1 users.
Sep 11 21:28:11 raspi rtkit-daemon[1013]: Supervising 8 threads of 5 processes of 1 users.
Sep 11 21:28:11 raspi rtkit-daemon[1013]: Supervising 8 threads of 5 processes of 1 users.
Sep 11 21:28:51 raspi rtkit-daemon[1013]: Supervising 8 threads of 5 processes of 1 users.
Sep 11 21:28:51 raspi rtkit-daemon[1013]: Supervising 8 threads of 5 processes of 1 users.
Sep 11 21:29:27 raspi rtkit-daemon[1013]: Supervising 8 threads of 5 processes of 1 users.
Sep 11 21:29:27 raspi rtkit-daemon[1013]: Supervising 8 threads of 5 processes of 1 users.
Sep 11 21:32:21 raspi rtkit-daemon[1013]: Supervising 8 threads of 5 processes of 1 users.
Sep 11 21:32:21 raspi rtkit-daemon[1013]: Supervising 8 threads of 5 processes of 1 users.
Sep 11 21:33:44 raspi rtkit-daemon[1013]: Supervising 8 threads of 5 processes of 1 users.
Sep 11 21:33:44 raspi rtkit-daemon[1013]: Supervising 8 threads of 5 processes of 1 users.
Sep 11 21:34:25 raspi rtkit-daemon[1013]: Supervising 8 threads of 5 processes of 1 users.
Sep 11 21:34:25 raspi rtkit-daemon[1013]: Supervising 8 threads of 5 processes of 1 users.
Sep 11 21:37:38 raspi rtkit-daemon[1013]: Supervising 8 threads of 5 processes of 1 users.
Sep 11 21:37:38 raspi rtkit-daemon[1013]: Supervising 8 threads of 5 processes of 1 users.
Sep 11 21:38:18 raspi rtkit-daemon[1013]: Supervising 8 threads of 5 processes of 1 users.
Sep 11 21:38:18 raspi rtkit-daemon[1013]: Supervising 8 threads of 5 processes of 1 users.
Sep 11 21:38:39 raspi /usr/sbin/gpm[757]: *** warning [daemon/old_main.c(254)]:
Sep 11 21:38:39 raspi /usr/sbin/gpm[757]: Data on strange file descriptor 4
Sep 11 21:39:02 raspi xdg-desktop-por[1327]: Backend call failed: Cannot invoke method; proxy is for the well-known name “org.freedesktop.ScreenSaver” without an owner, and proxy was constructed with the G_DBUS_PROXY_FLAGS_DO_NOT_AUTO_START flag
Sep 11 21:40:32 raspi /usr/sbin/gpm[757]: *** warning [daemon/old_main.c(254)]:
Sep 11 21:40:32 raspi /usr/sbin/gpm[757]: Data on strange file descriptor 4
Sep 11 21:42:08 raspi rtkit-daemon[1013]: Supervising 8 threads of 5 processes of 1 users.
Sep 11 21:42:08 raspi rtkit-daemon[1013]: Supervising 8 threads of 5 processes of 1 users.
Sep 11 21:42:48 raspi rtkit-daemon[1013]: Supervising 8 threads of 5 processes of 1 users.
Sep 11 21:42:48 raspi rtkit-daemon[1013]: Supervising 8 threads of 5 processes of 1 users.
Sep 11 21:43:25 raspi rtkit-daemon[1013]: Supervising 8 threads of 5 processes of 1 users.
Sep 11 21:43:25 raspi rtkit-daemon[1013]: Supervising 8 threads of 5 processes of 1 users.
Sep 11 21:44:54 raspi /usr/sbin/gpm[757]: *** warning [daemon/old_main.c(254)]:
Sep 11 21:44:54 raspi /usr/sbin/gpm[757]: Data on strange file descriptor 4
Sep 11 21:45:01 raspi rtkit-daemon[1013]: Supervising 8 threads of 5 processes of 1 users.
Sep 11 21:45:01 raspi rtkit-daemon[1013]: Supervising 8 threads of 5 processes of 1 users.
Sep 11 21:46:48 raspi rtkit-daemon[1013]: Supervising 8 threads of 5 processes of 1 users.
Sep 11 21:46:48 raspi rtkit-daemon[1013]: Supervising 8 threads of 5 processes of 1 users.

Sep 11 21:47:24 raspi /usr/sbin/gpm[757]: *** warning [daemon/old_main.c(254)]:
Sep 11 21:47:24 raspi /usr/sbin/gpm[757]: Data on strange file descriptor 4
Sep 11 21:47:28 raspi rtkit-daemon[1013]: Supervising 8 threads of 5 processes of 1 users.
Sep 11 21:47:28 raspi rtkit-daemon[1013]: Supervising 8 threads of 5 processes of 1 users.
```

pelwell, 2024-09-13:
What time did you return to the system and find it frozen? Are any of those log entries likely to be relevant?

qrp73, 2024-09-14:
> What time did you return to the system and find it frozen? Are any of those log entries likely to be relevant?

I return at about 21:45. There is no logs after 21:47:28 till the new boot. 

When I returned it shown Wayfire desktop, I'm not sure for 100% but probably I leave it in virtual terminal mode, because I surprised that it shows desktop. But the system was completely unresponsive, mouse didn't worked, keyboard didn't worked. I tried to ssh login to check the logs, but it appears that it don't listen even for ssh connections.

I don't remember such freeze for a long time, so I suspect it may be related to virtual terminal, gpm and clipboard with different keyboard shortcuts, like Alt+W, Ctrl+C/Ctrl+V, Cttrl+B and [, Ctrl+B and ], and other, which I played before that issue.

ViRb3, 2024-10-06:
Could this be related to #6349? You say "I don't remember such freeze for a long time", maybe the virtual terminal is just a coincidence and it's in reality the latest kernel update.