raspberrypi/linux #6352: The serial port number changed after the kernel was upgraded.

TopgunZh, 2024-09-12:
### Describe the bug

The serial port number changed after the kernel was upgraded from 5.0 to 6.0.

###  kernel 5.0
ttyAMA1  <--->  uart2
ttyAMA2  <--->  uart3
ttyAMA3  <--->  uart4
ttyAMA4  <--->  uart5
###  kernel 6.0
ttyAMA2  <--->  uart2
ttyAMA3  <--->  uart3
ttyAMA4  <--->  uart4
ttyAMA5  <--->  uart5

### Steps to reproduce the behaviour

###  kernel 5.0
[image]

###  kernel 6.0
[image]

### Device (s)

Raspberry Pi CM4

### System

###  kernel 5.0
[image]

###  kernel 6.0
[image]

pelwell, 2024-09-12:
That's correct, and it's intentional. Prior to our 6.0 kernels, UART numbering depended on which UARTs were enabled. Your table above for 5.0 is only correct if you've enable all of UART2, UART3, UART4 and UART5. With just UART5 enabled (for example), it would be ttyAMA2. In 6.0 and later, it is always ttyAMA5.

pelwell, 2024-09-12:
It won't change again.

mariusmotea, 2024-09-12:
I use a ZigBee controller connected on my PI5 serial interface flashed with z-stack firmware for Zigbee2MQTT. After i updated the packages and the serial adapter changed from ttyAMA1 to ttyAMA0 i fixed this in the app configuration but still Zigbee2MQTT fail to start. I can flash the zigbee controller firmware on the same serial interface with a python script, but zigbee2mqtt is unable to initiate the zigbee controller, like the communication is being corrupted. I used a separate USB to serial  adapter and it is working fine, so clearly something got broken on my PI5 after i update the kernel. I perform the serial loop test and it pass the test at same baud rate. I'm unable to downgrade to the old kernel since it is not present in the repository anymore, but downgrading some of the packages did not fixed the issue.

TopgunZh, 2024-09-13:
Then can I change the serial port number back by modifying the aliases of the device tree? As shown below.
[image]

The reason I want to change it back is:

- 1.My program already uses the ttyAMA1(uart2), ttyAMA2(uart3), ttyAMA3(uart4), ttyAMA4(uart5)serial port in kernel 6.0.

I don't want the program to need to determine the kernel version and switch the serial port after upgrading the kernel. It is troublesome for us to upgrade the kernel to users who have already shipped it. Usually we just upgrade our program.

- 2.Compatible with Raspberry Pi 5.
CM4(kernel 6.0)                                                        Pi5/CM5
 ttyAMA2（uart2, gpio0, gpio1）               ttyAMA1（uart1, gpio0, gpio1）
 ttyAMA3（uart3, gpio4, gpio5）               ttyAMA2（uart2, gpio4, gpio5）
 ttyAMA4（uart4, gpio8, gpio9）               ttyAMA3（uart3, gpio8, gpio9）
 ttyAMA5（uart5, gpio12, gpio13)              ttyAMA4（uart4, gpio12, gpio13)

The hardware io is the same, but the serial port number has changed. This requires me to determine the hardware platform and then switch the serial port.

pelwell, 2024-09-13:
> Then can I change the serial port number back by modifying the aliases of the device tree? 

Yes, you can, but it's a bit ugly. In an ideal world one would be able to put "serial1 = &uart2" in the overlay, but that doesn't work - the overlay compiler (dtc) doesn't know what "uart2" is, and the firmware can't fix it up later. It's also not possible to delete properties of the base DT from an overlay. What we can do is delete their content.

Here's an overlay which should work for you:
```
/dts-v1/;
/plugin/;

/ {
    compatible = "brcm,bcm2711";
    fragment@0 {
        target = <&aliases>;
        __overlay__ {
            serial1;
            serial2;
            serial3;
            serial4;
            serial5;
        };
    };
};
```
Write that into a file - I called it `serialrenumber-overlay.dts` - then compile and install it:
```
$ dtc -@ -Hepapr -I dts -O dtb -o serialrenumber.dtbo serialrenumber-overlay.dts
$ sudo cp serialrenumber.dtbo /boot/firmware/overlays/
```
You can now put `dtoverlay serialrenumber` in your config.txt.

There is one big caveat with this overlay - it doesn't play nicely with `dtoverlay=disable-bt`, so I hope you aren't using that.

TopgunZh, 2024-09-14:
Thank you very much.

closed 2024-09-14 by TopgunZh