raspberrypi/linux #6354: RPi5 SPI transfers: CS isn't kept low during words

heeplr, 2024-09-12:
### Describe the bug

On RPi5 with ```Linux mypi 6.6.31+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux``` and also custom ```Linux mypi 6.6.45-rt39-v8-16k+ #11 SMP Sat Aug 31 13:38:16 BST 2024 aarch64 GNU/Linux```, the configured CS (in my case CS0 of SPI.0 and SPI.1) is pulled high after $wordlength clock cycles.
It seems that earlier kernels/Pi models didn't show this behaviour.

### Steps to reproduce the behaviour

Reads from peripherals like the ICM42605 IMU or the SC16IS752 UART won't work if CS is pulled high before reading an answer.
(I'll use SC16IS752 on SPI1 as example, but also happens on SPI0):

* connect SC16IS752 to default SPI1 pins (SCLK=BCM21, MOSI=BCM20, MISO=BCM19, CS=BCM18, IRQ=BCM24)
* enable overlay in config.txt: ```dtoverlay=sc16is752-spi1```
* confirm ```/dev/ttySC0``` and ```/dev/ttySC1``` exist
* try TX: ``` echo 1 > /dev/ttySC0```
* this will hang and kernel will report error: ```sc16is7xx spi1.0: chip reports 255 free bytes in TX fifo, but it only has 64```
* set GPIO18 as output: 
  * before: 
```
$ pinctrl 18
18: a0    pd | hi // GPIO18 = SPI1_CE0
```
  * ```pinctrl set 18 op```
  * after: 
```
$ pinctrl 18
18: op dh pd | hi // GPIO18 = output
```
* now try TX again: ``` echo 1 > /dev/ttySC0``` and everything works as expected

connecting a logic analyzer shows the effect with an ICM42605:
D0 - SCLK
D1 = MISO
D2 = MOSI
D3 = CS

## works
[image]

## doesn't work
[image]

### Device (s)

Raspberry Pi 5

### System

```
System Information
------------------

Raspberry Pi 5 Model B Rev 1.0
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"

Raspberry Pi reference 2024-07-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 48efb5fc5485fafdc9de8ad481eb5c09e1182656, stage2

Linux ardupilot 6.6.45-rt39-v8-16k+ #11 SMP Sat Aug 31 13:38:16 BST 2024 aarch64 GNU/Linux
Revision	: c04170
Serial		: 68d12e48190205e8
Model		: Raspberry Pi 5 Model B Rev 1.0
Throttled flag  : throttled=0x0
Camera          : vc_gencmd_read_response returned -1 error=1 error_msg="Command not registered"

Videocore information
---------------------

[... 675 lines cut ...]
[   14.397459] input: pwr_button as /devices/platform/pwr_button/input/input0
[   14.661031] macb 1f00100000.ethernet eth0: Cadence GEM rev 0x00070109 at 0x1f00100000 irq 104 (m.m.m.m)
[   14.960311] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[   14.970938] Loaded X.509 cert 'benh@debian.org: 577e021cb980e0e820821ba7b54b4961b8b4fadf'
[   14.971530] Loaded X.509 cert 'romain.perier@gmail.com: 3abbc6ec146e09d1b6016ab9d6cf71dd233f0328'
[   14.972341] Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[   14.972917] Loaded X.509 cert 'wens: 61c038651aabdcf94bd0ac7ff06c7248db18c600'
[   15.015865] macb 1f00100000.ethernet end0: renamed from eth0
[   15.161935] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[   15.162251] usbcore: registered new interface driver brcmfmac
[   15.270824] spi1.0: ttySC0 at I/O 0x0 (irq = 163, base_baud = 921600) is a SC16IS752
[   15.285090] spi1.0: ttySC1 at I/O 0x1 (irq = 163, base_baud = 921600) is a SC16IS752
[   15.393366] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
[   15.394921] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Aug 29 2023 01:47:08 version 7.45.265 (28bca26 CY) FWID 01-b677b91b
[   16.071914] Adding 204784k swap on /var/swap.  Priority:-2 extents:12 across:761856k SS
[   17.569314] macb 1f00100000.ethernet end0: PHY [1f00100000.ethernet-ffffffff:01] driver [Broadcom BCM54213PE] (irq=POLL)
[   17.569328] macb 1f00100000.ethernet end0: configuring for phy/rgmii-id link mode
[   17.572373] pps pps0: new PPS source ptp0
[   17.575429] macb 1f00100000.ethernet: gem-ptp-timer ptp clock registered.
[   17.595388] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[  396.139520] systemd[641]: memfd_create() called without MFD_EXEC or MFD_NOEXEC_SEAL set
[  403.083104] sc16is7xx spi1.0: chip reports 255 free bytes in TX fifo, but it only has 64
[  403.083160] sc16is7xx spi1.0: chip reports 255 free bytes in TX fifo, but it only has 64
[  403.706563] sc16is7xx spi1.0: chip reports 255 free bytes in TX fifo, but it only has 64
```

pelwell, 2024-09-12:
There's something strange about your overlay, or perhaps some other software is changing the pinmux. When I load `sc16is752-spi1` I get:
```
pi@phil-pi5:~ $ pinctrl 18-21
18: op dl pd | lo // GPIO18 = output
19: a0    pd | lo // GPIO19 = SPI1_MISO
20: a0    pd | lo // GPIO20 = SPI1_MOSI
21: a0    pd | lo // GPIO21 = SPI1_SCLK
```
This is correct - the CS line is driven as a GPIO by the kernel SPI framework. I get the same result whether the overlay is loaded by the firmware or at runtime using the `dtoverlay` command.

Try this:
1. Comment out `dtoverlay=sc16is752-spi1` in config.txt.
2. Reboot the Pi.
3. What does `pinctrl 18-21` show?
4. Run `sudo dtoverlay sc16is752-spi1`.
5. What does `pinctrl 18-21` show now?

heeplr, 2024-09-12:
The results are strange indeed:

```
$ ls /dev/ttySC*
ls: cannot access '/dev/ttySC*': No such file or directory
$ pinctrl 18-21
18: a0    pd | hi // GPIO18 = SPI1_CE0
19: no    pd | -- // GPIO19 = none
20: no    pd | -- // GPIO20 = none
21: no    pd | -- // GPIO21 = none
$ sudo dtoverlay sc16is752-spi1
$ ls /dev/ttySC*
/dev/ttySC0  /dev/ttySC1
$ pinctrl 18-21
18: op dh pd | hi // GPIO18 = output
19: a0    pd | lo // GPIO19 = SPI1_MISO
20: a0    pd | lo // GPIO20 = SPI1_MOSI
21: a0    pd | lo // GPIO21 = SPI1_SCLK
```

The sc16is752-spi1-overlay.dts from 29f7f01091 i'm using is

```
dts-v1/;
/plugin/;

/ {
        compatible = "brcm,bcm2835";

        fragment@0 {
                target = <&gpio>;
                __overlay__ {
                        spi1_pins: spi1_pins {
                                brcm,pins = <19 20 21>;
                                brcm,function = <3>; /* alt4 */
                        };

                        spi1_cs_pins: spi1_cs_pins {
                                brcm,pins = <18>;
                                brcm,function = <1>; /* output */
                        };   

                        int_pins: int_pins@18 {
                                        brcm,pins = <24>;
                                        brcm,function = <0>; /* in */
                                        brcm,pull = <0>; /* none */
                        };
                };
        };

        fragment@1 {
                target = <&spi1>;
                __overlay__ {
                        #address-cells = <1>;
                        #size-cells = <0>;
                        pinctrl-names = "default";
                        pinctrl-0 = <&spi1_pins &spi1_cs_pins>;
                        cs-gpios = <&gpio 18 1>;
                        status = "okay";

                        sc16is752: sc16is752@0 {
                                compatible = "nxp,sc16is752";
                                reg = <0>; /* CE0 */
                                clocks = <&sc16is752_clk>;
                                interrupt-parent = <&gpio>;
                                interrupts = <24 2>; /* IRQ_TYPE_EDGE_FALLING */
                                pinctrl-0 = <&int_pins>;
                                pinctrl-names = "default";
                                gpio-controller;
                                #gpio-cells = <2>;
                                spi-max-frequency = <4000000>;
                        };
                };
        };

        fragment@2 {
                target = <&aux>;
                __overlay__ {
                        status = "okay";
                };
        };

        fragment@3 {
                target-path = "/";
                __overlay__ {
                        sc16is752_clk: sc16is752_spi1_0_clk {
                                compatible = "fixed-clock";
                                #clock-cells = <0>;
                                clock-frequency = <14745600>;
                        };
                };
        };

        __overrides__ {
                int_pin = <&sc16is752>,"interrupts:0", <&int_pins>,"brcm,pins:0",
                          <&int_pins>,"reg:0";
                xtal = <&sc16is752_clk>,"clock-frequency:0";
        };
};
```

pelwell, 2024-09-12:
You must have a script etc. that is changing the pinmux for GPIO 18, perhaps using pinctrl. If the pin had been changed from Device Tree then the overlay would not have applied successfully at runtime.

heeplr, 2024-09-12:
Indeed I can't reproduce this with a fresh image. That's strange since this is basically raspbian-lite + custom realtime kernel.
I'll close this. If it really affects other's too, it could be reopened.

@pelwell thanks for clarifying

closed 2024-09-12 by heeplr