raspberrypi/linux #6365: Pi 5 UARTs broken in 6.6.44 build of rpi-6.6.y.

pelwell, 2024-09-17:
### Describe the bug

As detailed by @lgeitner in the thread starting at https://github.com/raspberrypi/linux/pull/6144#issuecomment-2349489287, the MeteoPi application which requires UART access seems to be broken in the 6.6.44 build of rpi-6.6.y. It has also been shown to have worked in the 6.6.42 build. The corresponding source commits for those two builds are:

"kernel: Bump to 6.6.42" (1969fd7): bfbd468e81defa70d4356a1a3e467cc5d67ca89b
"kernel: Bump to 6.6.44" (44a287f): 2806f39528464ddcc56f77dd8ea4d81f71b7e619

### Steps to reproduce the behaviour

Run MeteoPi. Other test cases have yet to be determined.

### Device (s)

Raspberry Pi 5

### System

`Linux 6.6.44-v8-16k+ #1789 SMP PREEMPT Mon Aug 5 15:24:18 BST 2024 aarch64 GNU/Linux`

### Logs

N/A

pelwell, 2024-09-17:
The current list of suspect commits is:
```
2806f39528464 Revert "mfd: Partial revert of Add rp1 driver"
d0ea54e1c941a mfd: Partial revert of Add rp1 driver
216df57950849 DTS: bcm2712: enable SD slot CQE by default on Pi 5
3c319a466a1c7 dtoverlays: Document display_[width|height] on hd44780-lcd overlay
e94e761305fa2 dtoverlays: Add overlay for HD44780 via I2C PCF8574 backpack
a4bf61fad9fe1 fixup! pinctrl: bcm2712 pinctrl/pinconf driver
16d0ee22d2c0b hwmon: (adt7410) Add DT compatible strings
05e3687c6c973 overlays: i2c-rtc: Correct bq32000 property name
e9294823cf020 spi: dw: Clamp the minimum clock speed
199e611183de0 spi: dw: Fix non-DMA transmit-only transfers
70636ad110715 ARM: dts: bcm2712: Fix invalid polling-delay-passive setting
485d11cfa7df2 drm/vc4: Disable the 2pixel/clock odd timings workaround for interlaced
062434ab3be76 dts: rp1: restrict i2s burst lengths to 4
b6b4260fa546d sound/soc: dwc-i2s: choose FIFO thresholds based on DMA burst constraints
5112fd8cce4f1 DT: bindings: add a dma-maxburst property to snps,designware-i2s
f3cb675102a2a tty/serial: pl011: restrict RX burst FIFO threshold
1c4c90982462e Revert "spi: dw-dma: Get the last DMA scoop out of the FIFO"
ce56098eb4dc2 drivers: dw-axi-dmac: make more sensible choices about memory accesses
4c1a665b465fa dts: rp1: hobble DMA AXI burst lengths
3af7822df36e3 spi: dw: don't immediately kill DMA transfers if an error occurs
cd9084ceb606a spi: dw: Save bandwidth with the TMOD_RO feature
6014649de765a spi: dw: Save bandwidth with the TMOD_TO feature
31b9871b8895d staging: bcm2835-codec: Add support for H264 level 5.0 and 5.1
```
I'll try and create a PR from the 6.6.44 build head (2806f39528464) but with those commits reverted.

pelwell, 2024-09-17:
#6366 is the PR that will create the trial builds. After each update to it, wait about 45 minutes for the builds to complete then run `sudo rpi-update pulls/6366`.

pelwell, 2024-09-17:
Do you want to reconsider any of the above patches, @P33M?

P33M, 2024-09-17:
`f3cb675102a2a tty/serial: pl011: restrict RX burst FIFO threshold` may be causing side-effects. This isn't critical to UART operation with the restricted burst sizes - the RX timeout interrupt will gather any remaining bytes.

pelwell, 2024-09-17:
With f3cb675102a2a I see data loss at the end of transmission:
```
00009300: 72 69 61 6c 3a 20 6e 6f : 20 70 6c 61 74 66 6f 72
00009308: 20 44 4d 41 20 70 6c 61 : 6d 20 64 61 74 61 0a
00009310: 74 66 6f 72 6d 20 64 61 :
00009318: 74 61 0a                :
```
(where 931a is the last byte, and everything before 9300 matches)
With it reverted, I've not seen any data loss yet.

I've update #6366 to be top-of-tree rpi-6.6.y but with f3cb675102a2a reverted. As before, wait about 45 minutes then run `sudo rpi-update pulls/6366`.

lgeitner, 2024-09-17:
ran `sudo rpi-update pulls/6366`

received the error "Invalid artifact specified. Response: 404." 
please see below

pi@cumulusmxkitchen:~ $ sudo rpi-update pulls/6366
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
FW_REV:
BOOTLOADER_REV:d05f05c94fd7a1916942e56d7f486c142ae5661e
WANT_32BIT:0 WANT_64BIT:1 WANT_PI4:1 WANT_PI5:1
 *** Downloading specific artifact revision (this will take a few minutes)
curl  -L https://builds.raspberrypi.com/github/linux/8aa0ad843d49a58d738177fd892ec547015dd231/bcmrpi | zcat | tar xf - -C //root/.rpi-firmware --strip-components=2
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:04 --:--:--     0

gzip: stdin: unexpected end of file
tar: This does not look like a tar archive
tar: Exiting with failure status due to previous errors
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:04 --:--:--     0
Invalid artifact specified. Response: 404.
pi@cumulusmxkitchen:~ $

pelwell, 2024-09-17:
I thought this might happen. I updated the PR to include a merge-ready commit message, and you came online and tried to update in the interval while it was being rebuilt. It should be good now.

lgeitner, 2024-09-17:
ran `sudo rpi-update pulls/6366` then `sudo reboot`

the meteopi is not working!

the command `uname -a` displays the following:

pi@cumulusmxkitchen:~ $ uname -a
Linux cumulusmxkitchen 6.6.51-v8-16k+ #1 SMP PREEMPT Tue Sep 17 16:24:28 UTC 2024 aarch64 GNU/Linux

pelwell, 2024-09-17:
Did you reboot after the install?

There's another build with more reversions available via `sudo rpi-update pulls/6369`.

lgeitner, 2024-09-17:
I ran `sudo rpi-update pulls/6369` then `sudo reboot`

The meteopi is now working!

I would like to allow the meteopi to run overnight and check for any errors from the meteopi.  
I will let you know if there are any errors or problems sometime tomorrow.

Thank You!

output from the command  `uname- a`

pi@cumulusmxkitchen:~ $ uname -a
Linux cumulusmxkitchen 6.6.51-v8-16k+ #1 SMP PREEMPT Tue Sep 17 19:44:50 UTC 2024 aarch64 GNU/Linux
pi@cumulusmxkitchen:~ $

lgeitner, 2024-09-17:
I just noticed when I ran the command `dmesg` there is the following error in red

[   11.585942] pl011-axi 1f00030000.serial: unable to pause DMA transfer

pelwell, 2024-09-18:
`sudo rpi-update pulls/6371` will get you the release candidate patch. It simply disables DMA for UART0 until it can be shown to be reliable.

lgeitner, 2024-09-18:
I let the meteopi run overnight, there were no errors!  Its working as expected!

Thank you again for your help!

I will try `sudo rpi-update pulls/6371` sometime in the next couple hours and will let you know if I have any issues with it.

Thanks,
Lance

lgeitner, 2024-09-18:
I ran the command `sudo rpi-update pulls/6371` then `sudo reboot`.

The meteopi is working fine and the error "pl011-axi 1f00030000.serial: unable to pause DMA transfer" no longer appears in `dmesg`.

the output from `uname -a` is as follows:

pi@cumulusmxkitchen:~ $ uname -a
Linux cumulusmxkitchen 6.6.51-v8-16k+ #1 SMP PREEMPT Wed Sep 18 16:24:48 UTC 2024 aarch64 GNU/Linux
pi@cumulusmxkitchen:~ $ 

Thank you again for your help!

pelwell, 2024-09-18:
Thank you for reporting it - shipping that kernel in an image would have been embarrassing.

idulkoan, 2024-09-19:
Following as I am having issues after rebooting my pi (presumably kernel upgrade) on my rpi5, Klipper installation over UART (Pin 8/10 to btt skr e3 mini v2)

`Linux Ender5Pro 6.6.51-v8-16k+ #1 SMP PREEMPT Wed Sep 18 16:24:48 UTC 2024 aarch64 GNU/Linux`

```
###### Identifying MCU possibly connected via UART ...

 ● MCU #1: /dev/ttyS0
 ● MCU #2: /dev/ttyAMA0
```
```
mcu 'mcu': Unable to connect
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
```

Interestingly, after using 
` sudo rpi-update pulls/6371`
and restarting, I notice that with this version, If I let it sit for a long time, I eventually do get text on the printer screen randomly and have had a very brief window where I was able to start the homing process, but then it just dies completely. Its very random, and has only happened like 2-3 times in my testing.

I have created this python script:
```
pi@Ender5Pro:~/Downloads $ cat uart_test.py
import serial
import time

# Configure the serial port
port = '/dev/ttyAMA0'
baudrate = 250000
timeout = 1  # seconds

# Initialize the serial connection
ser = serial.Serial(port, baudrate, timeout=timeout)

# Function to write and read data
def test_uart():
    test_message = "Hello, UART Loopback!"
    ser.write(test_message.encode())  # Send data
    time.sleep(1)  # Wait for a short moment
    ser.flushInput()  # Clear the input buffer
    response = ser.read(len(test_message)).decode()  # Read the response

    if response == test_message:
        print("UART loopback test passed!")
    else:
        print("UART loopback test failed.")
        print(f"Sent: {test_message}")
        print(f"Received: {response}")

# Run the test
if __name__ == "__main__":
    test_uart()

# Close the serial port
ser.close()
```

To help test UART loopback after shorting pins 8 and 10 (TX RX) , But never had a successful result from the test everytime I have tried.

ausfas, 2024-11-02:
I have the same issue but I also have Zwave card on GPIO...installing sudo rpi-update pulls/6371
Stop my GPIO ?

I'm also and few others having issues when they reboot homeassistant, the nvme doesn't boot 2out of 3 times...and stuck in startup screen... could they both be linked?

lurch, 2024-11-02:
> I have the same issue but I also have Zwave card on GPIO...installing sudo rpi-update pulls/6371
Stop my GPIO ?

This issue has already been closed, and AFAIK the fix is already included in the kernels being shipped in the October 22nd 2024 versions of [Raspberry Pi OS](https://www.raspberrypi.com/software/operating-systems/). So if you're still having problems with the latest OS, it's probably best to open a new issue.

> I'm and few others having issue when they reboot homeassistant, the nvme doesn't boot 2 out of 3 times...and stuck in startup screen... could this be linked to this issue as I have changed Geekpi nvme POE HAT to waveshare but same issue

Depending on exactly _when_ it "gets stuck", problems with not being able to boot from an NVME SSD could either be related to https://github.com/raspberrypi/rpi-eeprom or to https://github.com/raspberrypi/linux

Also note that some SSDs simply aren't compatible with the Raspberry Pi (see e.g. the lists at https://shop.pimoroni.com/products/nvme-base ). The official Raspberry Pi [SSDs](https://www.raspberrypi.com/products/ssd/) will definitely work with the official [M.2 HAT+](https://www.raspberrypi.com/products/m2-hat-plus/), so will probably also work with most (or all) 3rd-party M.2 HATs too.

steneor, 2025-03-17:
Hello, For a long time (with pi2, pi3, pi4) I've been using the ttyAMA0 serial port to read information sent by an electric meter (at 1200 bps).
I recently bought a RPI5, but it no longer works with ttyAMA0.
However, I tried it with ttyAMA3 and it works.
I think the problem comes from the fact that ttyAMA0 is managed by DMA.

My question is how do I disable DMA on ttyAMA0?

Thanks in advance for your help and advice.

Some information:
root@PI5:~ # dmesg
4.011419] macb 1f00100000.ethernet: gem-ptp-timer ptp clock registered.
[ 8.099510] macb 1f00100000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
**[ 8.672770] pl011-axi 1f00030000.serial: DMA channel TX dma2chan0
[ 8.672782] pl011-axi 1f00030000.serial: DMA channel RX dma2chan1**

root@PI5:~ # uname –a
**Linux PI5 6.12.18-v8-16k+ #1862** SMP PREEMPT Wed Mar 12 12:33:09 GMT 2025 aarch64 GNU/Linux

root@PI5:~ # cat /sys/kernel/debug/dmaengine/summary
dma0 (1000010000.dma): number of channels: 5

dma1 (1000010600.dma): number of channels: 5
dma1chan0 | 107d004000.spi:tx
dma1chan1 | 107d004000.spi:rx
dma1chan2 | 107c701400.hdmi:audio-rx
dma1chan3 | 107c706400.hdmi:audio-rx

dma2 (1f00188000.dma): number of channels: 8
**dma2chan0 | 1f00030000.serial:tx
dma2chan1 | 1f00030000.serial:rx**

steneor, 2025-03-17:
how to apply this issue? I do this:    but -> "Invalid hash given"

`root@PI5:~ # rpi-update pulls/6365
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
FW_REV:
BOOTLOADER_REV:78d08e9763079c6608506e25eefeee5ceb0ceabc
 *** Invalid hash given
`

lurch, 2025-03-17:
@steneor I believe that the fix from this PR has already been included in the kernels that are installed by `apt`.

6by9, 2025-03-17:
CI builds are only retained for 90 days.

6365 is the issue number, not the PR number.

#6371 was the merged fix for this issue, and but was reverted via d47754e445e59f1face7b2b248f93d5d0b9e19a2 as fc4d644e0dcb7389666723f501f3c473e8ed6579 is the more correct/complete fix.

If you want to disable DMA again, then you'll need to build your own DT files with the same nodes commented out as in #6371.

pelwell, 2025-03-18:
https://github.com/raspberrypi/linux/pull/6724

steneor, 2025-03-18:
Thank you very much
rpi-update pulls/6724
Now /dev/ttyAMA0 works fine.

steneor, 2025-03-18:
The configuration is as follows:
In France, electricity meters continuously transmit a frame of information at 1200 bps.
frame: STX(0x02) dataset dataset dataset ......... dataset dataset ETX(0x03)
dataset: 0x0a ..(about 15 bytes) ..0x0d
Reading this frame with an RPI 1 2 3 4 is standard and works very well.
I think I was one of the first to try this with RPI5, and it no longer worked with ttyAMA0, but worked very well with ttyAMA[1-4].
Here's a website that describes the project well: https://www.magdiblog.fr/gpio/teleinfo-edf-suivi-conso-de-votre-compteur-electrique/

Thanks again for your help.

pelwell, 2025-03-19:
I've tried sending data to a Pi 5 via ttyAMA0/UART0 with those serial settings, and it works for me. What does the failure look like? Is it data corruption, or does it stall? Ae there kernel error messages (in the `dmesg` output)?

By the way, I'm considering disabling DMA in general for UART0, or at least providing a way to enable or disable it using a dtparam.

pelwell, 2025-03-20:
> I'm considering disabling DMA in general for UART0, or at least providing a way to enable or disable it using a dtparam.

I went with the latter - `dtparam=uart0_nodma` will enable UART0 without DMA.

steneor, 2025-03-22:
I wanted to try again this morning to provide you with some answers.
I created a new USB drive with the Raspberry Pi imager (PI5 Raspberry Pi OS Lite 64-bit).
Here's the result, which surprised me (6.6.51 without DMA):
```
root@pi5-usb:~ # uname –a
Linux pi5-usb 6.6.51+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux
```
This version doesn't support ttyAMA0 with DMA and therefore **works fine**.

Then I did `apt update apt upgrade` (6.6.74 with DMA)
```
root@pi5-usb:~ # uname –a
Linux pi5-usb 6.6.74+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.74-1+rpt1 (2025-01-27) aarch64 GNU/Linux
```
```
root@pi5-usb:~ # dmesg |grep serial
[0.012258] 107d001000.serial: ttyAMA10 at MMIO 0x107d001000 (irq = 15, base_baud = 0) is a PL011 rev3
[0.244251] 107d50c000.serial: ttyS0 at MMIO 0x107d50c000 (irq = 33, base_baud = 6000000) is a Broadcom BCM7271 UART
[ 0.244549] serial serial0: tty port ttyS0 registered
[0.868453] pl011-axi 1f00030000.serial: cts_event_workaround enabled
[ 0.869393] 1f00030000.serial: ttyAMA0 at MMIO 0x1f00030000 (irq = 126, base_baud = 0) is a PL011 AXI
[4.935271] hci_uart_bcm serial0-0: supply vbat not found, using dummy regulator
[4.935317] hci_uart_bcm serial0-0: supply vddio not found, using dummy regulator
[184.376116] pl011-axi 1f00030000.serial: DMA channel TX dma2chan0
[184.376130] pl011-axi 1f00030000.serial: DMA channel RX dma2chan1
```
With this new version 6.6.74 the ttyAMA0 no longer works.

To test ttyAMA0, I connect an electricity meter to ttyAMA0, then
```
root@pi5-usb:~ # /usr/bin/stty -F /dev/ttyAMA0 1200 sane evenp parenb cs7 –crtscts
root@pi5-usb:~ # cat /dev/ttyAMA0
```
The Pi5 remains blocked and there is no display.

If you want to keep ttyAMA0's DMA management, the solution you propose, `dtparam=uart0`_nodma, seems good to me.

lurch, 2025-03-24:
`-crtscts`
Ahh, do you also have the CTS and RTS lines wired up between your Pi and your electricity meter? Perhaps when Phil was doing his earlier test he was only using RX and TX? (I'm just guessing here, but maybe this is what is making the DMA-UART test work for Phil, but not for you?)

pelwell, 2025-03-24:
`-crtscts` means _no_ flow control.

lurch, 2025-03-24:
> `-crtscts` means _no_ flow control.

Doh, I'd skim-read the `stty` man page but glossed over the bit where it says `Optional - before SETTING indicates negation.` 🤦

pelwell, 2025-03-24:
The difference will be due to the timing of output from the electricity meter.

lgeitner, 2025-04-13:
I was on Kernel Version 6.6.74.  The MeteoPI was working perfectly. 

I performed the following commands:

sudo apt update
sudo apt full upgrade -y
sudo reboot

This upgraded me to Kernel Version 6.12.20.

I started noticing errors again with the MeteoPI.

In my /boot/firmware/config.txt, I added the following line to the file

dtparam=uart0_nodma

I saved my config.txt file then issued

sudo reboot

Unfortunately, I am still seeing the MeteoPI errors!

Is there something else I can try?

Thanks,
Lance

lurch, 2025-04-14:
@lgeitner There was a slight hiccup in he upgrade to the 6.12 kernel, which we've now fixed (see https://github.com/raspberrypi/bookworm-feedback/issues/401 ). Could you try another update please, and see if that fixes things?

pelwell, 2025-04-14:
The latest `stable` branch release, which is a build of 6.12.20, includes the patch that disables DMA for UART0. Are you sure you are running that release?
There are a few things you can check:
1. `uname -a`
2. If you have `dtparam=uart0_nodma` in confi.txt,  `sudo vclog -m` should show an error about an unknown parameter.
3. `cat /sys/kernel/debug/dmaengine/summary` should not mention any UARTs.

lgeitner, 2025-04-14:
Below are the results from the commands above.

pi@cumulusmxkitchen:~ $ uname -a
Linux cumulusmxkitchen 6.12.20+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.12.20-1+rpt1~bpo12+1 (2025-03-19) aarch64 GNU/Linux

pi@cumulusmxkitchen:~ $ sudo vclog -m
006841.250: Initial voltage 800000 temp 50469
007041.824: avs_2712: AVS pred 8947 894700 temp 52118
007045.431: vpred 894 mV +0
007673.701: FB framebuffer_swap 1
007693.099: Select resolution HDMI0/2 hotplug 1 max_mode 2
007710.136: HDMI0 edid block 0 offset 0
007712.516: 00ffffffffffff0022f08a3101010101
007718.189: 051a010380331d782ae595a656529d27
007723.861: 105054a10800d1c081c0a9c0b3009500
007729.534: 818081000101023a801871382d40582c
007735.207: 4500fd1e1100001e000000fd00323c1e
007740.880: 5011000a202020202020000000fc0048
007746.553: 5020323378770a2020202020000000ff
007752.225: 0036434d36303530364c510a2020019f
007770.884: HDMI0 edid block 1 offset 128
007773.442: 020319b149901f041303021211016703
007779.115: 0c0010000022e2002b023a801871382d
007784.788: 40582c4500fd1e1100001e023a80d072
007790.460: 382d40582c4500fd1e1100001e000000
007796.133: 00000000000000000000000000000000
007801.806: 00000000000000000000000000000000
007807.479: 00000000000000000000000000000000
007813.152: 00000000000000000000000000000091
007818.842: HDMI0: best-mode 2 (limit 2) 1920x1080 60 Hz CEA modes 1e000f80000000000000000000000000 extensions 1
007830.437: Select resolution HDMI1/2 hotplug 0 max_mode 2
007837.358: FB0 disp 0 max-fb 2 1920x1080 stride 3840 base 0x3f800000
008669.858: dtb_file 'bcm2712-rpi-5-b.dtb'
008782.697: Loaded HAT overlay
008784.300: dtparam: audio=on
008792.925: Unknown dtparam 'audio' - ignored
008865.460: Loaded overlay 'vc4-kms-v3d-pi5'
008868.394: dtparam: pciex1=true
008882.637: dtparam: pciex1_gen=3
008893.240: dtparam: uart0=on
008904.700: dtparam: uart0_console=on
009052.291: Loaded overlay 'uart0-pi5'
009054.605: dtparam: rtc=off
009341.100: RPM 7828, max RPM 7828
011147.360: Starting OS 11147 ms
011152.969: 00000040: -> 00000480
011154.818: 00000030: -> 00100080
011159.530: 00000034: -> 00100080
011164.243: 00000038: -> 00100080
011168.956: 0000003c: -> 00100080
011278.530: sdram: sdram refresh 2081->4162 (2)
071231.750: initial_turbo of 60 deactivated

pi@cumulusmxkitchen:~ $ cat /sys/kernel/debug/dmaengine/summary
dma0 (1000010000.dma): number of channels: 5

dma1 (1000010600.dma): number of channels: 5
 dma1chan0    | 107d004000.spi:tx
 dma1chan1    | 107d004000.spi:rx
 dma1chan2    | 107c701400.hdmi:audio-rx
 dma1chan3    | 107c706400.hdmi:audio-rx

dma2 (1f00188000.dma): number of channels: 8

lgeitner, 2025-04-14:
4.  Are they the same kinds of errors as you were getting with DMA?  Yes
5.  After I enter the command:

    sudo systemctl stop cumulusmx     'This stops the MeteoPI.
    cat /dev/ttyAMA0                            

I do not see any output, its blank

What are the stty settings mentioned elsewhere?

Thanks,
Lance

pelwell, 2025-04-14:
It's OK - I crossed the streams with a similar UART application where the device streams data at you, like a GPS device, but I don't know if that's the case with MeteoPi. You could confirm what settings it is using if you restart the service and run `sudo stty -F /dev/ttyAMA0 -a`.

lgeitner, 2025-04-14:
Here is the output from the command `sudo stty -F /dev/ttyAMA0 -a`

pi@cumulusmxkitchen:~ $ sudo stty -F /dev/ttyAMA0 -a
speed 19200 baud; rows 0; columns 0; line = 0;
intr = ^C; quit = ^\; erase = ^?; kill = ^U; eof = ^D; eol = <undef>;
eol2 = <undef>; swtch = <undef>; start = ^Q; stop = ^S; susp = ^Z; rprnt = ^R;
werase = ^W; lnext = ^V; discard = ^O; min = 1; time = 0;
-parenb -parodd -cmspar cs8 hupcl -cstopb cread clocal -crtscts
ignbrk -brkint -ignpar -parmrk -inpck -istrip -inlcr -igncr -icrnl -ixon -ixoff
-iuclc -ixany -imaxbel -iutf8
-opost -olcuc -ocrnl onlcr -onocr -onlret -ofill -ofdel nl0 cr0 tab0 bs0 vt0 ff0
-isig -icanon -iexten -echo -echoe -echok -echonl -noflsh -xcase -tostop -echoprt
echoctl echoke -flusho -extproc

lgeitner, 2025-04-15:
Kernel Version 6.12.20

```
sudo systemctl stop cumulusmx
sudo stty -F /dev/ttyAMA0 19200 raw -echo -echoe -echok
cat /dev/ttyAMA0
```

There is no output.

Kernel Version 6.6.74

```
sudo systemctl stop cumulusmx
sudo stty -F /dev/ttyAMA0 19200 raw -echo -echoe -echok
cat /dev/ttyAMA0
```

There is no output.

pelwell, 2025-04-15:
Thanks for trying. This is going to be hard to diagnose without some kind of indication about what the error is.

lurch, 2025-04-15:
> Kernel Version 6.6.74
> 
> ```
> sudo systemctl stop cumulusmx
> sudo stty -F /dev/ttyAMA0 19200 raw -echo -echoe -echok
> cat /dev/ttyAMA0
> ```
> 
> There is no output.

Presumably that means that the MeteoPi only sends a "response" in reply to a "query" that it gets from the cumulusmx software running on the Pi? If you have a USB-UART adapter (e.g. something like https://thepihut.com/products/usb-to-ttl-serial-cable-debug-console-cable-for-raspberry-pi ) then I guess you could connect that to the UART pins on the Pi's GPIO header to see what "query" strings the cumulusmx software is sending?

lgeitner, 2025-04-17:
I noticed here [](https://github.com/raspberrypi/firmware/commits) that there was a 6.6.75, 6.6.76, 6.6.77, 6.6.78 Kernel Versions.

I know the MeteoPI works fine on 6.6.74.  I thought I could upgrade to Kernel Version 6.6.75 by running the command

`sudo rpi-update 8e7a0cc`  Which is the commit version for the 6.6.75 Kernel.  However, I receive invalid hash given.

 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
FW_REV:
BOOTLOADER_REV:816bf7c594a4c117ab2815f443ad3d535be475a6
 *** Invalid hash given
 
I would like to try to update the kernel one version at a time until I start seeing the errors with the MeteoPI.

I would assume that the 6.12.20 Kernel Version would include all the changes from the latest 6.6 Kernel which would be 6.6.78?  Plus any changes from 6.12.1 thru 6.12.19.

If I can upgrade the Kernel one version at a time, I should be able to identify what Kernel Version the errors start occurring with the MeteoPI?

Would it be possible to upgrade the Kernel One Version at a time?

Thank You,
Lance

lurch, 2025-04-23:
> Would it be possible to upgrade the Kernel One Version at a time?

IIRC `rpi-update` works with commit hashes from the https://github.com/raspberrypi/rpi-firmware/ repo, so looking at https://github.com/raspberrypi/rpi-firmware/commits/master/ I guess you might be able to do `sudo rpi-update 146f82c` to get the 6.6.75 kernel?

lgeitner, 2025-04-23:
Thank You for the above information I was able to upgrade the Kernel to Version 6.6.75 using the command:

```
sudo rpi-update 146f82c
sudo reboot
```

I was looking at the commits from [](https://github.com/raspberrypi/firmware/commits), instead of the commits from [](https://github.com/raspberrypi/rpi-firmware/commits/master)

Over the next couple of days, I will upgrade the Kernel one version at a time.  I will post the results here.

Thank You,
Lance

lgeitner, 2025-04-28:
Results of Testing Kernel one version at a time.

Kernel 6.6.75   MeteoPI has no errors!
Kernel 6.6.76   MeteoPI has no errors!
Kernel 6.6.77   MeteoPI has no errors!
Kernel 6.6.78   MeteoPI has no errors!
Kernel 6.12.15  MeteoPI has no errors!
Kernel 6.12.16  MeteoPI has no errors!
Kernel 6.12.17  MeteoPI has no errors!
Kernel 6.12.18  MeteoPI has no errors!
Kernel 6.12.19  MeteoPI has no errors!
Kernel 6.12.20  MeteoPI has errors!  

I found the solution to prevent any errors from occurring with the MeteoPI was to remove the following line from /boot/firmware/config.txt

dtparam=uart0_nodma

then `sudo reboot`

I no longer have any errors with the MeteoPI on the 6.12.20 Kernel!

Thank You,
Lance

lurch, 2025-04-28:
> was to remove the following line from `/boot/firmware/config.txt`
```
dtparam=uart0_nodma
```

i.e. You reverted `config.txt` to it's "out of the box" setting?

> I no longer have any errors with the MeteoPI on the 6.12.20 Kernel!

Yay, pleased to hear that!

lgeitner, 2025-05-06:
I Updated my Raspberry PI 5 using the following commands:

```
sudo apt update
sudo apt full-upgrade
sudo reboot
```

The MeteoPI started producing errors again!

`uname -a` showed I was upgraded to 6.12.25 Kernel Version.

the command `cat /sys/kernel/debug/dmaengine/summary` showed no dma for serial:tx and serial:rx

dma0 (1000010000.dma): number of channels: 5

dma1 (1000010600.dma): number of channels: 5
 dma1chan0    | 107d004000.spi:tx
 dma1chan1    | 107d004000.spi:rx
 dma1chan2    | 107c701400.hdmi:audio-rx
 dma1chan3    | 107c706400.hdmi:audio-rx

dma2 (1f00188000.dma): number of channels: 8

I ran the following command to switch to the 6.12.21 Kernel Version

```
sudo rpi-update b07b133
sudo reboot
```

`uname -a` shows

Linux cumulusmxkitchen 6.12.21-v8-16k+ #1868 SMP PREEMPT Mon Mar 31 17:10:57 BST 2025 aarch64 GNU/Linux

the command `cat /sys/kernel/debug/dmaengine/summary` shows no dma for serial:tx and serial:rx

dma0 (1000010000.dma): number of channels: 5

dma1 (1000010600.dma): number of channels: 5
 dma1chan0    | 107d004000.spi:tx
 dma1chan1    | 107d004000.spi:rx
 dma1chan2    | 107c701400.hdmi:audio-rx
 dma1chan3    | 107c706400.hdmi:audio-rx

dma2 (1f00188000.dma): number of channels: 8

Below is the output for the Kernel Version 6.12.20 that works!

`cat /sys/kernel/debug/dmaengine/summary`

dma0 (1000010000.dma): number of channels: 5

dma1 (1000010600.dma): number of channels: 5
 dma1chan0    | 107d004000.spi:tx
 dma1chan1    | 107d004000.spi:rx
 dma1chan2    | 107c701400.hdmi:audio-rx
 dma1chan3    | 107c706400.hdmi:audio-rx

dma2 (1f00188000.dma): number of channels: 8
 **dma2chan0    | 1f00030000.serial:tx
 dma2chan1    | 1f00030000.serial:rx**

`dmesg | grep "serial"`

[    0.000000] Kernel command line: reboot=w coherent_pool=1M 8250.nr_uarts=1 pci=pcie_bus_safe cgroup_disable=memory numa_policy=interleave  numa=fake=8 system_heap.max_order=0 smsc95xx.macaddr=D8:3A:DD:BC:F2:62 vc_mem.mem_base=0x3fc00000 vc_mem.mem_size=0x40000000  root=PARTUUID=6f803e6a-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
[    0.037885] 107d001000.serial: ttyAMA10 at MMIO 0x107d001000 (irq = 16, base_baud = 0) is a PL011 rev3
[    0.451972] 107d50c000.serial: ttyS0 at MMIO 0x107d50c000 (irq = 33, base_baud = 6000000) is a Broadcom BCM7271 UART
[    0.452011] serial serial0: tty port ttyS0 registered
[    0.749595] pl011-axi 1f00030000.serial: cts_event_workaround enabled
[    0.749667] 1f00030000.serial: ttyAMA0 at MMIO 0x1f00030000 (irq = 131, base_baud = 0) is a PL011 AXI
[    1.594463] systemd[1]: Created slice system-serial\x2dgetty.slice - Slice /system/serial-getty.
[    2.204655] hci_uart_bcm serial0-0: supply vbat not found, using dummy regulator
[    2.204716] hci_uart_bcm serial0-0: supply vddio not found, using dummy regulator
**[    7.865762] pl011-axi 1f00030000.serial: DMA channel TX dma2chan0
[    7.865776] pl011-axi 1f00030000.serial: DMA channel RX dma2chan1**

**How do I re-enable DMA on the Kernel for Kernel Version 6.12.21 or later?**

pelwell, 2025-05-06:
There's a new PR giving the Pi 5 family a `uart0_dma` parameter: #6838
In about 40 minutes you will be able to install a trial build using `sudo rpi-update pulls/6838`, after which you should find that putting `dtparam=uart0_dma` in config.txt re-enables DMA.

lgeitner, 2025-05-06:
added `dtparam=uart0_dma` to /boot/firmware/config.txt

then ran `sudo rpi-update pulls/6838` then `sudo reboot`

`uname -a` displays the following:

Linux cumulusmxkitchen 6.12.25-v8-16k+ #1 SMP PREEMPT Tue May  6 13:42:48 UTC 2025 aarch64 GNU/Linux
 
`cat /sys/kernel/debug/dmaengine/summary` displays the following:

dma0 (1000010000.dma): number of channels: 5

dma1 (1000010600.dma): number of channels: 5
 dma1chan0    | 107d004000.spi:tx
 dma1chan1    | 107d004000.spi:rx
 dma1chan2    | 107c701400.hdmi:audio-rx
 dma1chan3    | 107c706400.hdmi:audio-rx

dma2 (1f00188000.dma): number of channels: 8
 dma2chan0    | 1f00030000.serial:tx
 dma2chan1    | 1f00030000.serial:rx
pi@cumulusmxkitchen:~ $ 

There are no errors anymore with the MeteoPI !

Thank You,
Lance

pelwell, 2025-05-06:
I'll get that merged, and it will become a standard feature.

ausfas, 2025-05-08:
I'm not sure if this issue will resolve the DMA problem that I've been facing. 
https://github.com/networkupstools/nut/issues/2805

[image]

pelwell, 2025-05-08:
There's a good chance that updating to the current kernel (6.12.25 or later) will get it working again.