raspberrypi/linux #6372: Enablement of dm-verity

JakubVanek, 2024-09-18:
### Describe the bug

I am working on a project that is currently compiling its own RPi kernel binaries. I'm considering moving instead to the kernel binaries provided by Raspberry Pi. However, I'm missing one kernel functionality in the official images: dm-verity. I depend on dm-verity for RAUC (project providing infrastructure for image-based updates). RAUC requires the following kernel options https://rauc.readthedocs.io/en/latest/integration.html#kernel-configuration and it seems to me that all, except for the `CONFIG_DM_VERITY`, are already satisfied. Would it be possible for you to enable `CONFIG_DM_VERITY=m` in your builds?

### Steps to reproduce the behaviour

Overview of what I am trying to achieve:
1. install RAUC on a RPi and on a PC
2. create a RAUC `verity` image on a PC
3. successfully flash the `verity` image into a new partition on the RPi

### Device (s)

Raspberry Pi 4 Mod. B, Raspberry Pi 5, Raspberry Pi CM4

### System

n/a (custom Buildroot-based OS)

pelwell, 2024-09-19:
Can you confirm that the config changes in #6375 are sufficient? Wait about 45 minutes, then run `sudo rpi-update pulls/6375` to install a trial build.

JakubVanek, 2024-09-19:
Yes, the changes are sufficient. I was able to insert the 2024-07-04-raspios-bookworm-armhf-lite.img.xz image (with the `rpi-update` applied) into my A/B partition layout and boot flow. I was then able to flash my current RAUC verity images into the inactive A/B partition from within Raspbian. Thank you!

closed 2024-09-19 by JakubVanek