raspberrypi/linux #6373: CM3 HDMI audio : bad sampling frequency and unstable audio channel status

jeanphi-francois, 2024-09-19:
### Describe the bug

When playing a sine or multitone signal (48 kHz sampling rate) into an AudioPrecision audio analyser.
Sound is played but Performance is horrible (15 % THD+N)
Sampling frequency reported by Analyzer is  50.5 kHz instead of the expected 48 kHz
Audio Status bits are not stable : The copyright, Emphasis and Category Code are moving.

### Steps to reproduce the behaviour

Set Video with modetest -M vc4 -s 32:1280x720. Video is fine.
aplay  --device sysdefault:CARD=vc4hdmi  /tmp/channel_multitone_48k.wav

I initially doubted my aplay command, and was not sure if the right status bits were sent, so I instrumented alsalib, and the status bits are correctly set, so the vc4-hdmi.conf file is correctly used.

### Device (s)

Raspberry Pi CM3+

### System

OS : buildroot based
Firmware : 
[    0.070392] raspberrypi-firmware soc:firmware: Attached to firmware from 2022-01-17T19:22:03, variant start
[    0.080405] raspberrypi-firmware soc:firmware: Firmware hash is bd34f55ef7b01b0a367f131060b561a2a58b80bb

Kernel : 
Linux isp-host 6.1.21-v4 #1 SMP Tue Sep 17 16:58:25 CEST 2024 armv7l GNU/Linux
(raspberry kernel, using KMS driver)

Alsalib is 1.2.8

### Additional context

I initially suspected I was maybe doing something wrong on the alsa side, so I went to the troubleshooting section of the forum.
I am now confident that it is not a bad aplay command or alsa conf issue.
Forum issue : https://forums.raspberrypi.com/viewtopic.php?t=376835

popcornmix, 2024-09-19:
Perhaps have a read of https://github.com/raspberrypi/linux/issues/5525
If it is that there are a couple of workarounds in there.

jeanphi-francois, 2024-09-19:
Yeah I read that, doesn't look like the same issue to me.

popcornmix, 2024-09-19:
Did you try https://github.com/raspberrypi/linux/issues/5525#issuecomment-1615922687

pelwell, 2024-09-19:
Does the audio analyser have an HDMI input? What's your configuration.

popcornmix, 2024-09-19:
I have an analyser I can check with later.
I could see csb instability when investigating the issue I linked (but only when libalsa was using the wrong sampling freq for AES). But it was stable when sample freq agreed.
I would have been testing on Pi4/5, so I can try on a Pi3.

jeanphi-francois, 2024-09-19:
It works with a CM4S
Audio channel status are stable
THD+N is at -96 dB when playing a 16 bit file.
Audio clock is measured at 48.0005 kHz

pelwell, 2024-09-19:
Is that running exactly the same image?

jeanphi-francois, 2024-09-19:
Yes, same image

jeanphi-francois, 2024-09-19:
Which aplay command should I use to play  8 channel audio ?
The iec958 plugin is always configured for two channel.

popcornmix, 2024-09-19:
I normally test with speaker-test. e.g.
```
speaker-test -D sysdefault:CARD=vc4hdmi -r48000 -c2
```
My analyser shows stable csb on Pi3+ (it shows 0x4,0x82,0x0,0x2,0x2) with this command.

Is it unstable for you?

popcornmix, 2024-09-20:
Also tried with:
```
aplay --device sysdefault:CARD=vc4hdmi dell/audio/testout2.wav```
for 48kHz, 44.1kHz and 8 channel wav files, and all have stable csb for me.

Also tried with `modetest -M vc4 -s 33:1280x720` active.

I do note you are on 6.1 kernel, and using your own builtroot.

Probably best for you to test with RPiOS bookworm (using 6.6 kernel) and see if that works for you.
If it does we can try to narrow down what is the difference (likely kernel, but could be alsa).

jeanphi-francois, 2024-09-23:
Ok I checked with an RPI3B and I have the same issue.
I will check with 6.6 kernel

jeanphi-francois, 2024-10-08:
Hi,

I updated my kernel to 6.6 using linux-stable_20240529 tag in the repo.
I still have the same behaviour : it is working fine on the CM4S, but not on RPI3.

popcornmix, 2024-10-08:
Can you test a clean install of RPiOS bookworm and run:
```speaker-test -D sysdefault:CARD=vc4hdmi -r48000 -c2```
and see what your analyser shows?

jeanphi-francois, 2024-10-08:
I will do what you ask. In the meantime, I tested with kernel hosted on this github in the firmware repo, and the corresponding modules. Still no luck.
So far it doesn't look like a kernel or userspace issue, since both are the same between a rpi3b / cm3+ and my cm4.
Maybe I should look into the config.txt and the device tree ?
But first lets give a try to the RPiOS

jeanphi-francois, 2024-10-08:
Well, it works fine with the last RPiOS Lite. I will try to copy kernel and modules to another unit.

jeanphi-francois, 2024-10-09:
Hi,
New data points :
RPI Os Kernel + RPI Os userspace + RPI Os firmware / config => OK
RPI Os kernel + My userspace + my Firmware => Not OK
My Kernel + RPI Os userspace + RPI Os firmware / config => OK

Since my userspace + kernel combination works with CM4, I don't think the userspace is at fault.
So I will experiment with

- The firmware files. I think I already tried that
- The overlay files. I already use the latest dtbo for the vc4-kms-v3d overlay
- The config.txt file

If you have any advice on where to look first, I am all ears :)

popcornmix, 2024-10-09:
It's very unlikely to be firmware (assuming you mean botcode.bin / start*.elf / fixup*.dat).
When you use the vc4-kms-v3d overlay, the hdmi hardware is not touched by the firmware apart from the initial splash.
Everything hdmi related is handled on the arm.

Make sure kernel (kernel*.img) and dtbs (bcm*.dtb and overlays/*.dtbo) are kept in sync - never mix and match them from different builds.

I can't think of anything in the config.txt that could affect this.

So, my guess would be alsa userspace. Either the version of libalsa, or the alsa config/card files.

jeanphi-francois, 2024-10-10:
Turns out it was the firmware.
Switchin between two sets of start.elf + fixup.dat would either solve the issue or bring it back.

I have a question though, why is there a v7l and a v7 variant, when it is perfectly possible to have a single kernel to rule them all ?
What does the v7l version improves ?
It is unrelated to the issue at hand, but I couldn't find an explanation for it.

Thank you for guiding me through this debugging journey.

closed 2024-10-10 by jeanphi-francois

pelwell, 2024-10-10:
The `l` in `v7l` stands for Large Physical Address Extension. It's an ARM technique for accessing a physical address space greater than 32-bits from a 32-bit virtual address. We need LPAE on the Pi 4 to address the hardware because most of it sits above the 4GB boundary - there's no room otherwise with 4GB RAM. However, LPAE has a cost in terms of RAM usage, so we don't want to enable it for Pi 3 etc.

If we were to make it change to the kernel line-up it would be to drop the v7l variant and force Pi 4 to run a 64-bit kernel, which is the recommended configuration.

popcornmix, 2024-10-10:
> Switchin between two sets of start.elf + fixup.dat would either solve the issue or bring it back.

What was the `vcgencmd version` for the good and bad firmware?

jeanphi-francois, 2024-10-10:
The working one is 
```
[ 0.070071] raspberrypi-firmware soc:firmware: Attached to firmware from 2024-05-24T15:31:28, variant start
[ 0.080089] raspberrypi-firmware soc:firmware: Firmware hash is 4942b7633c0ff1af1ee95a51a33b56a9dae47529

```
Bad one is
```

[ 0.070392] raspberrypi-firmware soc:firmware: Attached to firmware from 2022-01-17T19:22:03, variant start
[ 0.080405] raspberrypi-firmware soc:firmware: Firmware hash is bd34f55ef7b01b0a367f131060b561a2a58b80bb
```

popcornmix, 2024-10-10:
Okay, that is pretty damned old. I don't think it's worth trying to bisect exactly what the change was, as you have the fix.

As the firmware won't be touching hdmi hardware when you are outputting audio, I'd guess the difference is something more subtle (perhaps clock related), but in two and a half years, there's been lots of fixes.