raspberrypi/linux #6396: cgroups v1 no longer available in standard settings of kernel 6.12

by, 2024-10-03:
### Describe the bug

Please remember that cgroups v1 is now only available via CONFIG_CPUSETS_V1, which is OFF by default in kernel 6.12 (v2 is always available).
So if you are using cpuset or alike, which still relies on cgroups v1, you need to switch it on explicitly in kernel configurations.

### Steps to reproduce the behaviour

N/A

### Device (s)

Other

### System

All

### Logs

N/A

### Additional context

See also https://lore.kernel.org/lkml/20240827062111.580296-1-chenridong@huawei.com/T/

pelwell, 2024-10-03:
Thanks. This could also have been expressed in the form of a pull request.

by, 2024-10-03:
Agreed, @pelwell, but I was unsure about your future rpi-kernel strategy vs. upstream re. cgroups, so no proposal on my side, just FYI.

pelwell, 2024-10-03:
Understood. I guess I'm saying that we wouldn't consider a sensibly worded PR to be presumptuous - just efficient.

pelwell, 2024-10-03:
See https://github.com/raspberrypi/linux/pull/6397.

closed 2024-10-03 by by