raspberrypi/linux #6402: [rpi zero 2w][KMSDRM] CX31993 USB sound card; crackle sound; USB driver issue; ALSA lib pcm.c:8570:(snd_pcm_recover) underrun occurred 0B f=0/0 #265

qrp73, 2024-10-05:
### Describe the bug

When using the CX31993 USB sound card on a Raspberry Pi Zero 2W with application running in KMSDRM mode, the audio is choppy, even when selecting a low sampling rate. However, the same USB sound card works without interruptions on a Raspberry Pi 4B. Additionally, using a different USB sound card on the Raspberry Pi Zero 2W does not result in audio issues. 

This problem has been [observed](https://github.com/raspberrypi/bookworm-feedback/issues/265) for a long time ago and remains unresolved, even with the latest updates. 

It is suspected that the issue is related to the USB device driver, as the interruptions only occur with this specific device RPI02W and CX31993 USB sound card combination.

### Steps to reproduce the behaviour

1) Use Raspberry Pi Zero 2W with the default clock speed, running on latest Raspi-OS lite aarch64.

2) Connect an HDMI display (using a 1280x1024@60 resolution).

3) Connect a USB hub with a keyboard and a CX31993 USB sound card.

4) Run ```sudo reboot``` to restart the system.

5) Create a test directory and navigate to it:
```
mkdir TEST && cd TEST
```
6) Download the test audio file:
```
wget https://www.mediacollege.com/audio/tone/files/1kHz_44100Hz_16bit_05sec.wav
```

7) Play the audio file using aplay:
```
aplay 1kHz_44100Hz_16bit_05sec.wav
```
Expected result: The sound is played correctly, without issues.

8) Play the same audio file using ```ffplay``` with ```SDL_VIDEODRIVER=kmsdrm```:
```
SDL_VIDEODRIVER=kmsdrm ffplay 1kHz_44100Hz_16bit_05sec.wav
```

Expected result: The sound should be played correctly.

Actual result: The sound is played choppy with crackling noises, and multiple errors appear in the console output:
```
ALSA lib pcm.c:8570:(snd_pcm_recover) underrun occurred  0B f=0/0
ALSA lib pcm.c:8570:(snd_pcm_recover) underrun occurred  0B f=0/0
ALSA lib pcm.c:8570:(snd_pcm_recover) underrun occurred  0B f=0/0
...
```

### Device (s)

Raspberry Pi Zero 2 W

### System

```
$ uname -a && cat /etc/rpi-issue && vcgencmd version
Linux rpi02w 6.6.51+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt2 (2024-10-01) aarch64 GNU/Linux
Raspberry Pi reference 2024-03-15
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, f19ee211ddafcae300827f953d143de92a5c6624, stage2
Aug 30 2024 19:19:11 
Copyright (c) 2012 Broadcom
version 2808975b80149bbfe86844655fe45c7de66fc078 (clean) (release) (start)
```

### Logs

```
ALSA lib pcm.c:8570:(snd_pcm_recover) underrun occurred  0B f=0/0
ALSA lib pcm.c:8570:(snd_pcm_recover) underrun occurred  0B f=0/0
ALSA lib pcm.c:8570:(snd_pcm_recover) underrun occurred  0B f=0/0
...
```

### Additional context

This bug is not related to the ```ffplay``` code. I initially encountered this issue in my own SDL application that uses the ```librtaudio-dev``` library for audio playback. Upon further investigation, I found that the same problem occurs in ```ffplay``` when running in KMSDRM mode. Therefore, to simplify reproducing the issue, I used the standard ```ffplay``` program.

popcornmix, 2024-10-05:
You would probably be better off with a 32-bit install.

32-bit kernel uses the FIQ to improve performance of the dtw_otg usb driver.
There is not FIQ support on the 64-bit kernel, which reduces USB performance.

pelwell, 2024-10-05:
You will also get a bit more free memory on 32-bit, which can make a useful difference on a Zero 2 W.

qrp73, 2024-10-05:
I prefer a 64-bit kernel as it enables 64-bit integer computations with reduced overhead. I experimented with a 32-bit kernel, but encountered significant performance bottlenecks for my algorithms and additional issues.