raspberrypi/linux #6404: vc4 driver crashing while watching videos on kodi

EdiDD, 2024-10-07:
### Describe the bug

While watching videos on kodi 20.5 journal is flooded with errors:
`kernel: [drm:validate_tile_binning_config [vc4]] *ERROR* Failed to allocate binner memory: -12`
causing problems with watching movies.

### Steps to reproduce the behaviour

Watch movies without issues.

### Device (s)

Raspberry Pi 3 Mod. B

### System

- raspbian based on debian 12.7
- kernel 6.6.51+rpt-rpi-v8

### Additional context

config.txt:
dtoverlay=vc4-kms-v3d
gpu_mem=160

pelwell, 2024-10-07:
1. Is this a recent change in behaviour?
2. Does increasing of decreasing gpu_mem alter the behaviour?

EdiDD, 2024-10-07:
1. Yes. this is a recent change but i can't tell when (from which kernel ver.) this has started.  
2. Increasing gpu_mem to 256M or decreasing to 120M (so low causes movies to glitches / stuttering) has no effect on vc4 errors .

popcornmix, 2024-10-07:
Are you using the default skin (estuary)?
Are you booting to desktop or console?
If desktop are you running kodi or kodi fullscreen?
If desktop, does it work if you switch to boot to console then run kodi?

popcornmix, 2024-10-08:
In order to reproduce this, it would be useful to have a precise, minimal  list of things you do to see the first complaint in dmesg.

Is launching kodi and playing one video enough to provoke? If not, what is needed?
So we are comparing like with like, can you provoke the issue playing a known, public file? (e.g. [this](https://test-videos.co.uk/vids/bigbuckbunny/mp4/h264/1080/Big_Buck_Bunny_1080_10s_1MB.mp4))

EdiDD, 2024-10-08:
I will try to find some conditions (if any) to provoke this issue but for now this is some example of movie where the problem occurred:

- Video: h264 (High) (avc1), yuv420p(tv, progressive), 1920x1038 [SAR 1:1 DAR 320:173], 3041 kb/s, 23.98 fps, 23.98 tbr, 24k tbn (default)
- Audio: aac (LC) (mp4a), 48000 Hz, 5.1, fltp, 341 kb/s (default)
- With subtitles  *.srt - this can be important - when subtitles hang (just displays one line constantly) I know for sure that errors has started flooding

Errors appear after some time not instantly from the beginning of a movie. It can be 1min or 10min after.

EdiDD, 2024-10-11:
Can't provoke it. It starts flooding randomly with or without subtitles.  Meaningless if double or triple buffer is set in kodi or direct to plane or egl render method is used. Additionally I checked that there are no other problems:

- no storage read errors (badblocks)
- no memory problems (memtester)
- no additional kernel parameters (sysctl)
- no overheating

Looks like vc4 driver is faulty.

EdiDD, 2025-03-17:
Is raspberry pi3 support dropped? because this error still occur with kernel `6.6.74+rpt-rpi-v8` but it is not reproducible with kernel `linux-image-6.1.0-rpi8-rpi-v8`

popcornmix, 2025-03-17:
Are you saying the only change between it working and failing is switching the kernel between 6.6.74 and 6.1.0?
You say you are on "raspbian based on debian 12.7"?

Raspbian isn't the name of an OS (it's a rebuild of debian packages for 32-bit arm with hardfp which may be used by a number of OS's).
Did you install RPi OS bookworm? 32-bit or 64-bit?

EdiDD, 2025-03-17:
Yes. One and the only change is switching kernel to 6.1.0 from repo and the problem disappears. RPI OS is 64bit

popcornmix, 2025-03-17:
I've just tried a 1080p h264 video file with kodi.
I used [this](https://download.blender.org/demo/movies/BBB/bbb_sunflower_1080p_30fps_normal.mp4.zip) file.
I copied an srt subtitle file (naming it bbb_sunflower_1080p_30fps_normal.srt).
And tried playing it on RPiOS Bookworm 64-bit on a Pi3. It seems fine.

I haven't adjusted gpu_mem, as the default (76M) should play any level 4.1 (that is BluRay spec).
We don't claim support for higher level profiles.
Please try with default gpu_mem and the file I linked to.
If that plays for you, post a mediainfo of a file that doesn't play.

EdiDD, 2025-03-17:
This is too short video, as i mentioned above these errors and freezes are randomly, it can occur after 15min or more like 45 min, but happens on every movie. Could you try with normal full length movie?  
I tested it with default `gpu_mem` and with higher values like 256M - nothing changed.

popcornmix, 2025-03-17:
Does a short video (e.g. the one I linked) set to repeat have the issue after 45m?
(you can enable repeat by adding video to playlist, then in the playlist sidebar enable repeat).

I've got BluRay iso of The Amazing Spider-Man here (2h 16m). I've enabled subtitles.
I'll see if that plays to the end.

EdiDD, 2025-03-18:
Thank you for checking this spider-man. Please let me know if this worked for you?

popcornmix, 2025-03-18:
The Amazing Spider-Man played to the end for me on Pi3. In fact I played it on loop and it's still playing now (17 hours later).

EdiDD, 2025-03-18:
You have different kernel, b3plus ? Mine is:
`Linux raspberrypi 6.6.74+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.74-1+rpt1 (2025-01-27) aarch64 GNU/Linux`

popcornmix, 2025-03-18:
I originally tested with 6.12 kernel (which is in beta testing) which was also fine. 
I build the 6.6.78 kernel myself, just to confirm the 6.12 tree didn't fix it.

Can you try without the gpu_mem line (which is wasting 84M).
If you get a crash (or it fails to play) the post the mediainfo of the file you are playing.

EdiDD, 2025-03-18:
Ok. I will try but as i remembered correctly with such low gpu_mem as 76M movies where not playing smoothly - playback was very choppy and all is playing smoothly with gpu_mem about 160M. Unfortunately, I won't be able to check it until after the weekend. Maybe than i will try to play some free youtube movie which causes these kind of problems and give you a link.