raspberrypi/linux #6406: mcp251xfd and kernel panic

pompushko, 2024-10-08:
### Describe the bug

Hello

I have pretty simple config with mcp2518fd SPI connected CAN-bus controllers.
With super fresh build of Raspbian, I start facing kernel panic during boot. Only disconnecting of CAN-bus HAT can help.

### Steps to reproduce the behaviour

Just setup and use any of CAN-bus HAT with mcp2518fd. 

### Device (s)

Raspberry Pi Zero 2 W

### System

Raspberry Pi reference 2024-10-08
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, dbadbe2868e2fe5548024f145ea2726e9c58d71e, stage3

Aug 30 2024 19:19:11
Copyright (c) 2012 Broadcom
version 2808975b80149bbfe86844655fe45c7de66fc078 (clean) (release) (start)

Linux canbus 6.6.51+rpt-rpi-v7 #1 SMP Raspbian 1:6.6.51-1+rpt2 (2024-10-01) armv7l GNU/Linux

### Logs

Oct 08 21:16:31 canbus kernel: 8<--- cut here ---
Oct 08 21:16:31 canbus kernel: Unable to handle kernel NULL pointer dereference at virtual address 00000000 when read
Oct 08 21:16:31 canbus kernel: [00000000] *pgd=03ac1835, *pte=00000000, *ppte=00000000
Oct 08 21:16:31 canbus kernel: Internal error: Oops: 17 [#1] SMP ARM
Oct 08 21:16:31 canbus kernel: Modules linked in: brcmfmac_wcc brcmfmac vc4 snd_soc_hdmi_codec drm_display_helper brcmutil cec hci_uart drm_dma_helper btbcm drm_kms_helper bluetooth cfg80211 snd_soc_core snd_compress raspberrypi_hwmon snd_pcm_dmaengine binfmt_misc bcm2835_codec(C) mcp251xfd bcm2835_v4l2(C) ecdh_generic ecc can_dev bcm2835_isp(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_vmalloc rfkill videobuf2_dma_contig snd_bcm2835(C) videobuf2_memops videobuf2_v4l2 snd_pcm videodev snd_timer raspberrypi_gpiomem snd videobuf2_common vc_sm_cma(C) mc uio_pdrv_genirq uio can_gw can drm fuse drm_panel_orientation_quirks dm_mod backlight ip_tables x_tables ipv6 spidev i2c_bcm2835 spi_bcm2835aux spi_bcm2835 fixed
Oct 08 21:16:31 canbus kernel: CPU: 0 PID: 479 Comm: ip Tainted: G         C         6.6.51+rpt-rpi-v7 #1  Raspbian 1:6.6.51-1+rpt2
Oct 08 21:16:31 canbus kernel: Hardware name: BCM2835
Oct 08 21:16:31 canbus kernel: PC is at timecounter_read+0x14/0xac
Oct 08 21:16:31 canbus kernel: LR is at mcp251xfd_ring_init+0x1f0/0x738 [mcp251xfd]
Oct 08 21:16:31 canbus kernel: pc : [<801c1418>]    lr : [<7f2be43c>]    psr: 20000013
Oct 08 21:16:31 canbus kernel: sp : 9cbe9998  ip : ffffc000  fp : 00000000
Oct 08 21:16:31 canbus kernel: r10: 00000001  r9 : 00000460  r8 : 00000460
Oct 08 21:16:31 canbus kernel: r7 : 00000000  r6 : 8377a600  r5 : 00000000  r4 : 8377a970
Oct 08 21:16:31 canbus kernel: r3 : 83779000  r2 : 8377a970  r1 : 00000004  r0 : 00000000
Oct 08 21:16:31 canbus kernel: Flags: nzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
Oct 08 21:16:31 canbus kernel: Control: 10c5383d  Table: 1ba8406a  DAC: 00000055
Oct 08 21:16:31 canbus kernel: Register r0 information: NULL pointer
Oct 08 21:16:31 canbus kernel: Register r1 information: non-paged memory
Oct 08 21:16:31 canbus kernel: Register r2 information: non-slab/vmalloc memory
Oct 08 21:16:31 canbus kernel: Register r3 information: non-slab/vmalloc memory
Oct 08 21:16:31 canbus kernel: Register r4 information: non-slab/vmalloc memory
Oct 08 21:16:31 canbus kernel: Register r5 information: NULL pointer
Oct 08 21:16:31 canbus kernel: Register r6 information: non-slab/vmalloc memory
Oct 08 21:16:31 canbus kernel: Register r7 information: NULL pointer
Oct 08 21:16:31 canbus kernel: Register r8 information: non-paged memory
Oct 08 21:16:31 canbus kernel: Register r9 information: non-paged memory
Oct 08 21:16:31 canbus kernel: Register r10 information: non-paged memory
Oct 08 21:16:32 canbus kernel: Register r11 information: NULL pointer
Oct 08 21:16:32 canbus kernel: Register r12 information: non-paged memory
Oct 08 21:16:32 canbus kernel: Process ip (pid: 479, stack limit = 0xefae9b57)
Oct 08 21:16:32 canbus kernel: Stack: (0x9cbe9998 to 0x9cbea000)
Oct 08 21:16:32 canbus kernel: 9980:                                                       8160d000 00000000
Oct 08 21:16:32 canbus kernel: 99a0: 8377a600 7f2be43c 83779000 7f2bbfd8 83778600 81555800 8377a970 ffffc000
Oct 08 21:16:32 canbus kernel: 99c0: 00000000 00000000 83778600 81555800 816f3c00 8377a000 8377818c 00000001
Oct 08 21:16:32 canbus kernel: 99e0: 00000000 7f2bbfe8 00000000 00000000 00000000 816f3c00 00000000 83778600
Oct 08 21:16:32 canbus kernel: 9a00: 83778000 7f2bc13c 7f2fa0b4 00040080 83778000 9cbe9cbc 8244d8c0 7f2fa0b4
Oct 08 21:16:32 canbus kernel: 9a20: 83778024 8377818c 00000001 8096b584 834fd100 834fd11c 9cbe9b84 83778000
Oct 08 21:16:32 canbus kernel: 9a40: 9cbe9cbc fd4725ac 83778000 00040081 8244d8c0 9cbe9cbc 00040080 8096ba2c
Oct 08 21:16:32 canbus kernel: 9a60: 01ac7000 01ae7fff 9b95940c 76ec4000 ffffffff fd4725ac 83778000 83d17c00
Oct 08 21:16:32 canbus kernel: 9a80: 00000000 00040080 7f2fa0b4 8210aad0 81ecfe40 8096bac4 00000000 83fee6c0
Oct 08 21:16:32 canbus kernel: 9aa0: 01ac7000 83778000 83d17c00 00000000 9cbe9cbc 8097c6e4 00000002 83fee6c0
Oct 08 21:16:32 canbus kernel: 9ac0: 01ac7000 01ae7fff 9b95940c 76ec4000 ffffffff 9b959500 00000002 00000000
[... 39 lines cut ...]
Oct 08 21:16:32 canbus kernel: 9fc0: 0009cca0 00000000 00000002 00000128 67059320 00083954 0009cca0 00000000
Oct 08 21:16:32 canbus kernel: 9fe0: 0009cd68 7eda6618 0007f4dc 76e0412c 20000010 00000003 00000000 00000000
Oct 08 21:16:32 canbus kernel:  timecounter_read from mcp251xfd_ring_init+0x1f0/0x738 [mcp251xfd]
Oct 08 21:16:32 canbus kernel:  mcp251xfd_ring_init [mcp251xfd] from mcp251xfd_chip_start+0x244/0x2a8 [mcp251xfd]
Oct 08 21:16:32 canbus kernel:  mcp251xfd_chip_start [mcp251xfd] from mcp251xfd_open+0x80/0x210 [mcp251xfd]
Oct 08 21:16:32 canbus kernel:  mcp251xfd_open [mcp251xfd] from __dev_open+0x114/0x1c8
Oct 08 21:16:32 canbus kernel:  __dev_open from __dev_change_flags+0x194/0x20c
Oct 08 21:16:32 canbus kernel:  __dev_change_flags from dev_change_flags+0x20/0x5c
Oct 08 21:16:32 canbus kernel:  dev_change_flags from do_setlink+0x35c/0x1020
Oct 08 21:16:32 canbus kernel:  do_setlink from rtnl_newlink+0x530/0x950
Oct 08 21:16:32 canbus kernel:  rtnl_newlink from rtnetlink_rcv_msg+0x24c/0x2f8
Oct 08 21:16:32 canbus kernel:  rtnetlink_rcv_msg from netlink_rcv_skb+0xc0/0x120
Oct 08 21:16:32 canbus kernel:  netlink_rcv_skb from netlink_unicast+0x194/0x270
Oct 08 21:16:32 canbus kernel:  netlink_unicast from netlink_sendmsg+0x1f8/0x480
Oct 08 21:16:32 canbus kernel:  netlink_sendmsg from __sock_sendmsg+0x44/0x78
Oct 08 21:16:32 canbus kernel:  __sock_sendmsg from ____sys_sendmsg+0x1f4/0x21c
Oct 08 21:16:32 canbus kernel:  ____sys_sendmsg from ___sys_sendmsg+0x9c/0xd0
Oct 08 21:16:32 canbus kernel:  ___sys_sendmsg from sys_sendmsg+0x78/0xbc
Oct 08 21:16:32 canbus kernel:  sys_sendmsg from __sys_trace_return+0x0/0x10
Oct 08 21:16:32 canbus kernel: Exception stack(0x9cbe9fa8 to 0x9cbe9ff0)
Oct 08 21:16:32 canbus kernel: 9fa0:                   0009cca0 00000000 00000003 7eda6670 00000000 00000001
Oct 08 21:16:32 canbus kernel: 9fc0: 0009cca0 00000000 00000002 00000128 67059320 00083954 0009cca0 00000000
Oct 08 21:16:32 canbus kernel: 9fe0: 0009cd68 7eda6618 0007f4dc 76e0412c
Oct 08 21:16:32 canbus kernel: Code: e52de004 e28dd004 e1a04000 e5900000 (e5903000) 
Oct 08 21:16:32 canbus kernel: ---[ end trace 0000000000000000 ]---

pelwell, 2024-10-08:
From the crash dump I'd say the timecounter `priv->tc`'s `cc` field was NULL at the time when `timecounter_read` was called from `mcp251xfd_ring_init`. It looks as though it should already have been initialised by the call to `timecounter_init` from `mcp251xfd_timestamp_start`, but that may not be the case.

@marckleinebudde It looks like a group of patches from you were merged into this time interval (6.6.31 to 6.6.51) - do you have any ideas what may be going wrong?

pompushko, 2024-10-08:
https://nvd.nist.gov/vuln/detail/CVE-2024-41088

That's could be an issue?

pelwell, 2024-10-08:
To me, the CVE doesn't seem to fit the symptoms you are seeing, but I'm not going to rule out the possibility.

marckleinebudde, 2024-10-09:
> @marckleinebudde It looks like a group of patches from you were merged into this time interval (6.6.31 to 6.6.51) - do you have any ideas what may be going wrong?

Known problem, see https://lore.kernel.org/all/20240924-truthful-authentic-basilisk-aaab90-mkl@pengutronix.de/

Please upgrade to v6.6.53 or cherry pick:
51b2a7216122 ("can: mcp251xfd: properly indent labels")
a7801540f325 ("can: mcp251xfd: move mcp251xfd_timestamp_start()/stop() into mcp251xfd_chip_start/stop()")

pelwell, 2024-10-09:
Thanks, Marc. The timeline on 6.6 is surprising, but I guess the lack of a Fixes: tag meant it slipped through the back-ports net.

@pompushko The fixes are already in the latest rpi-update build - `sudo rpi-update`. Note that this is beta software and should be treated with caution - back up valuable data just in case.

pompushko, 2024-10-09:
> Thanks, Marc. The timeline on 6.6 is surprising, but I guess the lack of a Fixes: tag meant it slipped through the back-ports net.
>
> @pompushko The fixes are already in the latest rpi-update build - [...]

Thank you for fast bugfix!

Fix in fresh version of kernel? Or how it looks like?

marckleinebudde, 2024-10-09:
I mainlined the series https://lore.kernel.org/all/20240628-mcp251xfd-workaround-erratum-6-v4-0-53586f168524@pengutronix.de via can-next and didn't include Fixes Tags nor put stable on Cc, as the changes were rather intrusive and the bug triggers very rarely in normal use cases.

The stable team (in person or an algorithm) picked some patches from that series but not all. Cherry picking the above mentioned patches fixes the problem.

pelwell, 2024-10-09:
> Thank you for fast bugfix!

The fix was already in the latest build before we were aware of the problem.

> Fix in fresh version of kernel? Or how it looks like?

I'm not sure I understand the question, but the change affects the mcp251xfd module, and is part of a complete 6.6.54 kernel.

pompushko, 2024-10-09:
I mean, how `sudo rpi-update` will fix kernel :D

pelwell, 2024-10-09:
rpi-update installs complete kernel (and firmware) builds. The latest kernel build is 6.6.54, which includes the fix.

pompushko, 2024-10-09:
Oh, now I get it :D
I thought that with new fixes, should be incremented version of kernel...

pelwell, 2024-10-09:
At some point in the future the standard kernel package in "apt" will also pick up the fix, but that will take longer.

pompushko, 2024-10-09:
> At some point in the future the standard kernel package in "apt" will also pick up the fix, but that will take longer.
Is there any way to check, if apt "take" this bugfix?

pelwell, 2024-10-09:
Actually, I think it should be there already. Both the most recent releases to the "stable" branch are 6.6.53, and so will include the fix. Try:
```
sudo apt-update
sudo apt-upgrade
```

popcornmix, 2024-10-09:
> Actually, I think it should be there already. Both the most recent releases to the "stable" branch are 6.6.53, and so will include the fix. Try:

Latest stable kernel is:
```
Linux version 6.6.51+ (dom@buildbot) (arm-linux-gnueabihf-gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #1802 Tue Oct  8 17:23:04 BST 2024
```
Current apt kernel is marginally older:
```
Linux version 6.6.51+rpt-rpi-v8 (serge@raspberrypi.com) (gcc-12 (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT Debian 1:6.6.51-1+rpt2 (2024-10-01)
```
and will be bumped to stable version imminently. However both are 6.6.51.

pompushko, 2024-10-09:
> Actually, I think it should be there already. Both the most recent releases to the "stable" branch are 6.6.53, and so will include the fix. Try:
> 
> ```
> sudo apt-update
> sudo apt-upgrade
> ```

unfortunately, still no any updates

pelwell, 2024-10-09:
> unfortunately, still no any updates

Yes - @popcornmix was explaining that I'd made a mistake.

marckleinebudde, 2024-10-09:
FYI: There is one unsolved problem in one of the patches. In the error case the driver will bail out with `IRQ handler mcp251xfd_handle_tefif() returned -22.`. See following discussion for more details:

https://lore.kernel.org/all/FR3P281MB155216711EFF900AD9791B7ED9692@FR3P281MB1552.DEUP281.PROD.OUTLOOK.COM/
https://lore.kernel.org/all/20241001-mcp251xfd-fix-length-calculation-v1-1-598b46508d61@pengutronix.de/

pelwell, 2024-10-09:
And then there's #6407.

pompushko, 2024-10-09:
Hour ago came this one packages:

```
The following packages will be upgraded:
  linux-headers-6.6.51+rpt-common-rpi linux-headers-6.6.51+rpt-rpi-v7
  linux-headers-6.6.51+rpt-rpi-v7l linux-headers-rpi-v7 linux-headers-rpi-v7l
  linux-image-6.6.51+rpt-rpi-v7 linux-image-6.6.51+rpt-rpi-v7l linux-image-rpi-v7
  linux-image-rpi-v7l linux-kbuild-6.6.51+rpt linux-libc-dev
```

After update, I get:
`Linux canbus 6.6.51+rpt-rpi-v7 #1 SMP Raspbian 1:6.6.51-1+rpt3 (2024-10-08) armv7l GNU/Linux`

But still have the issue :(

pelwell, 2024-10-09:
6.6.54 has missed the cut-off for the next apt kernel release, but we'll try to get it out soon.

pompushko, 2024-10-20:
> 6.6.54 has missed the cut-off for the next apt kernel release, but we'll try to get it out soon.

when it could be? thank you

ragazenta, 2024-10-23:
> > 6.6.54 has missed the cut-off for the next apt kernel release, but we'll try to get it out soon.
> 
> when it could be? thank you

Uhm, how about downgrading to 6.6.28, if you prefer `apt` over `rpi-update`? I got segmentation fault in 6.6.51 on Pi 5. The package name for Pi Zero is a bit different though.

```
sudo apt install linux-image-6.6.28+rpt-rpi-2712 --no-install-recommends
sudo cp /boot/vmlinuz-6.6.28+rpt-rpi-2712 /boot/firmware/kernel_2712.img
```

henzef, 2024-11-14:
> FYI: There is one unsolved problem in one of the patches. In the error case the driver will bail out with `IRQ handler mcp251xfd_handle_tefif() returned -22.`. See following discussion for more [...]

Is there any update on this? Maybe a patch to try out?

marckleinebudde, 2024-11-14:
The original problem ("https://github.com/raspberrypi/linux/issues/6406#issue-2574144693") is solved with:

Please upgrade to v6.6.53 or cherry pick:
51b2a72 ("can: mcp251xfd: properly indent labels")
a780154 ("can: mcp251xfd: move mcp251xfd_timestamp_start()/stop() into mcp251xfd_chip_start/stop()")

The `IRQ handler mcp251xfd_handle_tefif() returned -22.` problem is partly fixed with:
3c1c18551e6a ("can: mcp251xfd: mcp251xfd_get_tef_len(): fix length calculation")

Bit it sometimes still triggers:
| https://lore.kernel.org/all/CAJ7t6HgaeQ3a_OtfszezU=zB-FqiZXqrnATJ3UujNoQJJf7GgA@mail.gmail.com/
I'll look into it tomorrow.

mattie47, 2024-11-19:
Just noting here I've asked on the Pi forums when we might expect to see v6.6.53 in pi-gen images or in apt. 

https://forums.raspberrypi.com/viewtopic.php?t=379698

pompushko, 2024-11-28:
Almost 2 month passed, but no fresh version in apt :D
Is there any update? Thank you

Some of my prod devices was broken permanently with that update :(

6by9, 2024-11-28:
Try looking again. Current kernel image for Bookworm is 6.6.62
```
$ apt-cache policy linux-image-rpi-v8
linux-image-rpi-v8:
  Installed: 1:6.6.62-1+rpt1
  Candidate: 1:6.6.62-1+rpt1
  Version table:
 *** 1:6.6.62-1+rpt1 500
        500 http://archive.raspberrypi.com/debian bookworm/main arm64 Packages
        100 /var/lib/dpkg/status
```
https://archive.raspberrypi.com/debian/pool/main/l/linux/ gives it a file datestamp of 2024-11-28 11:24

apt is only updated relatively infrequently for stability.

ragazenta, 2024-11-29:
> https://archive.raspberrypi.com/debian/pool/main/l/linux/ gives it a file datestamp of 2024-11-28 11:24

I noticed that the new tag has not been pushed to GitHub yet.

popcornmix, 2024-11-29:
> I noticed that the new tag has not been pushed to GitHub yet.

@XECDesign was this something you usually do?

XECDesign, 2024-12-02:
It was when the kernel was a part of the raspberrypi-firmware package. I was adding tags to match the version of the package, but I don't think I added the stable_* tags. Now that it is in its own package which doesn't match the same version format, I don't know what sort of tag would be expected or helpful.

ragazenta, 2024-12-02:
> I don't know what sort of tag would be expected or helpful.

https://downloads.raspberrypi.com/raspios_armhf/release_notes.txt

The 6.6.51 release is 5aeecea9f4a45248bcf564dec924965e066a7bfd, which is tagged as `stable_20241008`. (I wonder why that commit was tagged before being merged to main branch)

At least it should be noted in release notes, the commit id. It might be helpful to backport some driver patches from upstream on current released kernel.

mgrouch, 2024-12-11:
Is this fixed in 6.6.62?
Is it safe to upgrade?

Thanks

mattie47, 2024-12-11:
6.6.62 has been working reliably for me. I have upgraded around 20x RPI 4s
with 2x MCP2518FD each to it.

Thanks,

Matt

marckleinebudde, 2024-12-11:
@pompushko does it make sense to close this issue?

wrclark, 2024-12-11:
does not work properly with 6.6.64-v8+

marckleinebudde, 2024-12-11:
@wrclark Can you explain the problem?

wrclark, 2024-12-12:
> @wrclark Can you explain the problem?

[image]
It runs fine for a couple of minutes then I get this crc error and it refuses to restart. can1 is a slcan usb device that works fine and they are connected with good wire
I reboot and it is seemingly dead
[image]

kernel `6.6.64-v8+ `

```
$ modinfo can
filename:       /lib/modules/6.6.64-v8+/kernel/net/can/can.ko.xz
alias:          net-pf-29
author:         Urs Thuermann <urs.thuermann@volkswagen.de>, Oliver Hartkopp <oliver.hartkopp@volkswagen.de>
license:        Dual BSD/GPL
description:    Controller Area Network PF_CAN core
srcversion:     9FCDE9DC80F2506E3093972
depends:
intree:         Y
name:           can
vermagic:       6.6.64-v8+ SMP preempt mod_unload modversions aarch64
parm:           stats_timer:enable timer for statistics (default:on) (int)
```
```
$ modinfo mcp251xfd
filename:       /lib/modules/6.6.64-v8+/kernel/drivers/net/can/spi/mcp251xfd/mcp251xfd.ko.xz
license:        GPL v2
description:    Microchip MCP251xFD Family CAN controller driver
author:         Marc Kleine-Budde <mkl@pengutronix.de>
srcversion:     B043E0D3F878F0CE694064E
alias:          spi:mcp251xfd
alias:          spi:mcp251863
alias:          spi:mcp2518fd
alias:          spi:mcp2517fd
alias:          of:N*T*Cmicrochip,mcp251xfdC*
alias:          of:N*T*Cmicrochip,mcp251xfd
alias:          of:N*T*Cmicrochip,mcp251863C*
alias:          of:N*T*Cmicrochip,mcp251863
alias:          of:N*T*Cmicrochip,mcp2518fdC*
alias:          of:N*T*Cmicrochip,mcp2518fd
alias:          of:N*T*Cmicrochip,mcp2517fdC*
alias:          of:N*T*Cmicrochip,mcp2517fd
depends:        can-dev
intree:         Y
name:           mcp251xfd
vermagic:       6.6.64-v8+ SMP preempt mod_unload modversions aarch64
```

Then I downgrade to `6.1.77-v8+`
I have the same problem as above with instant bus-off.

```
$ modinfo can
filename:       /lib/modules/6.1.77-v8+/kernel/net/can/can.ko.xz
alias:          net-pf-29
author:         Urs Thuermann <urs.thuermann@volkswagen.de>, Oliver Hartkopp <oliver.hartkopp@volkswagen.de>
license:        Dual BSD/GPL
description:    Controller Area Network PF_CAN core
srcversion:     76F20AE7FC3EED720718288
depends:
intree:         Y
name:           can
vermagic:       6.1.77-v8+ SMP preempt mod_unload modversions aarch64
parm:           stats_timer:enable timer for statistics (default:on) (int)
```
```
$ modinfo mcp251xfd
filename:       /lib/modules/6.1.77-v8+/kernel/drivers/net/can/spi/mcp251xfd/mcp251xfd.ko.xz
license:        GPL v2
description:    Microchip MCP251xFD Family CAN controller driver
author:         Marc Kleine-Budde <mkl@pengutronix.de>
srcversion:     BEAFC3261FECB9E6BF50DDA
alias:          spi:mcp251xfd
alias:          spi:mcp251863
alias:          spi:mcp2518fd
alias:          spi:mcp2517fd
alias:          of:N*T*Cmicrochip,mcp251xfdC*
alias:          of:N*T*Cmicrochip,mcp251xfd
alias:          of:N*T*Cmicrochip,mcp251863C*
alias:          of:N*T*Cmicrochip,mcp251863
alias:          of:N*T*Cmicrochip,mcp2518fdC*
alias:          of:N*T*Cmicrochip,mcp2518fd
alias:          of:N*T*Cmicrochip,mcp2517fdC*
alias:          of:N*T*Cmicrochip,mcp2517fd
depends:        can-dev
intree:         Y
name:           mcp251xfd
vermagic:       6.1.77-v8+ SMP preempt mod_unload modversions aarch64
```

wrclark, 2024-12-12:
I think the wiring was the issue please ignore

pelwell, 2024-12-12:
This seems like a good point to close it.

closed 2024-12-12 by pelwell

woutVDW, 2024-12-14:
Hi, I am encountering the same issue. With my raspberry pi 5 on Ubuntu 24.04 with kernel 6.8.0-1016. I need this version for certain software that is running on this device. Do somebody know when this fix is ported to 6.8.y version? Or is it maybe possible to build the driver from source with this fix?

pelwell, 2024-12-14:
The 6.8 kernel is End Of Life - we won't be making any changes to rpi-6.8.y.

filogold, 2025-02-02:
Hi everyone,

I encountered the same IRQ handler issue. I updated the kernel, which seemed to improve the situation, but in reality, it only managed the problem better by postponing the fault.

My setup consists of a Raspberry Pi 5 with two CAN HAT 2CH FD modules, each using the MCP251XFD IC.

To ensure the problem is related to the transmitter and not the receiver, I tested two different scenarios:

1. Using two channels of the Raspberry Pi to transmit and the remaining two channels to receive (all configured with the same parameters: bitrate, data rate, and sampling time).

2. Using two channels of the Raspberry Pi to transmit and two channels of a Vector VN1640 (with the 1057Gcap installed) to receive. Also in this case, I carefully verified the CAN configuration.

In both cases the termination are correctly verified, providing 60 ohms in the buses.

I am aware that I am stressing the module, as I am sending multiple periodic messages (128 periodic messages per CAN FD, generating a 40% bus load). The messages require some processing to calculate the internal CRC and MC in the payload, which is correctly handled by the Python code.

Based on the information provided, the issue appears to be related to overrun packets. I suspect this is the cause of the failure.

```
ifconfig
can0: flags=193<UP,RUNNING,NOARP> mtu 72
unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00 txqueuelen 65536
(UNSPEC)
RX packets 6455202 bytes 68902927 (65.7 MiB)
RX errors 0 dropped 0 overruns 0 frame 0
TX packets 6455202 bytes 68902927 (65.7 MiB)
TX errors 0 dropped 0 overruns 8471 carrier 0 collisions 0
device interrupt 186
can1: flags=193<UP,RUNNING,NOARP> mtu 72
unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00 txqueuelen 65536
(UNSPEC)
RX packets 13736158 bytes 146632299 (139.8 MiB)
RX errors 0 dropped 0 overruns 0 frame 0
TX packets 13736158 bytes 146632299 (139.8 MiB)
TX errors 0 dropped 0 overruns 11052 carrier 0 collisions 0
device interrupt 187
can2: flags=193<UP,RUNNING,NOARP> mtu 72
unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00 txqueuelen 65536
(UNSPEC)
RX packets 13736162 bytes 146632347 (139.8 MiB)
RX errors 0 dropped 0 overruns 3 frame 0
TX packets 0 bytes 0 (0.0 B)
TX errors 0 dropped 0 overruns 0 carrier 0 collisions 0
device interrupt 188
can3: flags=193<UP,RUNNING,NOARP> mtu 72
unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00 txqueuelen 65536
(UNSPEC)
RX packets 6455206 bytes 68902975 (65.7 MiB)
RX errors 0 dropped 0 overruns 1 frame 0
TX packets 0 bytes 0 (0.0 B)
TX errors 0 dropped 0 overruns 0 carrier 0 collisions 0
device interrupt 189
```

```
dmesg | grep can
[ 3.165808] mcp251xfd spi0.1 can0: MCP2518FD rev0.0 (-RX_INT -PLL
-MAB_NO_WARN +CRC_REG +CRC_RX +CRC_TX +ECC -HD o:40.00MHz c:40.00MHz
m:20.00MHz rs:17.00MHz es:16.66MHz rf:17.00MHz ef:16.66MHz) successfully initialized.
[ 3.170311] mcp251xfd spi0.0 can1: MCP2518FD rev0.0 (-RX_INT -PLL -MAB_NO_WARN
+CRC_REG +CRC_RX +CRC_TX +ECC -HD o:40.00MHz c:40.00MHz m:20.00MHz
rs:17.00MHz es:16.66MHz rf:17.00MHz ef:16.66MHz) successfully initialized.
[ 3.179808] mcp251xfd spi1.1 can2: MCP2518FD rev0.0 (-RX_INT -PLL
-MAB_NO_WARN +CRC_REG +CRC_RX +CRC_TX +ECC -HD o:40.00MHz c:40.00MHz
m:20.00MHz rs:17.00MHz es:16.66MHz rf:17.00MHz ef:16.66MHz) successfully initialized.
[ 3.186278] mcp251xfd spi1.0 can3: MCP2518FD rev0.0 (-RX_INT -PLL
-MAB_NO_WARN +CRC_REG +CRC_RX +CRC_TX +ECC -HD o:40.00MHz c:40.00MHz
m:20.00MHz rs:17.00MHz es:16.66MHz rf:17.00MHz ef:16.66MHz) successfully initialized.
[ 112.788097] can: controller area network core
[ 112.795770] can: raw protocol
[ 2339.305706] mcp251xfd spi0.1 can0: IRQ handler mcp251xfd_handle_tefif() returned -22.
[ 2339.305714] mcp251xfd spi0.1 can0: IRQ handler returned -22 (intf=0x3f1a0014).
[ 4809.135009] mcp251xfd spi0.0 can1: IRQ handler mcp251xfd_handle_tefif() returned -22.
[ 4809.135017] mcp251xfd spi0.0 can1: IRQ handler returned -22 (intf=0x3f1a0014).
```

```
uname -a
Linux raspberrypi 6.6.74+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.74-1+rpt1
(2025-01-27) aarch64 GNU/Linux
```

```
modinfo can
filename: /lib/modules/6.6.74+rpt-rpi-2712/kernel/net/can/can.ko.xz
alias: net-pf-29
author: Urs Thuermann <urs.thuermann@volkswagen.de>, Oliver Hartkopp
<oliver.hartkopp@volkswagen.de>
license: Dual BSD/GPL
description: Controller Area Network PF_CAN core
srcversion: 37BF79470A254916FD8C595
depends:
intree: Y
name: can
vermagic: 6.6.74+rpt-rpi-2712 SMP preempt mod_unload modversions aarch64
parm: stats_timer:enable timer for statistics (default:on) (int)
```

```
modinfo mcp251xfd
filename:
/lib/modules/6.6.74+rpt-rpi-2712/kernel/drivers/net/can/spi/mcp251xfd/mcp251xfd.ko.xz
license: GPL v2
description: Microchip MCP251xFD Family CAN controller driver
author: Marc Kleine-Budde <mkl@pengutronix.de>
srcversion: 672D5B0159EB5AA465D4A79
alias: spi:mcp251xfd
alias: spi:mcp251863
alias: spi:mcp2518fd
alias: spi:mcp2517fd
alias: of:N*T*Cmicrochip,mcp251xfdC*
alias: of:N*T*Cmicrochip,mcp251xfd
alias: of:N*T*Cmicrochip,mcp251863C*
alias: of:N*T*Cmicrochip,mcp251863
alias: of:N*T*Cmicrochip,mcp2518fdC*
alias: of:N*T*Cmicrochip,mcp2518fd
alias: of:N*T*Cmicrochip,mcp2517fdC*
alias: of:N*T*Cmicrochip,mcp2517fd
depends: can-dev
intree: Y
name: mcp251xfd
vermagic: 6.6.74+rpt-rpi-2712 SMP preempt mod_unload modversions aarch64
```

The information I provided above is essentially the same as in the second case, where I use the Vector hardware as the receiver. One observation I have made is that after a bus failure, the overruns seem to stop or decrease. However, after a certain period (several hours), at some point, the second bus also fails.

Thanks in advance

marckleinebudde, 2025-02-02:
Hey @filogold,

this is unrelated to the original issue, can you please open a new issue and @-mention me!

Thanks!