raspberrypi/linux #6413: Kernel Oops when communicating with an USB device

mdrodrigo, 2024-10-11:
### Describe the bug

We have a device that uses CDC Port for communicating and when we send data to the device, we receive this set of kernel Oops.

### Steps to reproduce the behaviour

Any communication with this specific device, triggers the Oops. 

### Device (s)

Raspberry Pi 3 Mod. B+, Raspberry Pi 4 Mod. B

### System

Raspberry Pi reference 2024-07-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 0b115f302a8f1e5bd3523614d7f45b9d447434c7, stage4

Oct 10 2024 18:27:31
Copyright (c) 2012 Broadcom
version 0e63dd5ec215290186b4f3a41496b4425155878e (clean) (release) (start)

Linux raspberrypi 6.6.56-v7+ #1804 SMP Thu Oct 10 18:20:27 BST 2024 armv7l GNU/Linux

### Logs

```
[   63.311793] 8<--- cut here ---
[   63.314920] Unable to handle kernel paging request at virtual address 0a0d3458 when read
[   63.323146] [0a0d3458] *pgd=00000000
[   63.326794] Internal error: Oops: 5 [#1] SMP ARM
[   63.331489] Modules linked in: rfcomm snd_seq_dummy snd_hrtimer snd_seq snd_seq_device cmac algif_hash aes_arm_bs crypto_simd cryptd algi6
[   63.401495] CPU: 3 PID: 1496 Comm: python3 Tainted: G         C         6.6.56-v7+ #1804
[   63.409720] Hardware name: BCM2835
[   63.413165] PC is at __kmem_cache_alloc_node+0x110/0x48c
[   63.418565] LR is at 0xa54d
[   63.421395] pc : [<8036041c>]    lr : [<0000a54d>]    psr: a0000093
[   63.427747] sp : bbd3dce0  ip : bbd3dce0  fp : aec93580
[   63.433042] r10: 00000000  r9 : 00000038  r8 : 811057c4
[   63.438337] r7 : 00000920  r6 : 0a0d3438  r5 : 00000000  r4 : 81401100
[   63.444957] r3 : 00000020  r2 : 39a6b000  r1 : bab06378  r0 : 0000a54c
[   63.451603] Flags: NzCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment user
[   63.458978] Control: 10c5383d  Table: 0476406a  DAC: 00000055
[   63.464811] Register r0 information: non-paged memory
[   63.469937] Register r1 information: non-slab/vmalloc memory
[   63.475675] Register r2 information: non-paged memory
[   63.480796] Register r3 information: non-paged memory
[   63.485916] Register r4 information: slab kmem_cache start 81401100 pointer offset 0 size 124
[   63.494576] Register r5 information: NULL pointer
[   63.499343] Register r6 information: non-paged memory
[   63.504463] Register r7 information: non-paged memory
[   63.509583] Register r8 information: non-slab/vmalloc memory
[   63.515322] Register r9 information: non-paged memory
[   63.520441] Register r10 information: NULL pointer
[   63.525297] Register r11 information: slab kmalloc-128 start aec93580 pointer offset 0 size 128
[   63.534131] Register r12 information: 2-page vmalloc region starting at 0xbbd3c000 allocated at kernel_clone+0x9c/0x374
[   63.545080] Process python3 (pid: 1496, stack limit = 0x818146ce)
[   63.551260] Stack: (0xbbd3dce0 to 0xbbd3e000)
[   63.555678] dce0: 00000000 7f2f47d4 0000a54d ffffffff bab03480 00000038 8087e804 81401100
[   63.563972] dd00: 00000001 00000920 81dba400 00000000 aec93580 80317bc0 8087e804 bab03480
[   63.572264] dd20: 00000000 80113a8c 81743e10 00000820 00000000 00000038 00000001 81ca0000
[   63.580557] dd40: 81dba400 00000000 aec93580 8087e804 00000000 bbd3dda0 39a6b000 00000820
[   63.588850] dd60: 00000002 83236a00 84f6d000 8087eeec 81ca0000 00000000 82b21200 00000200
[   63.597144] dd80: 00000002 81743e10 00000000 7f2f47d4 00000002 83236a00 00000000 81ca0000
[   63.605437] dda0: 84f6d000 00000820 83236a0c 84f6f8c4 00000000 8084efcc 81525c00 81554e00
[   63.613730] ddc0: 81525c05 80b26c40 835d6a80 80b26c40 835d6a80 80b26c40 81401300 8085f2b4
[   63.622022] dde0: 00000000 80360350 00003817 ffffffff 00000666 ffffffff 80850378 00000200
[   63.630317] de00: 8085f2b4 81401300 01954d58 00002cc0 812ddfc0 8085f464 00000000 80317bc0
[   63.638610] de20: 8085f2b4 00000002 00000820 7f2f47d4 00000001 83236c80 83236c80 98ca7000
[   63.646903] de40: 83236c90 00000000 812ddfc0 84f6f8c4 00000000 8085f464 bbd3de70 00000004
[   63.655196] de60: 00000001 80173384 aecfadc0 00000000 00000000 00000001 aec93580 00000000
[... 297 lines cut ...]
[  353.209108] CPU1: stopping
[  353.209113] CPU: 1 PID: 0 Comm: swapper/1 Tainted: G      D  C         6.6.56-v7+ #1804
[  353.209119] Hardware name: BCM2835
[  353.209123]  unwind_backtrace from show_stack+0x18/0x1c
[  353.209136]  show_stack from dump_stack_lvl+0x50/0x68
[  353.209146]  dump_stack_lvl from do_handle_IPI+0x1a8/0x1e4
[  353.209158]  do_handle_IPI from ipi_handler+0x20/0x28
[  353.209169]  ipi_handler from handle_percpu_devid_irq+0x84/0x1c0
[  353.209179]  handle_percpu_devid_irq from generic_handle_domain_irq+0x30/0x40
[  353.209191]  generic_handle_domain_irq from bcm2836_arm_irqchip_handle_ipi+0x94/0xb4
[  353.209203]  bcm2836_arm_irqchip_handle_ipi from generic_handle_domain_irq+0x30/0x40
[  353.209216]  generic_handle_domain_irq from generic_handle_arch_irq+0x34/0x44
[  353.209229]  generic_handle_arch_irq from call_with_stack+0x18/0x20
[  353.209240]  call_with_stack from __irq_svc+0x84/0x94
[  353.209249] Exception stack(0xbb871f68 to 0xbb871fb0)
[  353.209257] 1f60:                   00000003 00000001 00000001 60000093 815ab540 8110501c
[  353.209263] 1f80: 815ab540 81105078 81256d50 80e28bd4 00000000 00000000 00000007 bb871fb8
[  353.209268] 1fa0: 80b24908 80b24d60 60000013 ffffffff
[  353.209272]  __irq_svc from default_idle_call+0x38/0xa0
[  353.209283]  default_idle_call from do_idle+0xbc/0x11c
[  353.209296]  do_idle from cpu_startup_entry+0x30/0x34
[  353.209310]  cpu_startup_entry from secondary_start_kernel+0x11c/0x124
[  353.209322]  secondary_start_kernel from 0x101280
[  353.617219] ---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b ]---
```

pelwell, 2024-10-14:
It looks as though something is corrupting kernel memory - some memory locations that are being interpreted as addresses actually seem to hold text strings. User space code should not be able to write to kernel memory in this way, so I suspect you have a bad driver.

When you say CDC Port, are you referring to using the dwc2 driver to bring up the USB interface on the power port as a device running a virtual serial interface?

P33M, 2024-10-14:
For more useful diagnostic output, can you boot with `slub_debug=FPUZ` in /boot/firmware/config.txt and provoke a crash?

mdrodrigo, 2024-10-14:
Hi @P33M 
> For more useful diagnostic output, can you boot with `slub_debug=FPUZ` in /boot/firmware/config.txt and provoke a crash?

The output with `slub_debug=FPUZ `

 ```
[   88.504658] 8<--- cut here ---
[   88.507791] Unable to handle kernel paging request at virtual address 0a0d3458 when read
[   88.516006] [0a0d3458] *pgd=00000000
[   88.519655] Internal error: Oops: 5 [#1] SMP ARM
[   88.524345] Modules linked in: rfcomm snd_seq_dummy snd_hrtimer snd_seq snd_seq_device cmac algif_hash aes_arm_bs crypto_simd cryptd algif_skcipher af_alg bnep vc4 brcmfmac_wcc snd_soc_hdmi_codec drm_display_helper cec hci_uart drm_dma_helper btbcm drm_kms_helper brcmfmac bluetooth brcmutil snd_soc_core bcm2835_isp(C) bcm2835_v4l2(C) cfg80211 binfmt_misc bcm2835_codec(C) cdc_acm v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_vmalloc videobuf2_dma_contig raspberrypi_hwmon snd_compress videobuf2_memops snd_pcm_dmaengine videobuf2_v4l2 videodev i2c_bcm2835 snd_bcm2835(C) snd_pcm ecdh_generic ecc rfkill videobuf2_common snd_timer mc snd raspberrypi_gpiomem vc_sm_cma(C) fixed uio_pdrv_genirq uio i2c_dev drm fuse drm_panel_orientation_quirks dm_mod backlight ip_tables x_tables ipv6
[   88.594261] CPU: 2 PID: 1498 Comm: python3 Tainted: G         C         6.6.56-v7+ #1804
[   88.602487] Hardware name: BCM2835
[   88.605940] PC is at __kmem_cache_alloc_node+0x110/0x48c
[   88.611350] LR is at 0xb5c7
[   88.614186] pc : [<8036041c>]    lr : [<0000b5c7>]    psr: a0000093
[   88.620544] sp : bbf41cd0  ip : bbf41cd0  fp : aece8300
[   88.625846] r10: 00000000  r9 : 0000002c  r8 : 811057c4
[   88.631148] r7 : 00000920  r6 : 0a0d3438  r5 : 00000000  r4 : 81401100
[   88.637774] r3 : 00000020  r2 : 39a59000  r1 : baaf4378  r0 : 0000b5c6
[   88.644400] Flags: NzCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment user
[   88.651750] Control: 10c5383d  Table: 0285006a  DAC: 00000055
[   88.657613] Register r0 information: non-paged memory
[   88.662788] Register r1 information: non-slab/vmalloc memory
[   88.668558] Register r2 information: non-paged memory
[   88.673687] Register r3 information: non-paged memory
[   88.678816] Register r4 information: slab kmem_cache start 81401100 pointer offset 0 size 124
[   88.687493] Register r5 information: NULL pointer
[   88.692270] Register r6 information: non-paged memory
[   88.697397] Register r7 information: non-paged memory
[   88.702524] Register r8 information: non-slab/vmalloc memory
[   88.708271] Register r9 information: non-paged memory
[   88.713399] Register r10 information: NULL pointer
[   88.718263] Register r11 information: slab kmalloc-128 start aece8300 pointer offset 0 size 128
[   88.727114] Register r12 information: 2-page vmalloc region starting at 0xbbf40000 allocated at kernel_clone+0x9c/0x374
[   88.738077] Process python3 (pid: 1498, stack limit = 0xd564c68c)
[   88.744265] Stack: (0xbbf41cd0 to 0xbbf42000)
[   88.748689] 1cc0:                                     84400240 00225510 0000b5c7 ffffffff
[   88.756991] 1ce0: 00000000 0000002c 80883e18 81401100 aece8320 00000920 816cd800 00000000
[   88.765294] 1d00: aece8300 80317bc0 80883e18 00000000 aece8300 80317bc0 8087e804 816cd800
[   88.773596] 1d20: 84b6e140 00000001 aece8320 81749e00 816cd800 00000000 aece8300 80883e18
[   88.781898] 1d40: 816cd800 84b6e140 00000001 8087d3c8 00000000 bbf41da0 00000000 00000002
[   88.790200] 1d60: 828b5700 84b6e140 81749e00 8087f014 00000200 00000000 00000000 00000001
[   88.798502] 1d80: 00000000 81743e10 60000093 9fff39ec 00000002 828b5700 00000000 81749e00
[   88.806804] 1da0: 84fa8400 00000820 828b570c 821bc8c4 00000000 8084efcc 81525b00 81554c00
[   88.815106] 1dc0: 81525b05 80b26c40 83450000 80b26c40 83450000 80b26c40 81401300 8085f2b4
[   88.823409] 1de0: 00000000 80360350 00004706 ffffffff 00007348 ffffffff 80850378 00000200
[   88.831711] 1e00: 8085f2b4 81401300 01c93b58 00002cc0 812ddfc0 8085f464 00000000 80317bc0
[   88.840012] 1e20: 8085f2b4 00000002 00000820 9fff39ec 00000001 84947780 84947780 84b6e440
[   88.848315] 1e40: 84947790 00000000 812ddfc0 821bc8c4 00000000 8085f464 bbf41e70 00000004
[   88.856617] 1e60: 00000001 80173384 aeca3a20 00000000 00000000 00000001 aece8300 00000000
[   88.864918] 1e80: 00000001 00000000 828b5700 00000000 84947790 00000000 00000000 825f8180
[   88.873219] 1ea0: baae9a80 801b728c 825f8180 baae9a40 baae9a80 801b7eac 00000000 00000000
[   88.881522] 1ec0: 00000000 20000093 00000000 801b7350 00008103 00000000 00000000 01c59fc8
[   88.889823] 1ee0: 00000200 00000000 00000000 00000000 00000000 00000000 01c59120 9fff39ec
[   88.898127] 1f00: 00000000 802c550a 00000000 8454af01 01c93b58 00000009 8454af00 aeca3a20
[   88.906429] 1f20: 01c93b58 8038fe70 00000000 00000008 00000000 00000001 00000161 80100298
[   88.914731] 1f40: 83450000 00000161 00000000 803d79b8 00000000 00000000 00000000 ae695301
[   88.923032] 1f60: 00000058 00000000 23d47858 80100298 00000000 00000000 00000000 9fff39ec
[   88.931333] 1f80: 00000000 00000001 01c59120 00000001 00000036 80100298 83450000 00000036
[   88.939637] 1fa0: 01c93b58 80100040 00000001 01c59120 00000009 802c550a 01c93b58 7ec6bbb0
[   88.947940] 1fc0: 00000001 01c59120 00000001 00000036 00004000 00000000 00000001 01c93b58
[   88.956243] 1fe0: 00000000 7ec6bb98 76e1c000 76d9f0e4 60000010 00000009 00000000 00000000
[   88.964548]  __kmem_cache_alloc_node from __kmalloc+0x4c/0x180
[   88.970487]  __kmalloc from dwc_otg_hcd_qtd_create+0x24/0x88
[... cut ...]
[  119.731348]  cpu_startup_entry from secondary_start_kernel+0x11c/0x124
[  119.737976]  secondary_start_kernel from 0x101280
[  125.043556] raspberrypi-exp-gpio soc:firmware:expgpio: Failed to get GPIO 4 state (-110 84)
[  126.083552] i2c-bcm2835 3f805000.i2c: i2c transfer timed out
[  127.123549] i2c-bcm2835 3f805000.i2c: i2c transfer timed out
[  128.163550] i2c-bcm2835 3f805000.i2c: i2c transfer timed out
[  129.193552] i2c-bcm2835 3f805000.i2c: i2c transfer timed out
[  130.243549] i2c-bcm2835 3f805000.i2c: i2c transfer timed out
```

mdrodrigo, 2024-10-14:
Hi @pelwell 

> It looks as though something is corrupting kernel memory - some memory locations that are being interpreted as addresses actually seem to hold text strings. User space code should not be able to [...]

The raspberrypi is in Host mode, as the other device is providing a device (using a g_serial).