raspberrypi/linux #6414: DSI display text flickers on latest kernel

mgineer85, 2024-10-14:
### Describe the bug

I have a [waveshare 4.3 dsi display](https://www.waveshare.com/wiki/4.3inch_DSI_LCD) displaying text weird after booting to the console (lite system).

### Steps to reproduce the behaviour

**Make it fail:**
Just start the system, waiting for the console.

- First the display is OK, which makes me assume, the display is OK.
- then after a short black flicker quite at the end of the boot, the text gets strange flickerish:

[image]

**Make it work:**
Downgrade to 6.6.16 using "rpi-update next" (currently next seems outdated, I just randomly pick an older kernel for testing).

[image]

### Device (s)

Raspberry Pi 3 Mod. B

### System

Which kernel version (uname -a)?

```
Linux wigglecam-main 6.6.56-v8+ #1804 SMP PREEMPT Thu Oct 10 18:23:14 BST 2024 aarch64 GNU/Linux
```

```
Raspberry Pi reference 2024-07-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 48efb5fc5485fafdc9de8ad481eb5c09e1182656, stage2
```

Which firmware version (vcgencmd version)?
```
Oct 10 2024 18:27:31
Copyright (c) 2012 Broadcom
version 0e63dd5ec215290186b4f3a41496b4425155878e (clean) (release) (start)
```

### Logs

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 6.6.56-v8+ (dom@buildbot) (aarch64-linux-gnu-gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #1804 SMP PREEMPT Thu Oct 10 18:23:14 BST 2024
[    0.000000] KASLR enabled
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001e400000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x000000001e400000..0x000000002e3fffff (262144 KiB) map reusable linux,cma
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] On node 0, zone DMA: 19456 pages in unavailable ranges
[    0.000000] percpu: Embedded 30 pages/cpu s85672 r8192 d29016 u122880
[    0.000000] pcpu-alloc: s85672 r8192 d29016 u122880 alloc=30*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[... 414 lines cut ...]
[   12.803237] [drm] Initialized vc4 0.0.0 20140616 for soc:gpu on minor 0
[   12.839285] imx708 10-001a: camera module ID 0x0301
[   13.235216] Console: switching to colour frame buffer device 100x30
[   13.246106] vc4-drm soc:gpu: [drm] fb0: vc4drmfb frame buffer device
[   13.279772] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   13.279814] Bluetooth: BNEP filters: protocol multicast
[   13.279841] Bluetooth: BNEP socket layer initialized
[   13.347587] input: 10-0038 generic ft5x06 (79) as /devices/platform/soc/3f205000.i2c/i2c-11/i2c-10/10-0038/input/input2
[   13.357716] Bluetooth: MGMT ver 1.22
[   13.375949] NET: Registered PF_ALG protocol family
[   13.941167] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[   13.943787] smsc95xx 1-1.1:1.0 eth0: Link is Down
[   13.970087] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   35.438848] systemd[804]: memfd_create() called without MFD_EXEC or MFD_NOEXEC_SEAL set

```

### Additional context

- PI 3b (tested 2 boards)
- system up to date (apt update and apt full-upgrade)
- Bookworm lite 64bit 
- The system is fresh installed, only change was to add `dtoverlay=vc4-kms-dsi-7inch`

I tried some ideas I picked up in forums, but nothing worked except the kernel downgrade.

mgineer85, 2024-10-14:
Hey, I tested when it starts not working any more, see as follows. Maybe it's helpful?

6.6.26 -> works
...
6.6.29 -> works
6.6.30 -> not working
...
6.6.56 -> not working

popcornmix, 2024-10-14:
Based on the identified version (I assume you are testing these versions using rpi-update with commits from master branch of rpi-firmware?) it looks like this PR is most likely:
https://github.com/raspberrypi/linux/pull/6094

mgineer85, 2024-10-14:
Yes, I did it like that.
I will check later again if that commit is causing it.

6by9, 2024-10-14:
Most likely it is https://github.com/raspberrypi/linux/pull/6094/commits/98b60c70cc0d36dcb6680b12cd63f348a7018c1d

This is the problem with 3rd party displays pretending to be the Pi display - they aren't necessarily 100% compatible, so improvements for the Pi panel can break support for them.

We won't be reverting that patch as it is an improvement for the Pi panel. The Waveshare panel should get a dedicated overlay, and can lose the TC358743 configuration as it isn't using it (it has an ICN6211 configured by their MCU).
I think we may have one of those Waveshare panels in the office, however I'm not there at present. Possibly later in the week.

mgineer85, 2024-10-14:
Ok, I am fine with that. If I can do additional testing, just let me know. Will test later today for that specific commit.

You plan to support the device in native software?otherwise I would return the device and try to find a different panel that suits my needs.
I know its probably waveshare to take that job, so I can understand if you do not support from your side.

thx for your quick additional information!

6by9, 2024-10-14:
> You plan to support the device in native software?

It's a kernel update, so it'll be released through the normal channels - initially rpi-update, and eventually apt.

It will need config.txt modifications to remove `display_auto_detect=1` and add `dtoverlay=vc4-dsi-waveshare-cloned-panel` (or whatever the overlay gets called. The device still looks like a Pi panel over I2C, so auto_detect would treat it as such.

mgineer85, 2024-10-14:
OK, unfortunately I cannot switch to the older pulls for testing, maybe they are not available any more?

Great to hear, that the device will still be included in the default kernel - changing the configuration is ok to me.

When there is something I can test, I am happy to do so. Can test also on Pi4 or Pi5.

6by9, 2024-10-14:
#6417 is a stab at the required overlay.

Once CI has run (takes about 40 mins) you should be able to use `sudo rpi-update pulls/6417` to install the build.
`dtoverlay=vc4-kms-dsi-waveshare-800x480` is then the required line to load it.

mgineer85, 2024-10-14:
I am sorry, the display is still flickering as before.

```
Linux wigglecam-main 6.6.56-v8+ #1 SMP PREEMPT Mon Oct 14 18:39:27 UTC 2024 aarc 64

```

```
pi@wigglecam-main:~ $ kmsprint
Connector 0 (32) HDMI-A-1 (disconnected)
  Encoder 0 (31) TMDS
Connector 1 (42) DSI-1 (connected)
  Encoder 1 (41) DSI
    Crtc 2 (84) 800x480@59.94 29.729 800/24/72/96/- 480/3/10/7/+ 60 (59.94) U
      Plane 2 (73) fb-id: 317 (crtcs: 2) 0,0 800x480 -> 0,0 800x480 (XR24 AR24 A                                                                                                 B24 XB24 RG16 BG16 AR15 XR15 RG24 BG24 YU16 YV16 YU24 YV24 YU12 YV12 NV12 NV21 N                                                                                                 V16 NV61 RGB8 BGR8 XR12 AR12 XB12 AB12 BX12 BA12 RX12 RA12)
        FB 317 800x480 RG16
```

```
pi@wigglecam-main:~ $ dmesg | grep dsi
[    0.059493] platform 3f101000.cprman: Fixed dependency cycle(s) with /soc/dsi@7e700000
[    0.062067] platform 3f101000.cprman: Fixed dependency cycle(s) with /soc/dsi@7e700000
[    0.062198] platform 3f700000.dsi: Fixed dependency cycle(s) with /soc/dsi@7e700000/panel-dsi-generic@0
[    0.062257] platform 3f700000.dsi: Fixed dependency cycle(s) with /soc/cprman@7e101000
[   11.087001] vc4_dsi 3f700000.dsi: Fixed dependency cycle(s) with /soc/dsi@7e700000/panel-dsi-generic@0
[   11.087191] mipi-dsi 3f700000.dsi.0: Fixed dependency cycle(s) with /soc/dsi@7e700000
[   12.454933] vc4-drm soc:gpu: bound 3f700000.dsi (ops vc4_dsi_ops [vc4])
pi@wigglecam-main:~ $ dmesg | grep pane
[    0.062198] platform 3f700000.dsi: Fixed dependency cycle(s) with /soc/dsi@7e700000/panel-dsi-generic@0
[   11.087001] vc4_dsi 3f700000.dsi: Fixed dependency cycle(s) with /soc/dsi@7e700000/panel-dsi-generic@0
pi@wigglecam-main:~ $ dmesg | grep i2c
[    0.065315] platform 3f801000.csi: Fixed dependency cycle(s) with /soc/i2c0mux/i2c@1/imx708@1a
[   10.169361] i2c i2c-11: Added multiplexed i2c bus 0
[   10.257743] i2c i2c-11: Added multiplexed i2c bus 10
[   12.964596] input: 10-0038 generic ft5x06 (79) as /devices/platform/soc/3f205000.i2c/i2c-11/i2c-10/10-0038/input/input2
pi@wigglecam-main:~ $ dmesg | grep vc4
[   11.087001] vc4_dsi 3f700000.dsi: Fixed dependency cycle(s) with /soc/dsi@7e700000/panel-dsi-generic@0
[   12.439243] vc4-drm soc:gpu: bound 3f400000.hvs (ops vc4_hvs_ops [vc4])
[   12.443001] rc rc0: vc4-hdmi as /devices/platform/soc/3f902000.hdmi/rc/rc0
[   12.443183] input: vc4-hdmi as /devices/platform/soc/3f902000.hdmi/rc/rc0/input0
[   12.450672] input: vc4-hdmi HDMI Jack as /devices/platform/soc/3f902000.hdmi/sound/card0/input1
[   12.451337] vc4-drm soc:gpu: bound 3f902000.hdmi (ops vc4_hdmi_ops [vc4])
[   12.454933] vc4-drm soc:gpu: bound 3f700000.dsi (ops vc4_dsi_ops [vc4])
[   12.455348] vc4-drm soc:gpu: bound 3f004000.txp (ops vc4_txp_ops [vc4])
[   12.455630] vc4-drm soc:gpu: bound 3f206000.pixelvalve (ops vc4_crtc_ops [vc4])
[   12.455807] vc4-drm soc:gpu: bound 3f207000.pixelvalve (ops vc4_crtc_ops [vc4])
[   12.455985] vc4-drm soc:gpu: bound 3f807000.pixelvalve (ops vc4_crtc_ops [vc4])
[   12.456103] vc4-drm soc:gpu: bound 3fc00000.v3d (ops vc4_v3d_ops [vc4])
[   12.464594] [drm] Initialized vc4 0.0.0 20140616 for soc:gpu on minor 0
[   12.601873] vc4-drm soc:gpu: [drm] fb0: vc4drmfb frame buffer device

```

```
# Additional overlays and parameters are documented
# /boot/firmware/overlays/README

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=0

# Automatically load initramfs files, if found
auto_initramfs=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Don't have the firmware create an initial video= setting in cmdline.txt.
# Use the kernel's default instead.
disable_fw_kms_setup=1

# Run in 64-bit mode
arm_64bit=1

# Disable compensation for displays with overscan
disable_overscan=1

# Run as fast as firmware / board allows
arm_boost=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[cm5]
dtoverlay=dwc2,dr_mode=host

[all]
#dtoverlay=vc4-kms-dsi-7inch
dtoverlay=vc4-kms-dsi-waveshare-800x480
dtoverlay=imx708,rotation=0
```

6by9, 2024-10-15:
```
    Crtc 2 (84) 800x480@59.94 29.729 800/24/72/96/- 480/3/10/7/+ 60 (59.94) U
```
Something squiffy there as I get 
```
    Crtc 2 (90) 840x480@57.46 27.777 840/59/2/45/? 480/7/2/22/? 57 (57.46) P|D 
```
And I do now see the typo of making the display 840x480, not 800x480.
Fix that (PR now updated) and I get
```
    Crtc 2 (90) 800x480@60.00 27.777 800/59/2/45/? 480/7/2/22/? 60 (60.00) P|D 
```
which is what I expect.

mgineer85, 2024-10-15:
I updated to the latest pull build now and it works.

With that update I get now:
```
pi@wigglecam-main:~ $ kmsprint
Connector 0 (32) HDMI-A-1 (disconnected)
  Encoder 0 (31) TMDS
Connector 1 (42) DSI-1 (connected)
  Encoder 1 (41) DSI
    Crtc 2 (84) 800x480@60.00 27.777 800/59/2/45/? 480/7/2/22/? 60 (60.00) P|D
      Plane 2 (73) fb-id: 317 (crtcs: 2) 0,0 800x480 -> 0,0 800x480 (XR24 AR24 AB24 XB24 RG16 BG16 AR15 XR15 RG24 BG24 YU16 YV16 YU24 YV24 YU12 YV12 NV12 NV21 NV16 NV61 RGB8 BGR8 XR12 AR12 XB12 AB12 BX12 BA12 RX12 RA12)
        FB 317 800x480 RG16
```

Before I also had added 
```
video=DSI-1:800x480M@60,rotate=180
```
to the cmdline.txt. With that the display is still broken so I removed it to ensure it's working. But I need to rotate the screen unfortunately. Is it possible?

6by9, 2024-10-15:
```
video=DSI-1:800x480M@60,rotate=180
```
Should just end up setting the rotation property on the primary plane to rot-180 (0x04), and shouldn't change the mode at all. What does `kmsprint` report as the mode if you add that?

mgineer85, 2024-10-15:
Without:

```

pi@wigglecam-main:~ $ kmsprint
Connector 0 (32) HDMI-A-1 (disconnected)
  Encoder 0 (31) TMDS
Connector 1 (42) DSI-1 (connected)
  Encoder 1 (41) DSI
    Crtc 2 (84) 800x480@60.00 27.777 800/59/2/45/? 480/7/2/22/? 60 (60.00) P|D
      Plane 2 (73) fb-id: 317 (crtcs: 2) 0,0 800x480 -> 0,0 800x480 (XR24 AR24 AB24 XB24 RG16 BG16 AR15 XR15 RG24 BG24 YU16 YV16 YU24 YV24 YU12 YV12 NV12 NV21 NV16 NV61 RGB8 BGR8 XR12 AR12 XB12 AB12 BX12 BA12 RX12 RA12)
        FB 317 800x480 RG16

```

With the video= prepended:
```
pi@wigglecam-main:~ $ sudo vi /boot/firmware/cmdline.txt
video=DSI-1:800x480M@60,rotate=180 console=tty1 root=PARTUUID=0dfd3080-02 rootfstype=ext4 fsck.repair=yes rootwait cfg80211.ieee80211_regdom=DE
```

reboot

```
pi@wigglecam-main:~ $ kmsprint
Connector 0 (32) HDMI-A-1 (disconnected)
  Encoder 0 (31) TMDS
Connector 1 (42) DSI-1 (connected)
  Encoder 1 (41) DSI
    Crtc 2 (84) 800x480@60.00 27.777 800/59/2/45/? 480/7/2/22/? 60 (60.00) P|U|D
      Plane 2 (73) fb-id: 317 (crtcs: 2) 0,0 800x480 -> 0,0 800x480 (XR24 AR24 AB24 XB24 RG16 BG16 AR15 XR15 RG24 BG24 YU16 YV16 YU24 YV24 YU12 YV12 NV12 NV21 NV16 NV61 RGB8 BGR8 XR12 AR12 XB12 AB12 BX12 BA12 RX12 RA12)
        FB 317 800x480 RG16
```

Now it looks still good even with rotated display. ✨✨✨
Maybe it failed because of the 840 / 800?

pelwell, 2024-10-15:
I'm happy to merge when you are ready.

6by9, 2024-10-15:
> Maybe it failed because of the 840 / 800?

Yes, as that didn't match any of the modes defined by the panel, it would have created one using the CVT formula
```
$cvt 800 480
# 800x480 59.48 Hz (CVT) hsync: 29.74 kHz; pclk: 29.50 MHz
Modeline "800x480_60.00"   29.50  800 824 896 992  480 483 493 500 -hsync +vsync
```
which matches you log earlier.

Now we have a matching mode it uses the existing mode and sets the rotation flag.

> I'm happy to merge when you are ready.

I'm happy.

mgineer85, 2024-10-15:
> > I'm happy to merge when you are ready.
> 
> I'm happy.

I'm very happy also :)  Thank you guys!

closed 2024-10-15 by mgineer85