raspberrypi/linux #6420: Raspberry Pi 5 sometimes has no LAN connection after booting

radekjuthner, 2024-10-15:
### Describe the bug

My raspberry Pi 5 sometimes (about one in seven) boots has no network connection. The Rpi receives an IP address from the DHCP server, but cannot ping any IP address from it. Interestingly, it is possible to ping the Rpi's IP address. The problem will be solved by disconnecting and then connecting the LAN cable or sudo ip link set end0 down and sudo ip link set end0 up. It's totally frustrating. I'm using Raspbian Bookworm with the latest update with Network manager. I know a similar problem was solved with Raspberry Pi 4 by adding genet.force_reneg=n to cmdline.txt. Unfortunately this doesn't work. Can anyone advise how to resolve this?

### Steps to reproduce the behaviour

I turn on or reboot the RPI5.
Sometimes the RPI does not have LAN access.
If this happens, the RPI will receive an IP address from the DHCP server and can be pinged from the router. It is not possible to ping any IP address from the RPI5.
Disconnecting and reconnecting the LAN cable will solve the problem. Or using sudo ip link set end0 down and sudo ip link set end0 up.

### Device (s)

Raspberry Pi 5

### System

cat /etc/rpi-issue
Raspberry Pi reference 2024-03-15
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, f19ee211ddafcae300827f953d143de92a5c6624, stage2

vcgencmd version
2024/10/10 00:36:47 
Copyright (c) 2012 Broadcom
version 0e63dd5e (release) (embedded)

uname -a
Linux raspi5 6.6.44-v8+ #1789 SMP PREEMPT Mon Aug  5 15:22:52 BST 2024 aarch64 GNU/Linux

### Logs

ip addr
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: end0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 2c:cf:67:5a:36:5b brd ff:ff:ff:ff:ff:ff
    inet 192.168.100.180/24 brd 192.168.100.255 scope global dynamic noprefixroute end0
       valid_lft 42914sec preferred_lft 42914sec
3: wlan0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 2c:cf:67:5a:36:5c brd ff:ff:ff:ff:ff:ff

ping 192.168.100.1
PING 192.168.100.1 (192.168.100.1) 56(84) bytes of data.
From 192.168.100.180 icmp_seq=10 Destination Host Unreachable

ip route
default via 192.168.100.1 dev end0 proto dhcp src 192.168.100.180 metric 20100 
192.168.100.0/24 dev end0 proto kernel scope link src 192.168.100.180 metric 100

dmesg | grep end0
[    1.047081] macb 1f00100000.ethernet end0: renamed from eth0
[    5.046914] macb 1f00100000.ethernet end0: PHY [1f00100000.ethernet-ffffffff:01] driver [Broadcom BCM54213PE] (irq=POLL)
[    5.046925] macb 1f00100000.ethernet end0: configuring for phy/rgmii-id link mode
[    9.150906] macb 1f00100000.ethernet end0: Link is Up - 1Gbps/Full - flow control tx

dmesg | grep eth0
[    0.698702] macb 1f00100000.ethernet eth0: Cadence GEM rev 0x00070109 at 0x1f00100000 irq 112 (2c:cf:67:5a:36:5b)
[    1.047081] macb 1f00100000.ethernet end0: renamed from eth0

sudo journalctl -e -f -u NetworkManager.service
říj 15 17:42:07 raspi5 systemd[1]: Starting NetworkManager.service - Network Manager...
říj 15 17:42:07 raspi5 NetworkManager[896]: <info>  [1729006927.8410] NetworkManager (version 1.42.4) is starting... (boot:4edd5f59-d5c3-4290-9682-c5fcf3adc2ba)
říj 15 17:42:07 raspi5 NetworkManager[896]: <info>  [1729006927.8411] Read config: /etc/NetworkManager/NetworkManager.conf (lib: 20-connectivity-debian.conf, no-mac-addr-change.conf)
říj 15 17:42:07 raspi5 NetworkManager[896]: <info>  [1729006927.8454] bus-manager: acquired D-Bus service "org.freedesktop.NetworkManager"
říj 15 17:42:07 raspi5 systemd[1]: Started NetworkManager.service - Network Manager.
říj 15 17:42:07 raspi5 NetworkManager[896]: <info>  [1729006927.8701] manager[0x55b29b5000]: monitoring kernel firmware directory '/lib/firmware'.
[... 32 lines cut ...]
říj 15 17:42:08 raspi5 NetworkManager[896]: <info>  [1729006928.1785] device (lo): state change: config -> ip-config (reason 'none', sys-iface-state: 'external')
říj 15 17:42:08 raspi5 NetworkManager[896]: <info>  [1729006928.1793] device (lo): state change: ip-config -> ip-check (reason 'none', sys-iface-state: 'external')
říj 15 17:42:08 raspi5 NetworkManager[896]: <info>  [1729006928.1853] modem-manager: ModemManager available
říj 15 17:42:08 raspi5 NetworkManager[896]: <info>  [1729006928.1956] device (lo): state change: ip-check -> secondaries (reason 'none', sys-iface-state: 'external')
říj 15 17:42:08 raspi5 NetworkManager[896]: <info>  [1729006928.1959] device (lo): state change: secondaries -> activated (reason 'none', sys-iface-state: 'external')
říj 15 17:42:08 raspi5 NetworkManager[896]: <info>  [1729006928.1965] device (lo): Activation: successful, device activated.
říj 15 17:42:12 raspi5 NetworkManager[896]: <info>  [1729006932.2622] device (end0): carrier: link connected
říj 15 17:42:12 raspi5 NetworkManager[896]: <info>  [1729006932.2626] device (end0): state change: unavailable -> disconnected (reason 'carrier-changed', sys-iface-state: 'managed')
říj 15 17:42:12 raspi5 NetworkManager[896]: <info>  [1729006932.2633] policy: auto-activating connection 'LAN' (2d47b88d-061d-4612-87bb-0bc3164bb538)
říj 15 17:42:12 raspi5 NetworkManager[896]: <info>  [1729006932.2639] device (end0): Activation: starting connection 'LAN' (2d47b88d-061d-4612-87bb-0bc3164bb538)
říj 15 17:42:12 raspi5 NetworkManager[896]: <info>  [1729006932.2639] device (end0): state change: disconnected -> prepare (reason 'none', sys-iface-state: 'managed')
říj 15 17:42:12 raspi5 NetworkManager[896]: <info>  [1729006932.2641] manager: NetworkManager state is now CONNECTING
říj 15 17:42:12 raspi5 NetworkManager[896]: <info>  [1729006932.2643] device (end0): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
říj 15 17:42:12 raspi5 NetworkManager[896]: <info>  [1729006932.2695] device (end0): state change: config -> ip-config (reason 'none', sys-iface-state: 'managed')
říj 15 17:42:12 raspi5 NetworkManager[896]: <info>  [1729006932.2700] dhcp4 (end0): activation: beginning transaction (timeout in 45 seconds)
říj 15 17:42:12 raspi5 NetworkManager[896]: <info>  [1729006932.2961] dhcp4 (end0): state changed new lease, address=192.168.100.180
říj 15 17:42:12 raspi5 NetworkManager[896]: <info>  [1729006932.2969] policy: set 'LAN' (end0) as default for IPv4 routing and DNS
říj 15 17:42:12 raspi5 NetworkManager[896]: <info>  [1729006932.3175] device (end0): state change: ip-config -> ip-check (reason 'none', sys-iface-state: 'managed')
říj 15 17:42:12 raspi5 NetworkManager[896]: <info>  [1729006932.3485] device (end0): state change: ip-check -> secondaries (reason 'none', sys-iface-state: 'managed')
říj 15 17:42:12 raspi5 NetworkManager[896]: <info>  [1729006932.3488] device (end0): state change: secondaries -> activated (reason 'none', sys-iface-state: 'managed')
říj 15 17:42:12 raspi5 NetworkManager[896]: <info>  [1729006932.3492] manager: NetworkManager state is now CONNECTED_SITE
říj 15 17:42:12 raspi5 NetworkManager[896]: <info>  [1729006932.3495] device (end0): Activation: successful, device activated.
říj 15 17:42:12 raspi5 NetworkManager[896]: <info>  [1729006932.3607] manager: startup complete
říj 15 17:42:12 raspi5 NetworkManager[896]: <info>  [1729006932.4731] manager: NetworkManager state is now CONNECTED_GLOBAL
říj 15 17:42:19 raspi5 NetworkManager[896]: <info>  [1729006939.7174] agent-manager: agent[795c5166a532d1cc,:1.56/org.freedesktop.nm-applet/1000]: agent registered

pelwell, 2024-10-15:
> Sometimes the RPI does not have LAN access.
> If this happens, the RPI will receive an IP address from the DHCP server and can be pinged from the router. It is not possible to ping any IP address from the RPI5.

There's an obvious contradiction in those statements - getting an IP address from the DHCP server requires traffic in both directions - but I'm not doubting the truth of what you say.

1. Can you explain the timeline in the jourmalctl log above - does it end in the "bad" state, or does it get fixed because the Ethernet cable was removed and reconnected sometine around 17:42:09?
2. When it's in the bad state, what does `ifconfig end0` report?
3. When it's in the bad state, does `sudo ifconfig end0 down; sudo ifconfig end0 up` change anything?

radekjuthner, 2024-10-15:
sudo ifconfig end0
end0: flags=4163<ENABLED,OMNIDIRECTION,RUNNING,MULTICAST> mtu 1500
 inet 192.168.100.180 net_mask 255.255.255.0 omnidirectional 192.168.100.255
 ether 2c:cf:67:5a:36:5b outgoing_queue_length 1000 (Ethernet)
 RX packets 23 bytes 1819 (1.7 KiB)
 RX Errors 0 Dropped 0 Overflow 0 Frames 0
 TX packets 130 bytes 14086 (13.7 KiB)
 TX errors 0 discarded 0 overflow 0 transmission 0 collisions 0
 device interrupt 112

Using sudo ifconfig end0 down and sudo ifconfig end0 up will restore the network connection.

pelwell, 2024-10-15:
And the answer to question 1?

pelwell, 2024-10-15:
1.  Asking it a different way, it would be interesting to see a journalctl log extract for what happens when cable is unplugged and reconnected.

4. Do the packet counts you get from `ifconfig end0` increase while pinging out from the Pi?

radekjuthner, 2024-10-16:
Bad state report 
sudo journalctl -e -f -u NetworkManager.service
říj 16 14:14:16 raspi5 systemd[1]: Starting NetworkManager.service - Network Manager...
říj 16 14:14:16 raspi5 NetworkManager[896]: <info>  [1729080856.6286] NetworkManager (version 1.42.4) is starting... (boot:01501af1-7305-425a-b0fc-3027e70fcea9)
říj 16 14:14:16 raspi5 NetworkManager[896]: <info>  [1729080856.6287] Read config: /etc/NetworkManager/NetworkManager.conf (lib: 20-connectivity-debian.conf, no-mac-addr-change.conf)
říj 16 14:14:16 raspi5 NetworkManager[896]: <info>  [1729080856.6324] bus-manager: acquired D-Bus service "org.freedesktop.NetworkManager"
říj 16 14:14:16 raspi5 systemd[1]: Started NetworkManager.service - Network Manager.
říj 16 14:14:16 raspi5 NetworkManager[896]: <info>  [1729080856.6524] manager[0x559801a000]: monitoring kernel firmware directory '/lib/firmware'.
říj 16 14:14:16 raspi5 NetworkManager[896]: <info>  [1729080856.6525] monitoring ifupdown state file '/run/network/ifstate'.
říj 16 14:14:16 raspi5 NetworkManager[896]: <info>  [1729080856.6575] hostname: hostname: using hostnamed
říj 16 14:14:16 raspi5 NetworkManager[896]: <info>  [1729080856.6576] hostname: static hostname changed from (none) to "raspi5"
říj 16 14:14:16 raspi5 NetworkManager[896]: <info>  [1729080856.6641] dns-mgr: init: dns=default,systemd-resolved rc-manager=symlink (auto)
říj 16 14:14:16 raspi5 NetworkManager[896]: <info>  [1729080856.6660] rfkill1: found Wi-Fi radio killswitch (at /sys/devices/platform/axi/1001100000.mmc/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/ieee80211/phy0/rfkill1) (driver brcmfmac)
říj 16 14:14:16 raspi5 NetworkManager[896]: <info>  [1729080856.6664] manager[0x559801a000]: rfkill: Wi-Fi hardware radio set disabled
říj 16 14:14:16 raspi5 NetworkManager[896]: <info>  [1729080856.6664] manager[0x559801a000]: rfkill: WWAN hardware radio set enabled
říj 16 14:14:16 raspi5 NetworkManager[896]: <info>  [1729080856.8493] Loaded device plugin: NMAtmManager (/usr/lib/aarch64-linux-gnu/NetworkManager/1.42.4/libnm-device-plugin-adsl.so)
říj 16 14:14:16 raspi5 NetworkManager[896]: <info>  [1729080856.8751] Loaded device plugin: NMTeamFactory (/usr/lib/aarch64-linux-gnu/NetworkManager/1.42.4/libnm-device-plugin-team.so)
říj 16 14:14:16 raspi5 NetworkManager[896]: <info>  [1729080856.8981] Loaded device plugin: NMWwanFactory (/usr/lib/aarch64-linux-gnu/NetworkManager/1.42.4/libnm-device-plugin-wwan.so)
říj 16 14:14:16 raspi5 NetworkManager[896]: <info>  [1729080856.9122] Loaded device plugin: NMBluezManager (/usr/lib/aarch64-linux-gnu/NetworkManager/1.42.4/libnm-device-plugin-bluetooth.so)
říj 16 14:14:16 raspi5 NetworkManager[896]: <info>  [1729080856.9337] Loaded device plugin: NMWifiFactory (/usr/lib/aarch64-linux-gnu/NetworkManager/1.42.4/libnm-device-plugin-wifi.so)
říj 16 14:14:16 raspi5 NetworkManager[896]: <info>  [1729080856.9341] manager: rfkill: Wi-Fi disabled by radio killswitch; disabled by state file
říj 16 14:14:16 raspi5 NetworkManager[896]: <info>  [1729080856.9342] manager: rfkill: WWAN enabled by radio killswitch; enabled by state file
říj 16 14:14:16 raspi5 NetworkManager[896]: <info>  [1729080856.9343] manager: Networking is enabled by state file
říj 16 14:14:16 raspi5 NetworkManager[896]: <info>  [1729080856.9360] settings: Loaded settings plugin: ifupdown ("/usr/lib/aarch64-linux-gnu/NetworkManager/1.42.4/libnm-settings-plugin-ifupdown.so")
říj 16 14:14:16 raspi5 NetworkManager[896]: <info>  [1729080856.9361] settings: Loaded settings plugin: keyfile (internal)
říj 16 14:14:16 raspi5 NetworkManager[896]: <info>  [1729080856.9361] ifupdown: management mode: unmanaged
říj 16 14:14:16 raspi5 NetworkManager[896]: <info>  [1729080856.9361] ifupdown:       interface-parser: parsing file /etc/network/interfaces
říj 16 14:14:16 raspi5 NetworkManager[896]: <info>  [1729080856.9362] ifupdown:       interface-parser: finished parsing file /etc/network/interfaces
říj 16 14:14:16 raspi5 NetworkManager[896]: <info>  [1729080856.9682] dhcp: init: Using DHCP client 'internal'
říj 16 14:14:16 raspi5 NetworkManager[896]: <info>  [1729080856.9686] manager: (lo): new Loopback device (/org/freedesktop/NetworkManager/Devices/1)
říj 16 14:14:16 raspi5 NetworkManager[896]: <info>  [1729080856.9703] device (lo): state change: unmanaged -> unavailable (reason 'connection-assumed', sys-iface-state: 'external')
říj 16 14:14:16 raspi5 NetworkManager[896]: <info>  [1729080856.9709] device (lo): state change: unavailable -> disconnected (reason 'connection-assumed', sys-iface-state: 'external')
říj 16 14:14:16 raspi5 NetworkManager[896]: <info>  [1729080856.9719] device (lo): Activation: starting connection 'lo' (54124b4f-73aa-41a8-bbc6-5af0e0c266a0)
říj 16 14:14:16 raspi5 NetworkManager[896]: <info>  [1729080856.9741] manager: (end0): new Ethernet device (/org/freedesktop/NetworkManager/Devices/2)
říj 16 14:14:16 raspi5 NetworkManager[896]: <info>  [1729080856.9747] device (end0): state change: unmanaged -> unavailable (reason 'managed', sys-iface-state: 'external')
říj 16 14:14:16 raspi5 NetworkManager[896]: <info>  [1729080856.9958] device (wlan0): driver supports Access Point (AP) mode
říj 16 14:14:16 raspi5 NetworkManager[896]: <info>  [1729080856.9968] manager: (wlan0): new 802.11 Wi-Fi device (/org/freedesktop/NetworkManager/Devices/3)
říj 16 14:14:16 raspi5 NetworkManager[896]: <info>  [1729080856.9974] device (wlan0): state change: unmanaged -> unavailable (reason 'managed', sys-iface-state: 'external')
[... cut ...]
ping: sendmsg: There is no path to the computer

and

sudo ifconfig end0
end0: flags=4163<ENABLED,OMNIDIRECTION,RUNNING,MULTICAST> mtu 1500
 ether 2c:cf:67:5a:36:5b outgoing_queue_length 1000 (Ethernet)
 RX packets 41 bytes 2647 (2.5 KiB)
 RX Errors 0 Dropped 0 Overflow 0 Frames 0
 TX packets 925 bytes 68361 (66.7 KiB)
 TX errors 0 discarded 0 overflow 0 transmission 0 collisions 0
 device interrupt 112

In bad state, it is possible to ping the RPI5. Network manager restart and sudo dhclient -v do not work on RPI

radekjuthner, 2024-10-18:
I have one more clue. The loss of network connection seems to occur already in the bootloader. I started RPI5, pressed SHIFT. RPI loaded Network installer and then lost network connection. Could not use Network installer. Replugging the network cable solved the problem. All of this means that the network connection problem will occur even before Linux is booted.
After booting bad state, the already mentioned problem occurs. Ping from RPI to router doesn't work, but router to RPI does. Some pings do not end with "Destination Host Unrecheable".

[image]

HovnovoD, 2024-11-07:
Hi,
I have exactly the same problem.
I tried the official Raspberry OS, Raspberry OS Light and Ubuntu.
The problem is for both systems.
Therefore, I also think that the problem is already booting firmware.
I always update to the new firmware and I am always disappointed that the problem has not been solved.
Unfortunately, I did not find any threads in the Internet discussions to describe this problem.
Is it just a unique problem?

Did you find any solution to this problem?

I apologize for my English language.

radekjuthner, 2024-11-08:
Hi.
I have added to the boot sequence for now

ifconfig eth0 down
ifconfig eth0 up. 

This will ensure that the LAN disconnects on boot and connects immediately. So far it seems to be working.

Instructions:

sudo nano /etc/systemd/system/network-restart.service

insert content:

[Unit]
Description=Restart Ethernet interface at boot
After=network.target

[Service]
Type=oneshot
ExecStart=/sbin/ifconfig eth0 down
ExecStartPost=/sbin/ifconfig eth0 up
RemainAfterExit=true

[Install]
WantedBy=multi-user.target

then
sudo systemctl enable network-restart.service
sudo systemctl start network-restart.service

HovnovoD, 2024-11-08:
Hi Radek,

Thank you for your advice. It seems that it works this way. I wanted to use these commands in this way as well, but unfortunately, I didn’t know how to run them during system boot. When I ran them as "autostart" after the system started, these commands didn't work this way.

I’ll let you know more, probably in about a month, to see if it continues to work like this.

Thank you Radek
Díky Radku : )

radekjuthner, 2024-11-27:
The problem seems to be going deeper. For about two months, the RPI booted without a problem. Network unavailable again today. Are you using the RPI itself or some accessories? I have a Pironman 5 with SSD. There might be a problem with the accessory. You can also write to radek.juthner@gmail.com. The community doesn't seem interested in investigating the problem in any way, let alone solving it.

peterharperuk, 2024-11-28:
Note: Duplicate report here https://github.com/raspberrypi/firmware/issues/1922

strayedstar, 2024-12-24:
OH,I have the same problem?

gearhead, 2024-12-29:
I experience the same thing. I do not, though, have any logs for it. I just unplug/re-plug. I develop on Pi2/3/4/5 and have never seen this except on the Pi5. I have never seen it grab a DHCP address, though. When it fails, it has always has a 169.x.x.x address.

alessandromrc, 2025-01-22:
Does the Ethernet NIC even blink at all or shows any activity? I don't really understand the fact that you are saying that the DHCP server is able to give you an IP address and also that you are able to ping the pi from another machine but not the other way around... are you sure that you aren't getting an [APIPA](https://wiki.wireshark.org/APIPA) address?

It would be interesting to know which router / managed switch you are using just to make sure the other side is not the issue, I personally tried on a pi5 with a managed Cisco Switch and an OpenWRT and everything seems to be fine

alessandromrc, 2025-01-22:
> The network card is active and the LEDs show activity.
> My RPI receives an IP address (192.168.100.180) from the router. Unfortunately, it is not possible to ping from it to other devices in the network, including the router. But the interesting thing is that RPI can be pinged from devices on the network.
> My router is built on Banana PI BPI-R3 and Arch linux. Another 6 devices (Smart TV, PC, RPI 4, Orange PI and others) are connected to it via LAN without apparent problems.

Interesting, are you able to use Wireshark or any other tool when this happens so that we can see if the Pi even sends the packet? It would be useful to know if the packets gets stuck while being sent or the other device responds back but the response gets ignored by the Pi.

radekjuthner, 2025-01-24:
Sorry I have no experience with wireshark.

alessandromrc, 2025-01-24:
@radekjuthner This is happening with the current version of Raspberry Pi OS too right? I want to perform some testings myself but I need to be sure that the latest available version still has this issue.

gearhead, 2025-01-24:
Yes. I have seen it with the latest PiOS 6.6.y. When I have queried it, I have seen it with an APIPA address. 169.254.x.x

radekjuthner, 2025-01-25:
@alessandromrc 

I keep the system up to date. Unfortunately, the error occurs very sporadically and randomly. It is difficult to confirm its occurrence at the moment.

alessandromrc, 2025-01-25:
Well @radekjuthner, when this issue occurs, could you run `tcpdump` on the Raspberry Pi to check if it’s sending and receiving packets?

To install tcpdump you need to be sudo and run this command:
```bash
sudo apt install tcpdump
```

Then if you just want to log incoming/outgoing packets you can just run
```bash
sudo tcpdump
```

In my case I would like you to run:
```bash
sudo tcpdump -w dump
```

That command will dump every packet that is outgoing and incoming on all the interfaces available on the Pi. (In this case you can just ping another machine like your gateway from your Pi)

I would like to have such dump so that I can analyze what is going on within the networking stack.

It would also be helpful to get an overview of your network interfaces. Please run:
```bash
ip a
```
and send the output here.

alessandromrc, 2025-01-25:
I just rebooted my Pi around 50 times and I was unable to replicate this, this is my kernel version:
```
Linux raspberrypi5 6.6.51+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux
```

and this is my rpi-issue:
```
Raspberry Pi reference 2024-11-19
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 891df1e21ed2b6099a2e6a13e26c91dea44b34d4, stage5
```

radekjuthner, 2025-01-30:
@alessandromrc 
An error occurred a while ago. I am attaching the required outputs.

`sudo tcpdump
tcpdump: verbose output suppressed, use -v[v]... for full protocol decode
listening on eth0, link-type EN10MB (Ethernet), snapshot length 262144 bytes
19:29:28.816899 ARP, Request who-has 192.168.100.1 tell 192.168.100.190, length 46
19:29:28.870753 ARP, Request who-has 192.168.100.130 tell 192.168.100.180, length 28
19:29:29.820434 IP 192.168.100.190.8001 > 224.0.0.7.8001: UDP, length 179
19:29:29.878315 ARP, Request who-has 192.168.100.130 tell 192.168.100.180, length 28
19:29:29.895528 IP 192.168.100.190.45626 > 192.168.100.255.15600: UDP, length 35
19:29:30.861395 ARP, Request who-has 192.168.100.1 tell 192.168.100.190, length 46
19:29:30.902316 ARP, Request who-has 192.168.100.130 tell 192.168.100.180, length 28
19:29:31.823247 IP 192.168.100.190.8001 > 224.0.0.7.8001: UDP, length 179
19:29:31.926403 ARP, Request who-has 192.168.100.130 tell 192.168.100.180, length 28
19:29:32.896208 IP 192.168.100.190.45997 > 239.255.255.250.15600: UDP, length 35
19:29:32.904775 ARP, Request who-has 192.168.100.1 tell 192.168.100.190, length 46
19:29:32.950314 ARP, Request who-has 192.168.100.130 tell 192.168.100.180, length 28
19:29:41.142510 ARP, Request who-has 192.168.100.130 tell 192.168.100.180, length 28
19:29:53.430515 ARP, Request who-has 192.168.100.130 tell 192.168.100.180, length 28
19:30:05.718540 ARP, Request who-has 192.168.100.130 tell 192.168.100.180, length 28
19:30:05.788797 IP 192.168.100.190.3955 > 239.255.255.250.1900: UDP, length 307
19:30:05.789690 IP 192.168.100.190.3955 > 239.255.255.250.1900: UDP, length 316
19:30:05.790547 IP 192.168.100.190.3955 > 239.255.255.250.1900: UDP, length 369
19:30:05.791362 IP 192.168.100.190.3955 > 239.255.255.250.1900: UDP, length 365
19:30:05.820786 IP 192.168.100.190.3955 > 239.255.255.250.1900: UDP, length 312
19:30:05.821598 IP 192.168.100.190.3955 > 239.255.255.250.1900: UDP, length 321
19:30:05.822278 IP 192.168.100.190.3955 > 239.255.255.250.1900: UDP, length 374
19:30:05.822882 IP 192.168.100.190.3955 > 239.255.255.250.1900: UDP, length 360
19:30:05.864974 IP 192.168.100.190.3955 > 239.255.255.250.1900: UDP, length 306
19:30:05.865684 IP 192.168.100.190.3955 > 239.255.255.250.1900: UDP, length 315
19:30:05.866320 IP 192.168.100.190.3955 > 239.255.255.250.1900: UDP, length 362
19:30:05.866953 IP 192.168.100.190.3955 > 239.255.255.250.1900: UDP, length 370
19:30:05.867550 IP 192.168.100.190.3955 > 239.255.255.250.1900: UDP, length 372
19:30:05.868137 IP 192.168.100.190.3955 > 239.255.255.250.1900: UDP, length 360
19:30:05.868829 IP 192.168.100.190.3955 > 239.255.255.250.1900: UDP, length 380
19:30:05.873551 IP 192.168.100.190.8001 > 224.0.0.7.8001: UDP, length 179
19:30:05.902903 IP 192.168.100.190.43101 > 192.168.100.255.15600: UDP, length 35
19:30:05.997839 IP 192.168.100.190.3955 > 239.255.255.250.1900: UDP, length 307
19:30:05.998867 IP 192.168.100.190.3955 > 239.255.255.250.1900: UDP, length 316
19:30:05.999784 IP 192.168.100.190.3955 > 239.255.255.250.1900: UDP, length 369
19:30:06.000812 IP 192.168.100.190.3955 > 239.255.255.250.1900: UDP, length 365
19:30:06.028887 IP 192.168.100.190.3955 > 239.255.255.250.1900: UDP, length 312
19:30:06.029584 IP 192.168.100.190.3955 > 239.255.255.250.1900: UDP, length 321
19:30:06.030211 IP 192.168.100.190.3955 > 239.255.255.250.1900: UDP, length 374
19:30:06.030796 IP 192.168.100.190.3955 > 239.255.255.250.1900: UDP, length 360
19:30:06.074841 IP 192.168.100.190.3955 > 239.255.255.250.1900: UDP, length 306
19:30:06.075575 IP 192.168.100.190.3955 > 239.255.255.250.1900: UDP, length 315
19:30:06.076424 IP 192.168.100.190.3955 > 239.255.255.250.1900: UDP, length 362
19:30:06.077249 IP 192.168.100.190.3955 > 239.255.255.250.1900: UDP, length 370
19:30:06.077883 IP 192.168.100.190.3955 > 239.255.255.250.1900: UDP, length 372
19:30:06.078472 IP 192.168.100.190.3955 > 239.255.255.250.1900: UDP, length 360
19:30:06.079047 IP 192.168.100.190.3955 > 239.255.255.250.1900: UDP, length 380
19:30:11.883985 IP 192.168.100.190.8001 > 224.0.0.7.8001: UDP, length 179
19:30:11.904097 IP 192.168.100.190.36031 > 192.168.100.255.15600: UDP, length 35
19:30:13.469929 ARP, Request who-has 192.168.100.1 tell 192.168.100.190, length 46
19:30:13.886569 IP 192.168.100.190.8001 > 224.0.0.7.8001: UDP, length 179
19:30:14.904749 IP 192.168.100.190.52739 > 239.255.255.250.15600: UDP, length 35
19:30:15.489311 ARP, Request who-has 192.168.100.1 tell 192.168.100.190, length 46
19:30:15.889697 IP 192.168.100.190.8001 > 224.0.0.7.8001: UDP, length 179
19:30:17.513538 ARP, Request who-has 192.168.100.1 tell 192.168.100.190, length 46
19:30:17.892769 IP 192.168.100.190.8001 > 224.0.0.7.8001: UDP, length 179
19:30:17.905326 IP 192.168.100.190.57780 > 192.168.100.255.15600: UDP, length 35
19:30:19.541614 ARP, Request who-has 192.168.100.1 tell 192.168.100.190, length 46
19:30:19.895984 IP 192.168.100.190.8001 > 224.0.0.7.8001: UDP, length 179
19:30:20.905945 IP 192.168.100.190.34879 > 239.255.255.250.15600: UDP, length 35
19:30:21.565381 ARP, Request who-has 192.168.100.1 tell 192.168.100.190, length 46
19:30:21.900348 IP 192.168.100.190.8001 > 224.0.0.7.8001: UDP, length 179
19:30:22.405985 f6:ed:fb:39:73:75 (oui Unknown) > Broadcast Null Unnumbered, xid, Flags [Response], length 6: 01 00
19:30:22.489886 IP 0.0.0.0 > 224.0.0.22: igmp v3 report, 1 group record(s)
[... 105 lines cut ...]
[... cut ...]
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 2c:cf:67:5a:36:5b brd ff:ff:ff:ff:ff:ff
    inet 192.168.100.180/24 brd 192.168.100.255 scope global dynamic noprefixroute eth0
       valid_lft 42314sec preferred_lft 42314sec
3: wlan0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 2c:cf:67:5a:36:5c brd ff:ff:ff:ff:ff:ff

Ping error via GUI:
[image]

Ping from routerboard  IP 192.168.100. 1to Raspberry Pi5 IP 192.168.100.180:
[image]

alessandromrc, 2025-02-01:
@radekjuthner,

The frequent ARP requests suggest that the Raspberry Pi is attempting to resolve the Gateway (RouterBOARD) MAC address but isn't able to get a response from other devices in the network.

Have you tried assigning a static IP to the Raspberry Pi to see if that resolves the issue? 

Also, it doesn't appear that the DHCP lease has expired, as you can see here: 

```valid_lft 42314sec preferred_lft 42314sec```

The Pi still has around 12 hours left on the lease.

When the issue occurs, could you run the following command in the terminal?

```bash
ip neighbour
```

In any case, assigning a static IP could potentially solve your problem.

radekjuthner, 2025-02-09:
@alessandromrc 
ip neighbor 
192.168.100.136 dev eth0 FAILED 
192.168.100.130 dev eth0 INCOMPLETE

alessandromrc, 2025-02-10:
> [@alessandromrc](https://github.com/alessandromrc) ip neighbor 192.168.100.136 dev eth0 FAILED 192.168.100.130 dev eth0 INCOMPLETE

That seems like you're having some issues with reaching other devices in the network as you're not able to get their MAC address... Are you able to ping those devices from the pi? I suspect that you cannot but maybe the other way around could. 

This is a quite weird issue, the only thing I can think of could be a lower-level sided thing within the network NIC management... I am not so sure about how the NIC works on the pi5 as there's the RP1 chip.

alessandromrc, 2025-02-10:
> Could this be some race condition? I only see it happen at boot. Saw it
> last night as well. It's as though it doesn't wait long enough for a DHCP
> response. When it fails to 'connect' or gets an apipa address. When I
> restart the network manager after it boots (connman or network manager) it
> will connect.
> […](#)

This is an interesting point of view as I haven't been able to reproduce it, do you have any idea on how to make this reproducible in a manner that is repetable by others?

I wonder if I can code my own DHCP server and "fake" a delay response so that it lands in such weird condition.

radekjuthner, 2025-02-10:
I suspect the problem is somewhere in the firmware. The Raspberry Pi 5 has a bootloader that starts before Linux. The bootloader can connect to the network, download the operating system and install it online. It uses DHCP. I noticed the IP address problem in the bootloader before Linux started.

alessandromrc, 2025-02-10:
> I suspect the problem is somewhere in the firmware. The Raspberry Pi 5 has a bootloader that starts before Linux. The bootloader can connect to the network, download the operating system and install it online. It uses DHCP. I noticed the IP address problem in the bootloader before Linux started.

Whenever I have some free time I will write a DHCP server and fake it being really slow to see if the problem is caused by the DHCP Client (pi).

alessandromrc, 2025-02-11:
@radekjuthner @gearhead pure curiosity... are you guys running an SD card or an NVME drive?

(All my tests have been done while running an NVME drive which is surely faster than any SD card... that could be something keeping the boot process to lose performances and pretty much risk to yield too much and lose the DHCP response from the server)

gearhead, 2025-02-11:
@alessandromrc All my testing has been with SD cards.

alessandromrc, 2025-02-11:
> @alessandromrc All my testing has been with SD cards. 

Are you aware of the brand and model?

I wonder if I can get something similar or identical to what you are running.

gearhead, 2025-02-11:
@alessandromrc I have a bunch of these:
SanDisk SDSQUAR-016G-GN6MM

alessandromrc, 2025-02-11:
> [@alessandromrc](https://github.com/alessandromrc) I have a bunch of these: SanDisk SDSQUAR-016G-GN6MM

I should have a 128GB similar variant around... I will look for it and see if I get any issues with that configuration whenever I got some free time.

Thanks!

radekjuthner, 2025-02-12:
@alessandromrc 
I use SSD by default. I also noticed the problem with the SD card with MX linux.

alessandromrc, 2025-02-13:
I have continued testing with the SD Card but still no issues at all : /

radekjuthner, 2025-02-13:
As I wrote above. The problem occurs once every 10-20 boots. Lately, more often after a reboot than during a cold start.
Maybe we simply have defective pieces of raspberry.

teh-masterer, 2025-04-01:
I have a similar issue. I can ping the RPI, and do a portscan to see ports are open. However, I couldn't initiate any connections, and outbound connections were not working either. It definitely seems like a firmware or hardware issue with the network card. 
This occurs sometimes, when the system is running, and it persists through boot. Sometimes it will start communicating again if I replug it into other Ethernet ports on my switch. 

I previously thought it was due to a PoE Hat, but it has been since removed and the issue has persisted. 

For now I will use a USB Ethernet Adapter and see if the issue persists. 

Difference from my setup and this is that I don't use DHCP.

alessandromrc, 2025-04-02:
> I have a similar issue. I can ping the RPI, and do a portscan to see ports are open. However, I couldn't initiate any connections, and outbound connections were not working either. It definitely [...]

Which network switch are you currently using?

radekjuthner, 2025-04-12:
One small observation: the connection problem occurs more often during reboots. It doesn't matter if the RPI is a DHCP client or if I have a static IP address.

malikbrh, 2025-06-10:
Any update on this? My Pi is unreachable except via Tailscale and no patch seems to be out until today.

malikbrh, 2025-06-10:
> > My Pi is unreachable except via Tailscale
> 
> What is the connectivity when you are using Tailscale (which is a VPN?)?

Pi is connected via internal (no USB adapter) Ethernet to LAN/WAN. Can access it anywhere using Tailscale (a wireguard mesh VPN), but the local IP 192.168.x.x is unreachable even when disabling the firewall, no SSH, no Ping, no HTTP response.

pelwell, 2025-06-10:
Break it down to a more basic level. Are the mesh VPN packets carried over Ethernet, WiFi or something else? If the answer is Ethernet, then you have routing problem rather than an Ethernet connectivity problem.

malikbrh, 2025-06-10:
Pure Ethernet, no WiFi. The Pi is reachable with its services over the VPN, but the local IP never replies.

Topology:

Pi <-> Switch <-> My PC
.240              .160 (local IPs)
.a                 .b (Mesh VPN addresses)

when calling my Pi with .a from .b, it works fine.
if i ping the .240 address from .160 it never replies
if i ping the .240 address directly from the Pi it works, meaning at least it assigned the address to itself correctly.

By searching Google about this issue, I ended up on this thread which seems to exactly describe my problem.

Hope my description was clearer than my previous message, thanks for your help !