raspberrypi/linux #6424: ath11k kernel panic on Pi 5 with 8GB RAM, but not on 2GB (DMA/PCI-E kernel panic)

omerk, 2024-10-17:
### Describe the bug

ath11k kernel module works as expected on a Raspberry Pi 5 board with 2 GB RAM but the same image (same boot media) fails with a DMA/memory related kernel panic on the 8GB unit.

Limiting the memory of the 8GB unit to 2GB (`mem=2G` in `cmdline.txt`) fixes the issue. Detailed logs below.

Test setup consists of two units:

Unit 1: Raspberry Pi 5, 2GB RAM, Official M.2 Hat, QCN9074 WiFi module (PCI-E)
Unit 2: Raspberry Pi 5, 8GB RAM, Official M.2 Hat, QCN9074 WiFi module (PCI-E)

WiFi modules used are identical brand/model and from the same batch, boot media is shared across the two units to ensure there are no config-related issues.

Not entirely sure if this is specifically an issue with the ath11k driver, as it is seems to work on other platforms, perhaps this is a BCM2712 DMA / PCI-E restriction? Speculating, of course, thanks in advance for your assistance.

### Steps to reproduce the behaviour

(On a fresh device, with no WiFi configuration)

  - Compile, install and boot up the custom kernel
  - Use `nmcli` to connect to a WiFi network
  - kernel panic

(With a valid WiFi configuration set up)

  - Boot device with custom kernel
  - kernel panic

### Device (s)

Raspberry Pi 5

### System

Kernel version:

```
$ git rev-parse HEAD
84ab77459e61c648299d32464127b89ca65de40a

$ uname -a
Linux raspberrypi 6.6.56-v8-16k-sifly+ #1 SMP PREEMPT Thu Oct 17 13:34:10 BST 2024 aarch64 GNU/Linux
```

`.config` used to compile the kernel, which is essentially the standard 2712 config with `ath11k` enabled, attached:  [kernel-config.zip](https://github.com/user-attachments/files/17416189/kernel-config.zip)

`config.txt` used on device:

```
dtoverlay=disable-wifi
dtoverlay=disable-bt

# For QCN9074
dtparam=pciex1
dtparam=pciex1_gen=3

# Force PCIe config to support 32bit DMA addresses at the expense of having to bounce buffers.
# https://github.com/raspberrypi/firmware/blob/b154632e320b87ea95c6ce8b59f96dbbe523ecf1/boot/overlays/README#L3597
dtoverlay=pcie-32bit-dma

# Compatibility features
# https://github.com/raspberrypi/firmware/blob/b154632e320b87ea95c6ce8b59f96dbbe523ecf1/boot/overlays/README#L3611
# no-mip: Use if a) more than 8 interrupt vectors are required or b) the EP requires DMA and MSI addresses to be 32bit.
dtoverlay=pciex1-compat-pi5,no-mip

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

[... 267 lines cut ...]
    inet 192.168.1.185/24 brd 192.168.1.255 scope global dynamic noprefixroute wlan0
       valid_lft 86384sec preferred_lft 86384sec
    inet6 fe80::5a4e:f962:55b2:ca18/64 scope link noprefixroute
       valid_lft forever preferred_lft forever

$ ping raspberrypi.com
PING raspberrypi.com (104.21.88.234) 56(84) bytes of data.
64 bytes from 104.21.88.234 (104.21.88.234): icmp_seq=1 ttl=58 time=7.68 ms
64 bytes from 104.21.88.234 (104.21.88.234): icmp_seq=2 ttl=58 time=7.94 ms
64 bytes from 104.21.88.234 (104.21.88.234): icmp_seq=3 ttl=58 time=8.53 ms
^C
--- raspberrypi.com ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 2002ms
rtt min/avg/max/mdev = 7.677/8.050/8.531/0.356 ms
```

By limiting the memory to 2GB, the 8GB unit works as expected.

### Additional context

I have tried various permutations of the following config options in `cmdline.txt` with no success:

  - `iommu=soft`
  - `iommu.strict=1`
  - `coherent_pool=1M`

P33M, 2024-11-13:
Are you saying that using
```
dtoverlay=pcie-32bit-dma
dtoverlay=pciex1-compat-pi5,no-mip
dtparam=pciex1
dtparam=pciex1_gen=3
```
in config.txt on an 8GB device results in a kernel panic?
What happens if you remove `dtparam=pciex1_gen=3`?

P33M, 2024-11-13:
It's likely this bug - https://patchwork.kernel.org/project/linux-wireless/patch/20231212031914.47339-1-imguzh@gmail.com/#25662182

Usage of virt_to_phys() is bad. This is fixed in kernel 6.9 and onwards. Our rpi-6.12.y branch is mostly functional (and will be the next target for rpi-update releases), can you try building that?

P33M, 2024-11-18:
Remove `dtoverlay=pcie-32bit-dma` from config.txt - it should not be necessary.

omerk, 2024-11-19:
With the system running as it was:
```
$ dmesg | grep -i cma
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000002000000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x0000000002000000..0x0000000005ffffff (65536 KiB) map reusable linux,cma
[    0.026104] Memory: 8166224K/8384512K available (13952K kernel code, 2294K rwdata, 4768K rodata, 5376K init, 837K bss, 145312K reserved, 65536K cma-reserved)
```

After adding `cma=128M@128M` to `cmdline.txt`:
```
$ dmesg | grep -i cma
[    0.000000] Reserved memory: bypass linux,cma node, using cmdline CMA params instead
[    0.000000] OF: reserved mem: node linux,cma compatible matching fail
[    0.000000] cma: Reserved 128 MiB at 0x0000000008000000 on node -1
[    0.000000] Kernel command line: reboot=w coherent_pool=1M 8250.nr_uarts=1 pci=pcie_bus_safe cgroup_disable=memory numa_policy=interleave  smsc95xx.macaddr=2C:CF:67:67:8D:23 vc_mem.mem_base=0x3fc00000 vc_mem.mem_size=0x40000000  console=ttyAMA10,115200 console=tty1 root=PARTUUID=8c5b1cb2-02 rootfstype=ext4 fsck.repair=yes cma=128M@128M rootwait
[    0.049943] Memory: 8100688K/8384512K available (13952K kernel code, 2294K rwdata, 4768K rodata, 5376K init, 837K bss, 145312K reserved, 131072K cma-reserved)
```

Config + ath11k output:
```
$ sed -n 1,4p /boot/firmware/config.txt
dtparam=pciex1
dtparam=pciex1_gen=3
dtoverlay=pcie-32bit-dma
dtoverlay=pciex1-compat-pi5,no-mip

$ dmesg | grep ath11k
[    7.250378] ath11k_pci 0000:01:00.0: BAR 0 [mem 0x1b80000000-0x1b801fffff 64bit]: assigned
[    7.250411] ath11k_pci 0000:01:00.0: enabling device (0000 -> 0002)
[    7.250682] ath11k_pci 0000:01:00.0: MSI vectors: 16
[    7.250694] ath11k_pci 0000:01:00.0: qcn9074 hw1.0
[    7.743254] ath11k_pci 0000:01:00.0: chip_id 0x0 chip_family 0x0 board_id 0xa0 soc_id 0xffffffff
[    7.743264] ath11k_pci 0000:01:00.0: fw_version 0x270206d0 fw_build_timestamp 2022-08-04 12:48 fw_build_id WLAN.HK.2.7.0.1-01744-QCAHKSWPL_SILICONZ-1
[    9.113350] ath11k_pci 0000:01:00.0: swiotlb buffer is full (sz: 1048583 bytes), total 32768 (slots), used 992 (slots)
[    9.113363] ath11k_pci 0000:01:00.0: failed to set up tcl_comp ring (0) :-12
[    9.113412] ath11k_pci 0000:01:00.0: failed to init DP: -12
```

No change, unfortunately.

P33M, 2024-11-20:
Can you build a rpi-6.12.y kernel with `CONFIG_SWIOTLB_DYNAMIC=y` and test?

omerk, 2024-11-20:
Trying to figure out if this is a problem with my config, but so far no luck building 12856cc6850854a062b7aa2ee55786257e266168:

```
...
  CC [M]  drivers/media/dvb-frontends/itd1000.o
  CC [M]  drivers/media/dvb-frontends/ves1820.o
  CC [M]  drivers/media/dvb-frontends/ves1x93.o
  CC [M]  drivers/media/dvb-frontends/zd1301_demod.o
  CC [M]  drivers/media/dvb-frontends/zl10036.o
  CC [M]  drivers/media/dvb-frontends/zl10039.o
  CC [M]  drivers/media/dvb-frontends/zl10353.o
  LD [M]  drivers/media/dvb-frontends/cxd2820r.o
  LD [M]  drivers/media/dvb-frontends/drxd.o
  LD [M]  drivers/media/dvb-frontends/drxk.o
  LD [M]  drivers/media/dvb-frontends/stb0899.o
  LD [M]  drivers/media/dvb-frontends/stv0900.o
  AR      drivers/built-in.a
  AR      built-in.a
  AR      vmlinux.a
  LD      vmlinux.o
  OBJCOPY modules.builtin.modinfo
  GEN     modules.builtin
  GEN     .vmlinux.objs
  MODPOST Module.symvers
ERROR: modpost: "__swiotlb_find_pool" [drivers/gpu/drm/vc4/vc4.ko] undefined!
make[2]: *** [scripts/Makefile.modpost:145: Module.symvers] Error 1
make[1]: *** [/tmp/raspios-builder/linux/Makefile:1888: modpost] Error 2
make: *** [Makefile:224: __sub-make] Error 2
```

P33M, 2024-11-20:
Hmm, reliance on an unexported symbol. It builds for me with `git revert 10da9e6868942` - shouldn't affect video on Pi 5 as it has IOMMUs.

6by9, 2024-11-20:
10da9e6868942 is using `is_swiotlb_buffer` which I thought was valid.

However that then uses `swiotlb_find_pool` if CONFIG_SWIOTLB_DYNAMIC is defined.
The usage changes between 6.6 and 6.12, so that may solve some of those issues.

omerk, 2024-11-22:
Still no luck even with 10da9e6 reverted.

Here's what I am doing, and the results with both stock cmdline.txt and the modified one:

```
git clone --branch rpi-6.12.y https://github.com/raspberrypi/linux
cd linux
git revert 10da9e6868942
KERNEL=kernel_2712
make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- bcm2711_defconfig
make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- menuconfig

Library routines  --->
  [*] Dynamic allocation of DMA bounce buffers

Device Drivers  --->
  [*] Network device support  --->
    [*]   Wireless LAN  --->
      <M>     Qualcomm Technologies 802.11ax chipset support
        <M>       Atheros ath11k PCI support
        [*]       QCA ath11k debugging
        [*]       ath11k tracing support

make -j16 ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- Image modules dtbs

<boot media mounted under /tmp/rpi-ssd>
RPI_SSD=/tmp/rpi-ssd

sudo cp arch/arm64/boot/Image $RPI_SSD/boot/$KERNEL.img
sudo cp arch/arm64/boot/dts/broadcom/*.dtb $RPI_SSD/boot/
sudo cp arch/arm64/boot/dts/overlays/*.dtb* $RPI_SSD/boot/overlays/
sudo cp arch/arm64/boot/dts/overlays/README $RPI_SSD/boot/overlays/

sudo env PATH=$PATH make -j16 ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- INSTALL_MOD_PATH=$RPI_SSD/rootfs modules_install

------------------

config.txt:

dtoverlay=disable-wifi
dtoverlay=disable-bt

dtparam=pciex1
dtparam=pciex1_gen=3

### -----------
### For QCN9074

# Force PCIe config to support 32bit DMA addresses at the expense of having to bounce buffers.
# https://github.com/raspberrypi/firmware/blob/b154632e320b87ea95c6ce8b59f96dbbe523ecf1/boot/overlays/README#L3597
dtoverlay=pcie-32bit-dma

# Compatibility features
# https://github.com/raspberrypi/firmware/blob/b154632e320b87ea95c6ce8b59f96dbbe523ecf1/boot/overlays/README#L3611
# no-mip: Use if a) more than 8 interrupt vectors are required or b) the EP requires DMA and MSI addresses to be 32bit.
dtoverlay=pciex1-compat-pi5,no-mip

### -----------

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Additional overlays and parameters are documented
# /boot/firmware/overlays/README

[... 61 lines cut ...]
$ cat /boot/firmware/cmdline.txt
console=serial0,115200 console=tty1 root=PARTUUID=8c5b1cb2-02 rootfstype=ext4 fsck.repair=yes cma=128M@128M rootwait

$ dmesg | grep cma
[    0.000000] Reserved memory: bypass linux,cma node, using cmdline CMA params instead
[    0.000000] OF: reserved mem: node linux,cma compatible matching fail
[    0.000000] cma: Reserved 128 MiB at 0x0000000008000000 on node -1
[    0.000000] Kernel command line: reboot=w coherent_pool=1M 8250.nr_uarts=1 pci=pcie_bus_safe cgroup_disable=memory numa_policy=interleave  smsc95xx.macaddr=2C:CF:67:67:8D:23 vc_mem.mem_base=0x3fc00000 vc_mem.mem_size=0x40000000  console=ttyAMA10,115200 console=tty1 root=PARTUUID=8c5b1cb2-02 rootfstype=ext4 fsck.repair=yes cma=128M@128M rootwait
[    0.025889] Memory: 7993620K/8384512K available (14016K kernel code, 2252K rwdata, 4668K rodata, 5120K init, 577K bss, 254376K reserved, 131072K cma-reserved)

$ dmesg | grep ath11
[    7.000662] ath11k_pci 0000:01:00.0: BAR 0 [mem 0x1b80000000-0x1b801fffff 64bit]: assigned
[    7.000688] ath11k_pci 0000:01:00.0: enabling device (0000 -> 0002)
[    7.000875] ath11k_pci 0000:01:00.0: MSI vectors: 16
[    7.000882] ath11k_pci 0000:01:00.0: qcn9074 hw1.0
[    7.476077] ath11k_pci 0000:01:00.0: chip_id 0x0 chip_family 0x0 board_id 0xa0 soc_id 0xffffffff
[    7.476085] ath11k_pci 0000:01:00.0: fw_version 0x270206d0 fw_build_timestamp 2022-08-04 12:48 fw_build_id WLAN.HK.2.7.0.1-01744-QCAHKSWPL_SILICONZ-1
[    8.842934] ath11k_pci 0000:01:00.0: swiotlb buffer is full (sz: 1048583 bytes), total 32768 (slots), used 992 (slots)
[    8.842947] ath11k_pci 0000:01:00.0: failed to set up tcl_comp ring (0) :-12
[    8.843087] ath11k_pci 0000:01:00.0: failed to init DP: -12
```

@6by9 you mentioned some changes between 6.6 and 6.12, does this mean dynamic swiotlb is currrently broken?

Thanks for your assistance.