raspberrypi/linux #6444: Latest updates break third party desktop integration

glanduin, 2024-10-28:
### Describe the bug

After the latest updates which are:

```
2024-10-28 20:07:19 upgrade libdrm-common:all 2.4.114-1 2.4.123-1~bpo12+1+rpt1
2024-10-28 20:07:20 upgrade libdrm2:arm64 2.4.114-1+b1 2.4.123-1~bpo12+1+rpt1
2024-10-28 20:07:20 upgrade libdrm-amdgpu1:arm64 2.4.114-1+b1 2.4.123-1~bpo12+1+rpt1
2024-10-28 20:07:21 upgrade libdrm-nouveau2:arm64 2.4.114-1+b1 2.4.123-1~bpo12+1+rpt1
2024-10-28 20:07:21 upgrade libdrm-radeon1:arm64 2.4.114-1+b1 2.4.123-1~bpo12+1+rpt1
2024-10-28 20:07:23 upgrade liblightdm-gobject-1-0:arm64 1.26.0-8+rpt3 1.26.0-8+rpt5
2024-10-28 20:07:24 upgrade libwayland-client0:arm64 1.22.0-2.1~bpo12+rpt1 1.23.0-1~bpo12+1+rpt1
2024-10-28 20:07:24 upgrade libwayland-cursor0:arm64 1.22.0-2.1~bpo12+rpt1 1.23.0-1~bpo12+1+rpt1
2024-10-28 20:07:25 upgrade libwayland-egl1:arm64 1.22.0-2.1~bpo12+rpt1 1.23.0-1~bpo12+1+rpt1
2024-10-28 20:07:26 upgrade libwayland-server0:arm64 1.22.0-2.1~bpo12+rpt1 1.23.0-1~bpo12+1+rpt1
2024-10-28 20:07:26 upgrade lightdm:arm64 1.26.0-8+rpt3 1.26.0-8+rpt5
2024-10-28 20:07:31 upgrade raspi-config:all 20240708 20241025
2024-10-28 20:07:31 upgrade raspberrypi-sys-mods:arm64 20240911 20240923
2024-10-28 20:07:31 upgrade rpi-imager:arm64 1.8.5+rpt1 1.9.1
2024-10-28 20:07:33 upgrade userconf-pi:all 0.9 0.10
```

Kde Plasma desktop does not start (in wayland) anymore. X11 is not affected but switching to wayland results in a black screen with a mouse cursor (which is immovable). System also becomes unresponsive and any shortcut to switch to a terminal does not respond. X11 works as expected. It is possible that other third party desktop environments are also affected from this regression.

### Steps to reproduce the behaviour

You have to be on a RPIOS with a KDE Plasma desktop on top. And select wayland as your desktop backend (which is the default). After the latest updates, a reboot results in a black screen with an immovable mouse cursor and an unresponsive system.

### Device (s)

Raspberry Pi 5

### System

Raspberry Pi reference 2024-03-15

2024/10/21 15:27:49 
Copyright (c) 2012 Broadcom
version 951e1cc9 (release) (embedded)

Linux raspberrypi 6.6.51+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux

### Logs

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x414fd0b1]
[    0.000000] Linux version 6.6.51+rpt-rpi-2712 (serge@raspberrypi.com) (gcc-12 (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08)
[    0.000000] KASLR enabled
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 5 Model B Rev 1.0
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000002000000, size 320 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x0000000002000000..0x0000000015ffffff (327680 KiB) map reusable linux,cma
[    0.000000] OF: reserved mem: 0x0000000000000000..0x000000000007ffff (512 KiB) nomap non-reusable atf@0
[    0.000000] OF: reserved mem: 0x000000003fd165e0..0x000000003fd165f9 (0 KiB) nomap non-reusable nvram@0
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x00000000ffffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000007ffff]
[    0.000000]   node   0: [mem 0x0000000000080000-0x000000003f7fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000000ffffffff]
[    0.000000] On node 0, zone DMA: 512 pages in unavailable ranges
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[... 484 lines cut ...]
[    6.437168] Bluetooth: hci0: BCM4345C0
[    6.437175] Bluetooth: hci0: BCM4345C0 (003.001.025) build 0000
[    6.444370] Bluetooth: hci0: BCM4345C0 'brcm/BCM4345C0.raspberrypi,5-model-b.hcd' Patch
[    6.660207] Adding 2097136k swap on /var/swap.  Priority:-2 extents:5 across:2269168k 
[    6.840874] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    6.840882] Bluetooth: BNEP filters: protocol multicast
[    6.840889] Bluetooth: BNEP socket layer initialized
[    7.180662] Bluetooth: hci0: BCM: features 0x2f
[    7.182186] Bluetooth: hci0: BCM43455 37.4MHz Raspberry Pi 3+-0190
[    7.182199] Bluetooth: hci0: BCM4345C0 (003.001.025) build 0382
[    7.182591] Bluetooth: hci0: BCM: Using default device address (43:45:c0:00:1f:ac)
[    7.217393] Bluetooth: MGMT ver 1.22
[    7.228532] NET: Registered PF_ALG protocol family
[    7.281145] Bluetooth: RFCOMM TTY layer initialized
[    7.281162] Bluetooth: RFCOMM socket layer initialized
[    7.281173] Bluetooth: RFCOMM ver 1.11
[    7.302519] Bluetooth: ISO socket layer initialized
[    7.366910] macb 1f00100000.ethernet eth0: PHY [1f00100000.ethernet-ffffffff:01] driver [Broadcom BCM54213PE] (irq=POLL)
[    7.366920] macb 1f00100000.ethernet eth0: configuring for phy/rgmii-id link mode
[    7.369585] pps pps0: new PPS source ptp0
[    7.369632] macb 1f00100000.ethernet: gem-ptp-timer ptp clock registered.
[    9.404098] macb 1f00100000.ethernet eth0: Link is Up - 100Mbps/Full - flow control off
[    9.450377] warning: `iwconfig' uses wireless extensions which will stop working for Wi-Fi 7 hardware; use nl80211
[   10.691263] systemd[1088]: memfd_create() called without MFD_EXEC or MFD_NOEXEC_SEAL set
[   14.626906] EXT4-fs (sda3): mounted filesystem 351f3d66-8e63-4157-a490-379cfdc8ba47 r/w with ordered data mode. Quota mode: none.

simpz, 2024-10-29:
I have this same issue, since upgrading to the latest updates. In my case plasma mobile is completely broken now.

pelwell, 2024-10-29:
There was an update yesterday that ensured labwc is installed. Another patch should be rolling out later today to fix the Network Manager GUI glitch.

simpz, 2024-10-29:
@pelwell Is this what is causing KDE Plasma to fail to start too or is this unrelated?

pelwell, 2024-10-29:
If our normal desktop starts after the update but KDE Plasma doesn't then it's probably unrelated.

pelwell, 2024-10-29:
You might get some more useful output from `journalctl --user -b`.

simpz, 2024-10-29:
Ah that's good, Xwayland seems to core dump 
```

Oct 28 22:44:15 piclock kwin_wayland_wrapper[1402]: wl_global_create: implemented version for 'wl_shm' higher than interface version (2 > 1)
Oct 28 22:44:17 piclock kwin_wayland_wrapper[1402]: kf.globalaccel.kglobalacceld: No desktop file found for service  "org.kde.dolphin.desktop"
Oct 28 22:44:17 piclock kwin_wayland_wrapper[1402]: OpenGL vendor string:                   Broadcom
Oct 28 22:44:17 piclock kwin_wayland_wrapper[1402]: OpenGL renderer string:                 V3D 7.1
Oct 28 22:44:17 piclock kwin_wayland_wrapper[1402]: OpenGL version string:                  3.1 (Core Profile) Mesa 23.2.1-1~bpo12+rpt3
Oct 28 22:44:17 piclock kwin_wayland_wrapper[1402]: OpenGL shading language version string: 1.40
Oct 28 22:44:17 piclock kwin_wayland_wrapper[1402]: Driver:                                 VideoCore 3D
Oct 28 22:44:17 piclock kwin_wayland_wrapper[1402]: GPU class:                              Unknown
Oct 28 22:44:17 piclock kwin_wayland_wrapper[1402]: OpenGL version:                         3.1
Oct 28 22:44:17 piclock kwin_wayland_wrapper[1402]: GLSL version:                           1.40
Oct 28 22:44:17 piclock kwin_wayland_wrapper[1402]: Mesa version:                           23.2.1
Oct 28 22:44:17 piclock kwin_wayland_wrapper[1402]: Linux kernel version:                   6.6.51
Oct 28 22:44:17 piclock kwin_wayland_wrapper[1402]: Requires strict binding:                no
Oct 28 22:44:17 piclock kwin_wayland_wrapper[1402]: GLSL shaders:                           yes
Oct 28 22:44:17 piclock kwin_wayland_wrapper[1402]: Texture NPOT support:                   yes
Oct 28 22:44:17 piclock kwin_wayland_wrapper[1402]: Virtual Machine:                        no
Oct 28 22:44:18 piclock kwin_wayland_wrapper[1402]: kwin_xkbcommon: XKB: inet:334:58: unrecognized keysym "XF86EmojiPicker"
Oct 28 22:44:18 piclock kwin_wayland_wrapper[1402]: kwin_wayland_drm: Failed to create gamma blob! Invalid argument
Oct 28 22:44:19 piclock kwin_wayland_wrapper[1440]: (EE)
Oct 28 22:44:19 piclock kwin_wayland_wrapper[1440]: (EE) Backtrace:
Oct 28 22:44:19 piclock kwin_wayland_wrapper[1440]: (EE) 0: /usr/bin/Xwayland (0x555633a30000+0x17a3d8) [0x555633baa3d8]
Oct 28 22:44:19 piclock kwin_wayland_wrapper[1440]: (EE) 1: /usr/bin/Xwayland (0x555633a30000+0x17e144) [0x555633bae144]
Oct 28 22:44:19 piclock kwin_wayland_wrapper[1440]: (EE) 2: linux-vdso.so.1 (__kernel_rt_sigreturn+0x0) [0x7ffecb7d87b0]
Oct 28 22:44:19 piclock kwin_wayland_wrapper[1440]: (EE) 3: /lib/aarch64-linux-gnu/libwayland-client.so.0 (wl_proxy_get_version+0x4) [0x7ffecb637894]
Oct 28 22:44:19 piclock kwin_wayland_wrapper[1440]: (EE) 4: /usr/bin/Xwayland (0x555633a30000+0x41ce8) [0x555633a71ce8]
Oct 28 22:44:19 piclock kwin_wayland_wrapper[1440]: (EE) 5: /usr/bin/Xwayland (0x555633a30000+0x3ecc0) [0x555633a6ecc0]
Oct 28 22:44:19 piclock kwin_wayland_wrapper[1440]: (EE) 6: /usr/bin/Xwayland (0x555633a30000+0x1077a8) [0x555633b377a8]
Oct 28 22:44:19 piclock kwin_wayland_wrapper[1440]: (EE) 7: /usr/bin/Xwayland (0x555633a30000+0xb2248) [0x555633ae2248]
Oct 28 22:44:19 piclock kwin_wayland_wrapper[1440]: (EE) 8: /usr/bin/Xwayland (0x555633a30000+0x9da2c) [0x555633acda2c]
Oct 28 22:44:19 piclock kwin_wayland_wrapper[1440]: (EE) 9: /usr/bin/Xwayland (0x555633a30000+0x9e9dc) [0x555633ace9dc]
Oct 28 22:44:19 piclock kwin_wayland_wrapper[1440]: (EE) 10: /usr/bin/Xwayland (0x555633a30000+0xab910) [0x555633adb910]
Oct 28 22:44:19 piclock kwin_wayland_wrapper[1440]: (EE) 11: /lib/aarch64-linux-gnu/libc.so.6 (0x7ffecaf30000+0x27740) [0x7ffecaf57740]
Oct 28 22:44:19 piclock kwin_wayland_wrapper[1440]: (EE) 12: /lib/aarch64-linux-gnu/libc.so.6 (__libc_start_main+0x98) [0x7ffecaf57818]
Oct 28 22:44:19 piclock kwin_wayland_wrapper[1440]: (EE) 13: /usr/bin/Xwayland (0x555633a30000+0x389f0) [0x555633a689f0]
Oct 28 22:44:19 piclock kwin_wayland_wrapper[1440]: (EE)
Oct 28 22:44:19 piclock kwin_wayland_wrapper[1440]: (EE) Segmentation fault at address 0x40
Oct 28 22:44:19 piclock kwin_wayland_wrapper[1440]: (EE)
Oct 28 22:44:19 piclock kwin_wayland_wrapper[1440]: Fatal server error:
Oct 28 22:44:19 piclock kwin_wayland_wrapper[1440]: (EE) Caught signal 11 (Segmentation fault). Server aborting
```

pelwell, 2024-10-29:
@spl237 FYI.

yutayu, 2024-10-29:
> There was an update yesterday that ensured labwc is installed. Another patch should be rolling out later today to fix the Network Manager GUI glitch.

Now update has come and fixed.
rpi4 32bit
Thanks.

simpz, 2024-10-29:
Looks like this?
[https://www.linuxquestions.org/questions/showthread.php?p=6505230](url)
Looks similar a move from  wayland-1.22.0 to wayland-1.23.0? But it's far from clear what was done to resolve this from the thread.

pelwell, 2024-10-29:
> Looks like this?

Yes, I'd say so.

> it's far from clear what was done to resolve this from the thread

Just two reports of rebuilding "kwin" fixing the issue.

spl237, 2024-10-29:
I think the problem here is expecting KDE to work with labwc.

KDE uses its own compositor, KWin, which already uses Wayland. If you install our updates to labwc etc, this should make no difference to KDE, but if you then choose to switch to labwc, what you are doing is changing the compositor from KWin to labwc. I suspect that KDE requires various features which are unique to KWin and which do not exist in labwc, such as specific autostarts and the like.

I don't think it is likely that KDE Plasma will run on top of labwc; I think if you are running KDE, you need not to switch to labwc. But installing the latest updates shouldn't affect KDE as long as you don't make the switch.

If it is the case that just installing the latest updates - without switching the compositor - breaks KDE, then we can investigate.

simpz, 2024-10-29:
@pelwell How can I revert to wayland-1.22.0 ?
@spl237 I haven't changed any config I'm aware of recently, simply applied the update from 28th Oct and my KDE plasma mobile session has stopped working.

spl237, 2024-10-29:
OK, in that case (and looking at your core dump), it may be that an incompatible version of Xwayland is being installed by labwc.

The short-term fix for that would be to add the relevant Conflicts: line in the control file of one of the affected packages, to prevent labwc being installed. 

It would be interesting to know what the version number of xwayland is on an image which is successfully running KDE, and to see if rolling back to that version is sufficient to fix it.

spl237, 2024-10-29:
It might be worth trying

sudo apt install --reinstall xwayland=2:22.1.9-1

Does that fix things?

pelwell, 2024-10-29:
The linked article suggests that wayland-1.22.0 is good, while wayland-1.23.0 is bad, and that seems to fit with the upgrade log above:
```
...
2024-10-28 20:07:24 upgrade libwayland-client0:arm64 1.22.0-2.1~bpo12+rpt1 1.23.0-1~bpo12+1+rpt1
2024-10-28 20:07:24 upgrade libwayland-cursor0:arm64 1.22.0-2.1~bpo12+rpt1 1.23.0-1~bpo12+1+rpt1
2024-10-28 20:07:25 upgrade libwayland-egl1:arm64 1.22.0-2.1~bpo12+rpt1 1.23.0-1~bpo12+1+rpt1
2024-10-28 20:07:26 upgrade libwayland-server0:arm64 1.22.0-2.1~bpo12+rpt1 1.23.0-1~bpo12+1+rpt1
...
```

simpz, 2024-10-29:
The XWayland downgrade still produces the crash on XWayland in the log.
I'll happily test a wayland downgrade if you can give a suitable method.

spl237, 2024-10-29:
In that case, I'm not sure what we can do about this. We need the newer version of wayland as it is a dependency of our version of wlroots - and we are going to move to an even newer version soon, which will presumably require an even newer version of wayland. I don't think we can have two different versions installed simultaneously, so it looks as if installing labwc over KDE will inevitably break it. We may be in an either-or situation here, and that is going to lead to problems with updates etc - even if someone installs KDE over a clean image, and hence downgrades wayland, a full-upgrade is likely to lead to it being updated again.

I can't think of any automatic way to handle this - a KDE user could set an apt pin on the relevant versions to prevent updates, but that's about it, and even that's not ideal, as it will make you vulnerable to security flaws etc.

glanduin, 2024-10-29:
What about letting in plasma 6 into the repositories so we can update to 6 which is claimed to not have booting issues? By the way, i built my setup on top of the "lite" version of rpios. So it shouldn't even bother with desktop dependencies and updates. I would expect it to just let whatever desktop is installed deal with those.

simpz, 2024-10-29:
[https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1077016](url)
Is this upstream Debian bug report?
Couldn't the fix of rebuild "Plasma 5" from the linuxquestions forum work (whatever that fix ACTUALLY is, I can't see), I guess I'm saying someone seemed to make this work but doesn't seem to say how.

pelwell, 2024-10-29:
(Careful with those links, @simpz - the URL is ending up as the visible text rather than the link, which is "url". I've patched them up.)

XECDesign, 2024-10-29:
I have a workaround that will probably brick your system, so don't do this if you don't have a backup:

/etc/apt/preferences.d/debian_wayland.pref:
```
Package: src:wayland
Pin: release o=Raspberry Pi Foundation
Pin-Priority: -1
```

That will prevent apt from installing or considering the newer wayland packages. If you already have them installed, you will need to remove and reinstall them or downgrade:

```
dpkg -l | awk '$1 == "ii" && $3 == "1.23.0-1~bpo12+1+rpt1" { print $2"=1.21.0-1"; }' | xargs sudo apt-get -y  --allow-downgrades --mark-auto install
```

Run without the `-y` first to check that it's not going to remove half of your system with it.

simpz, 2024-10-29:
I'm trying
`apt install libwayland-client0=1.21.0-1 libwayland-cursor0=1.21.0-1 libwayland-server0=1.21.0-1`
to get me going for the moment. Though it responds with a slightly alarming number of packages that are no longer required. 

But more processes are now up on the system and they Xwayland  crash is gone. I'm not beside the pi to test if this has resolved.

simpz, 2024-10-29:
Confirmed that the
`apt install libwayland-client0=1.21.0-1 libwayland-cursor0=1.21.0-1 libwayland-server0=1.21.0-1`
has restored KDE plasma back to working. 

But I guess you'll need to decide what the longer term fix should be with this, whether it's plasma 6 or something else.

norambna, 2024-10-31:
> Confirmed that the `apt install libwayland-client0=1.21.0-1 libwayland-cursor0=1.21.0-1 libwayland-server0=1.21.0-1` has restored KDE plasma back to working.
> 
> But I guess you'll need to decide what the longer term fix should be with this, whether it's plasma 6 or something else.

Thanks! I'm a Raspberry Pi OS Lite + Plasma user and this solves the Wayland issue for now.

TheOneDoc, 2024-11-01:
I have the same issue un my raspberry pi 400 and simpz fix worked for me as well

```
doesntmatter@raspberrypi400:~ $ sudo apt install libwayland-client0=1.21.0-1 libwayland-cursor0=1.21.0-1 libwayland-server0=1.21.
0-1
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages will be DOWNGRADED:
  libwayland-client0 libwayland-cursor0 libwayland-server0
0 upgraded, 0 newly installed, 3 downgraded, 0 to remove and 0 not upgraded.
Need to get 76.7 kB of archives.
After this operation, 9,216 B of additional disk space will be used.
Do you want to continue? [Y/n]
Get:1 http://deb.debian.org/debian bookworm/main arm64 libwayland-client0 arm64 1.21.0-1 [27.6 kB]
Get:2 http://deb.debian.org/debian bookworm/main arm64 libwayland-cursor0 arm64 1.21.0-1 [14.1 kB]
Get:3 http://deb.debian.org/debian bookworm/main arm64 libwayland-server0 arm64 1.21.0-1 [34.9 kB]
Fetched 76.7 kB in 0s (1,050 kB/s)
dpkg: warning: downgrading libwayland-client0:arm64 from 1.23.0-1~bpo12+1+rpt1 to 1.21.0-1
(Reading database ... 209821 files and directories currently installed.)
Preparing to unpack .../libwayland-client0_1.21.0-1_arm64.deb ...
Unpacking libwayland-client0:arm64 (1.21.0-1) over (1.23.0-1~bpo12+1+rpt1) ...
dpkg: warning: downgrading libwayland-cursor0:arm64 from 1.23.0-1~bpo12+1+rpt1 to 1.21.0-1
Preparing to unpack .../libwayland-cursor0_1.21.0-1_arm64.deb ...
Unpacking libwayland-cursor0:arm64 (1.21.0-1) over (1.23.0-1~bpo12+1+rpt1) ...
dpkg: warning: downgrading libwayland-server0:arm64 from 1.23.0-1~bpo12+1+rpt1 to 1.21.0-1
Preparing to unpack .../libwayland-server0_1.21.0-1_arm64.deb ...
Unpacking libwayland-server0:arm64 (1.21.0-1) over (1.23.0-1~bpo12+1+rpt1) ...
Setting up libwayland-server0:arm64 (1.21.0-1) ...
Setting up libwayland-client0:arm64 (1.21.0-1) ...
Setting up libwayland-cursor0:arm64 (1.21.0-1) ...
Processing triggers for libc-bin (2.36-9+rpt2+deb12u8) ...
```

simpz, 2024-11-04:
Just a question. Is there any hope that this will be fixed i.e. KDE Plasma 5 before a new release of Raspberry Pi OS (which would be a new Debian so Plasma 6 by default you'd imagine)? Or an out-of-band Plasma 6 release for Raspberry Pi OS?

TheOneDoc, 2024-11-05:
Can confirm it's working for me
```
Linux raspberrypi400 6.6.58-v8+ #1809 SMP PREEMPT Wed Oct 23 11:53:53 BST 2024 aarch64

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Fri Nov  1 14:06:23 2024 from 192.168.178.30
XXX@raspberrypi400:~ $ sudo vi /etc/apt/sources.list.d/raspi.list
XXX@raspberrypi400:~ $ sudo apt update
Hit:1 http://deb.debian.org/debian bookworm InRelease
Get:2 http://deb.debian.org/debian-security bookworm-security InRelease [48.0 kB]
Get:3 http://deb.debian.org/debian bookworm-updates InRelease [55.4 kB]
Get:4 http://archive.raspberrypi.com/debian bookworm InRelease [39.2 kB]
Get:5 http://deb.debian.org/debian-security bookworm-security/main Sources [126 kB]
Get:6 http://deb.debian.org/debian-security bookworm-security/main arm64 Packages [187 kB]
Get:7 http://deb.debian.org/debian-security bookworm-security/main armhf Packages [187 kB]
Get:8 http://archive.raspberrypi.com/debian bookworm/main Sources [120 kB]
Get:9 http://archive.raspberrypi.com/debian bookworm/main arm64 Packages [503 kB]
Get:10 http://archive.raspberrypi.com/debian bookworm/main armhf Packages [533 kB]
Get:11 http://archive.raspberrypi.com/debian bookworm/untested arm64 Packages [5,959 B]
Get:12 http://archive.raspberrypi.com/debian bookworm/untested armhf Packages [5,960 B]
Fetched 1,811 kB in 1s (1,493 kB/s)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
20 packages can be upgraded. Run 'apt list --upgradable' to see them.
XXX@raspberrypi400:~ $ sudo apt full-upgrade
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  kwin-common kwin-data kwin-wayland kwin-x11 libkwineffects14 libkwinglutils14 libvlc-bin libvlc5 libvlccore9 libwayland-client0 libwayland-cursor0 libwayland-server0 raspi-config vlc-data vlc-plugin-base
  vlc-plugin-video-output xserver-common xserver-xorg-core xserver-xorg-legacy zlib1g
20 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 28.6 MB of archives.
After this operation, 385 kB of additional disk space will be used.
Do you want to continue? [Y/n]
Get:1 http://archive.raspberrypi.com/debian bookworm/main arm64 zlib1g arm64 1:1.2.13.dfsg-1+rpt1 [94.3 kB]
Get:2 http://archive.raspberrypi.com/debian bookworm/untested arm64 kwin-x11 arm64 4:5.27.5-3+rpt1 [484 kB]
Get:3 http://archive.raspberrypi.com/debian bookworm/untested arm64 kwin-wayland arm64 4:5.27.5-3+rpt1 [671 kB]
Get:4 http://archive.raspberrypi.com/debian bookworm/untested arm64 kwin-common arm64 4:5.27.5-3+rpt1 [3,954 kB]
Get:5 http://archive.raspberrypi.com/debian bookworm/untested arm64 libkwineffects14 arm64 4:5.27.5-3+rpt1 [128 kB]
Get:6 http://archive.raspberrypi.com/debian bookworm/untested arm64 libkwinglutils14 arm64 4:5.27.5-3+rpt1 [74.6 kB]
Get:7 http://archive.raspberrypi.com/debian bookworm/untested arm64 kwin-data all 4:5.27.5-3+rpt1 [7,531 kB]
Get:8 http://archive.raspberrypi.com/debian bookworm/main arm64 libwayland-client0 arm64 1.23.0-1~bpo12+1+rpt1 [31.4 kB]
Get:9 http://archive.raspberrypi.com/debian bookworm/main arm64 libwayland-server0 arm64 1.23.0-1~bpo12+1+rpt1 [40.9 kB]
Get:10 http://archive.raspberrypi.com/debian bookworm/main arm64 libvlccore9 arm64 1:3.0.21-0+rpt3+deb12u1 [596 kB]
Get:11 http://archive.raspberrypi.com/debian bookworm/main arm64 libvlc5 arm64 1:3.0.21-0+rpt3+deb12u1 [154 kB]
Get:12 http://archive.raspberrypi.com/debian bookworm/main arm64 libvlc-bin arm64 1:3.0.21-0+rpt3+deb12u1 [94.6 kB]
Get:13 http://archive.raspberrypi.com/debian bookworm/main arm64 libwayland-cursor0 arm64 1.23.0-1~bpo12+1+rpt1 [13.0 kB]
Get:14 http://archive.raspberrypi.com/debian bookworm/main arm64 raspi-config all 20241105 [36.0 kB]
Get:15 http://archive.raspberrypi.com/debian bookworm/main arm64 vlc-plugin-base arm64 1:3.0.21-0+rpt3+deb12u1 [5,141 kB]
Get:16 http://archive.raspberrypi.com/debian bookworm/main arm64 vlc-data all 1:3.0.21-0+rpt3+deb12u1 [473 kB]
Get:17 http://archive.raspberrypi.com/debian bookworm/main arm64 vlc-plugin-video-output arm64 1:3.0.21-0+rpt3+deb12u1 [290 kB]
Get:18 http://archive.raspberrypi.com/debian bookworm/main arm64 xserver-common all 2:21.1.7-3+rpt3+deb12u8 [2,387 kB]
Get:19 http://archive.raspberrypi.com/debian bookworm/main arm64 xserver-xorg-legacy arm64 2:21.1.7-3+rpt3+deb12u8 [2,394 kB]
Get:20 http://archive.raspberrypi.com/debian bookworm/main arm64 xserver-xorg-core arm64 2:21.1.7-3+rpt3+deb12u8 [4,041 kB]
Fetched 28.6 MB in 1s (20.5 MB/s)
Reading changelogs... Done
Preconfiguring packages ...
setting xserver-xorg-legacy/xwrapper/allowed_users from configuration file
(Reading database ... 209821 files and directories currently installed.)
Preparing to unpack .../zlib1g_1%3a1.2.13.dfsg-1+rpt1_arm64.deb ...
Unpacking zlib1g:arm64 (1:1.2.13.dfsg-1+rpt1) over (1:1.2.13.dfsg-1) ...
Setting up zlib1g:arm64 (1:1.2.13.dfsg-1+rpt1) ...
[... 48 lines cut ...]
Setting up vlc-data (1:3.0.21-0+rpt3+deb12u1) ...
Setting up libwayland-client0:arm64 (1.23.0-1~bpo12+1+rpt1) ...
Setting up xserver-xorg-core (2:21.1.7-3+rpt3+deb12u8) ...
Setting up libvlc5:arm64 (1:3.0.21-0+rpt3+deb12u1) ...
Setting up kwin-common (4:5.27.5-3+rpt1) ...
[... cut ...]
Setting up kwin-x11 (4:5.27.5-3+rpt1) ...
Processing triggers for desktop-file-utils (0.26-1) ...
Processing triggers for hicolor-icon-theme (0.17-2) ...
Processing triggers for libc-bin (2.36-9+rpt2+deb12u8) ...
Processing triggers for man-db (2.11.2-2) ...
Processing triggers for mailcap (3.70+nmu1) ...
Processing triggers for libvlc-bin:arm64 (1:3.0.21-0+rpt3+deb12u1) ...
XXX@raspberrypi400:~ $ sudo reboot now

Broadcast message from root@raspberrypi400 on pts/1 (Tue 2024-11-05 16:31:02 CET):

The system will reboot now!

```

simpz, 2024-11-06:
I can confirm your packages in untested work for me too with KDE Plasma.

XECDesign, 2024-11-07:
I've move it to 'main' now. I'd recommend removing the 'untested' component from your sources.

Thanks for testing!

norambna, 2024-11-07:
> I've move it to 'main' now

Great! The latest versions of `libwayland-client0 libwayland-cursor0 libwayland-server0` work fine after this update.

simpz, 2024-11-08:
Thanks for your extra effort to keep Plasma working.