raspberrypi/linux #6445: RPi4 & RPi5 HDMI Video Broken

rtomasa, 2024-10-29:
### Describe the bug

Hi team,

I'm the developer of an upcoming gaming front end designed primarily for playing classic games on CRT TVs.
The system is based on the Raspberry Pi OS 64bit Lite created with Pi-Gen (official Raspberry tool) and it uses GBM+DRM for all video operations.
The focus on CRTs led me to initially use the DPI output, but following the recent discontinuation of interlaced video support in DPI, I re-engineered my frontend to use an HDMI-to-VGA DAC. My code is able to inject any kind of video mode on the fly, so that the DAC generates the proper signal to the CRT TV (BTW it uses super-resolutions). This adjustment allowed for compatibility across all Raspberry Pi models (Pi3, Pi4, and Pi5) and supported both progressive and interlaced modes—except for the PAL 50Hz interlaced mode, which remained problematic.

While researching the PAL 50Hz interlaced issue, I noticed a potential fix in the latest Raspberry Pi OS released today (October 28th) (https://github.com/raspberrypi/linux/pull/6282) and updated my system to test it. Unfortunately, after this update, a major regression occurred, rendering the system almost unusable. Below is a comparison of the previous and current kernel versions along with the observed outcomes across models and video modes:

| Model | Kernel Version | Progressive-60 | Progressive-50 | Interlaced-60 | Interlaced-50 |
|-------|----------------|----------------|----------------|---------------|---------------|
| Pi4   | 6.6.31         | OK             | OK             | OK            | --            |
| Pi5   | 6.6.31         | OK             | OK             | OK            | --            |
| Pi4   | 6.6.58         | KO             | KO             | KO            | --            |
| Pi5   | 6.6.58         | OK             | OK             | --            | --            |

Legend:  
- **OK** = Proper image displayed  
- **KO** = Image out of sync (bad display)  
- **--** = Mode not supported (no timing change)

Could you help me understand the cause of this regression and possibly guide me on how to resolve it?

Best regards,

### Steps to reproduce the behaviour

Since it is a frontend that it is on development I cannot provide you any repro steps but probably you could whether to set a custom dummy EDID via cmdline.txt to simulate things like 1920x240@60, 1920x288@50, 1920x480@60, 1920x576@50

### Device (s)

Raspberry Pi 5

### System

```
RePlay OS 2024-06-16
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, a63eb86d7b90633e69737f6e0ee818c978771dc1, stage2
```
```
vcgencmd version
Oct 17 2024 11:34:50 
Copyright (c) 2012 Broadcom
version b580e2acde306434ab07a913745a21451643ff55 (clean) (release) (start)
```
```
Linux replay 6.6.58-v8+ #1809 SMP PREEMPT Wed Oct 23 11:53:53 BST 2024 aarch64 GNU/Linux
```

6by9, 2024-10-29:
Please provide the full mode configuration that you are requesting - use kmsprint.

There was a fix which is probably in the same time frame where sync polarities weren't being implemented correctly, but now they are.
We did verify all the hdmi output properties on an analyser after that change, so it should be as requested.

rtomasa, 2024-10-29:
OK, some good news: you were right. After inverting the HSync and VSync, I managed to return to the previous state where both the Pi4 and Pi5 can now display images in both progressive and interlaced modes. However, I still can’t get interlaced 50Hz to work properly (DRM does not accept/set the same).

Since this mode isn’t being accepted, the `kmsprint` displays some default information:

```
Connector 0 (32) HDMI-A-1 (connected)
  Encoder 0 (31) TMDS
    Crtc 3 (100) 1024x768@60.00 65.000 1024/24/136/160/- 768/3/6/29/- 60 (60.00) P|D 
      Plane 3 (89) fb-id: 368 (crtcs: 3) 0,0 1024x768 -> 0,0 1024x768 (XR24 AR24 AB24 XB24 RG16 BG16 AR15 XR15 RG24 BG24 YU16 YV16 YU24 YV24 YU12 YV12 NV12 NV21 NV16 NV61 P030 XR30 AR30 AB30 XB30 RGB8 BGR8 XR12 AR12 XB12 AB12 BX12 BA12 RX12 RA12)
        FB 368 1024x768 RG16
Connector 1 (42) HDMI-A-2 (disconnected)
  Encoder 1 (41) TMDS
```

Additionally, since I’m using the Switchres library to calculate display modes, here is the information for the specific mode I’m attempting to set:

```
Switchres: Monitor range 15625.00-15750.00,49.50-65.00,2.000,4.700,8.000,0.064,0.192,1.024,0,0,192,288,448,576
Switchres(v2.2.0) add display[0]
Switchres: display[0] options: monitor[generic_15] generation[on]
Switchres: Monitor range 15625.00-15750.00,49.50-65.00,2.000,4.700,8.000,0.064,0.192,1.024,0,0,192,288,448,576
Switchres: Calculating best video mode for 1920x576@50.000000i orientation: normal

Switchres: (   0)x(   0)_(0=0.000000Hz)
   rng(0): 1920 x 576_50.000000i 15.675000 [integ] scale(1.000, 1.000, 1.000) diff(0.000, 0.000, 0.000)

Switchres: normal (1920x576@50.000000)->(1920x576@50.000000)
   rng(0): 1920 x 576_50.000000i 15.675000 [integ] scale(1.000, 1.000, 1.000) diff(0.000, 0.000, 0.000)
Switchres: Modeline "1920x576_50i 15.675000KHz 50.000000Hz" 39.109125 1920 1998 2182 2495 576 583 589 627 interlace  -hsync -vsync
sr_add_mode: got mode 1920x576@50.000000 type(20000009)
Switchres: success adding mode dummy timing "1920x576_50i 15.675000KHz 50.000000Hz" 39.109125 1920 1998 2182 2495 576 583 589 627 interlace  -hsync -vsync
Switchres: success deleting mode dummy timing "1920x576_50i 15.675000KHz 50.000000Hz" 39.109125 1920 1998 2182 2495 576 583 589 627 interlace  -hsync -vsync
```

rtomasa, 2024-11-02:
I apologize for any confusion caused by my last issue regarding PAL timings.
After several hours of testing, I discovered that the problem was on my end—the GBM buffer I was using wasn’t large enough to support the 576 resolution. Once I increased the buffer size, everything worked perfectly.

Thank you so much for your hard work! The Raspberry Pi and its VideoCore GPU now seem to be the best device out there for CRT enthusiasts.

closed 2024-11-02 by rtomasa