raspberrypi/linux #6446: Feature Request:  Add support for TCS3725 ahd VMEML6040 RGB Color Sensors

Jibun-no-Kage, 2024-10-29:
Feature Request:  Add support for TCS34725 ahd VMEML6040 RGB Color Sensors as device tree overlays.

Example of TCS34725 RGB Color Sensor...
https://www.adafruit.com/product/1334

VMEML6040 RGB Color Sensor...
https://www.taydaelectronics.com/veml6040-color-sensor-rgbw-i2c-module.html

Jibun-no-Kage, 2024-10-29:
Oh, right, TCS34725 is 0x29...  VMEML6040 is 0x10...

The use case is that I am using these sensors to read a 'tally' light on a remote floating head camera, at the end of 25 or so foot  jib or boom crane... which is beyond visibility to anyone at ground level.  A micro controlller, i.e. in this case Pi Zero 2W forwards this color state to a production broadcast monitor at ground level, that as a custom GPIO interface,  I have not decided if I will use solid-state relays, or NPN transistors, or even mosfets to do the actual GPIO pin state changes between the Pi and monitor as yet.  but since i have the solid state relays, likely use those for the initial testing.

Sensor->i2c->Pi->Evaluate RGB levels->Pi GPIO->Trigger applicable relay.  The broadcast monitor GPIO interface, Pin to GND, i.e. Low 'On', Pin High 'Off', for the respective color of tally status on-screen of monitor.  So I just close the relay to short pin to GND for the given pin, i.e. color, Red, Green, or None.

Typical broadcast switchers and cameras... Red is on-air, Green is stand-by or preview.  Or none... if camera is not pending preview or on-air.

Jibun-no-Kage, 2024-10-29:
You know... I could have sworn I posted this request to the 'firmware' section... Sorry about that.  I guess with all the page jumping I was doing... I did not  realize I was in the wrong section when I made the request.  Thanks for transferring it to the right section/queue.

pelwell, 2024-10-29:
See https://github.com/raspberrypi/linux/pull/6448. Wait about 40 minutes for the autobuilds to complete, then run `sudo rpi-update pulls/6448` to install a trial kernel, being careful to back up any important data first.

Jibun-no-Kage, 2024-10-29:
Yup... understood and well do.  Just a side question... ball park when this will go main stream?  Just curious.

pelwell, 2024-10-29:
I missed a few back-port commits - it should be good now, so the build will be available to try by the morning.

Stable kernels (and hence apt packages) are usually updated roughly every three months. There is likely to be at least one more 6.6 release before we move to the next LTS kernel - probably 6.12.

pelwell, 2024-10-30:
Let me know when you've tested #6448 and I'll merge it (if it works...)

Jibun-no-Kage, 2025-01-12:
Did you finally merge it?  I completely forgot the update the comments here.

pelwell, 2025-01-12:
I was waiting for feedback that never came. It's rebased now, so if you wait another ~15 minutes you'll be able to `sudo rpi-update pulls/6448` to test.

Jibun-no-Kage, 2025-01-14:
Cool.... It is late now, but late tomorrow I should have some feed back to post.  Thanks again.