raspberrypi/linux #6453: Problem mount HardDisk with LibreELEC

Aloyah1, 2024-11-03:
### Describe the bug

Hi,

I'm new to LibreELEC.

I was on a RPi 3B OSMC, then today I bought a RPI 5 and installed LibreELEC.

One thing I've noticed on startup is that my hard disk doesn't mount (even when unplugged and replugged). My USB flash drive, which does mount, takes a long time to mount.

My hard disk is powered by the HUB power supply, then I use a HUB to connect it to the RPi 5.

HD > HUB (PS) > RPi 5, only the microsd is powered by the RPi 5.

when I run a check on my Windows it detects errors that it corrects, and then the disk goes up it goes up, but if I reboot and remove then put it back again.

I enter the lsblk with putty to see, but nothing.

### Steps to reproduce the behaviour

start normaly

### Device (s)

Raspberry Pi 5

### System

LibreELEC

popcornmix, 2024-11-05:
You'll need to post output of dmesg after getting the failure to detect drive.

Aloyah1, 2024-11-05:
> You'll need to post output of dmesg after getting the failure to detect drive.

Yes, but I think I've understood libreelec doesn't detect ntfs disks very well.
https://paste.kodi.tv/ilosipuduz.kodi

BugfinderPro135, 2024-11-12:
Yes I saw many comments about this problem in the LibreElec forum as I've come across this problem myself. 

Once a drive gets unmounted (ntfs), it can't be seen again even if rebooted etc. If the drive is plugged into a Mac for instance, and a file is added or edited, then plugging the drive back into the pi the drive gets mounted again.

It's a nasty bug to get introduced into libreelec - and it still not being fixed after so much time - a few years. And yes it's still in Libreelec 12.1.

Boy do I wish Kodi 18.9 could just have python upgraded and nothing else - it worked so well with pi2 and pi3 - but no longer...

Aloyah1, 2024-11-12:
> Yes I saw many comments about this problem in the LibreElec forum as I've come across this problem myself.
> 
> Once a drive gets unmounted (ntfs), it can't be seen again even if rebooted etc. If the drive is plugged into a Mac for instance, and a file is added or edited, then plugging the drive back into the pi the drive gets mounted again.
> 
> It's a nasty bug to get introduced into libreelec - and it still not being fixed after so much time - a few years. And yes it's still in Libreelec 12.1.
> 
> Boy do I wish Kodi 18.9 could just have python upgraded and nothing else - it worked so well with pi2 and pi3 - but no longer... Reliability and usability have gone downhill fast since V18.9 - this bug, amongst many others, weren't in this version...

I was able to solve the problem with the nightly version, they have added the ntfs3g addon which allows you to mount and dismount without any problem.

BugfinderPro135, 2024-11-13:
Is there a link for this addon?

Aloyah1, 2024-11-13:
> Is there a link for this addon/update?
> 
> Let's hope an urgent update to LibreElec (12.2?) is released soon so we can eventually put this long standing problem behind us...

I don't think so, we'll have to wait until the end of the year for the next upgrade.

popcornmix, 2024-11-13:
You have many of these messages:
```
Nov 05 16:48:51.451438 LibreELEC kernel: usb usb2-port1: over-current change #1
```
This is a problem with power being supplied over USB.
It may be worth trying with `usb_max_current_enable=1` in config.txt.

If that doesn't help, it may be your powered hub cannot provide the power needed by the hub, or is otherwise behaving badly (e.g. a hub that back powers the Pi can provoke issues like this).

Aloyah1, 2024-11-13:
> You have many of these messages:
> 
> ```
> Nov 05 16:48:51.451438 LibreELEC kernel: usb usb2-port1: over-current change #1
> ```
> 
> This is a problem with power being supplied over USB. It may be worth trying with `usb_max_current_enable=1` in config.txt.
> 
> If that doesn't help, it may be your powered hub cannot provide the power needed by the drive, or is otherwise behaving badly (e.g. a hub that back powers the Pi can provoke issues like this).

I answered above, the nightly version solved the problem. I also changed power supply

closed 2024-11-13 by popcornmix