raspberrypi/linux #6460: dwc_otg: DEVICE:006 : update_urb_state_xfer_comp:750:trimming xfer length

ss8688, 2024-11-09:
### Describe the bug

I recently added a USB network adapter to my Raspberry Pi because the built-in network card, which lacks an external antenna, had poor signal reception. After doing this, my USB SSD started frequently disconnecting. My system boots from this SSD, and the dmesg logs show the following error messages:

`dwc_otg: DEVICE:006 : update_urb_state_xfer_comp:750:trimming xfer length`
`usb 1-1.3: reset high-speed USB device number 6 using dwc_otg`
Additionally, there are other errors like `brcmfmac: brcmf_set_channel: set chanspec ****** fail, reason -52`, and I’m unsure if they’re related to the USB SSD disconnection issue. By the way, I also have a Klipper 3D printer controller connected to this Raspberry Pi, so there are a total of three USB devices in use.

### Steps to reproduce the behaviour

The issue seems more reproducible when there are simultaneous hard drive read/write operations + WiFi data transfers, for example when running `apt upgrade`

### Device (s)

Raspberry Pi 3 Mod. B+

### System

Raspberry Pi reference 2024-03-15
version 2808975b80149bbfe86844655fe45c7de66fc078 (clean) (release) (start)
Linux voron0 6.6.51+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux

### Logs

```
$ dmesg -w
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 6.6.51+rpt-rpi-v8 (serge@raspberrypi.com) (gcc-12 (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08)
[    0.000000] KASLR enabled
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001e400000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x000000001e400000..0x000000002e3fffff (262144 KiB) map reusable linux,cma
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x0000000035ffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000035ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000035ffffff]
[    0.000000] On node 0, zone DMA: 8192 pages in unavailable ranges
[    0.000000] percpu: Embedded 30 pages/cpu s85672 r8192 d29016 u122880
[    0.000000] pcpu-alloc: s85672 r8192 d29016 u122880 alloc=30*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: ARM erratum 843419
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_hdmi=0  vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=fb33757d-02 rootfstype=ext4 fsck.repair=yes rootwait cfg80211.ieee80211_regdom=CN
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 217728
[    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
[    0.000000] Memory: 566216K/884736K available (13440K kernel code, 2212K rwdata, 4268K rodata, 4864K init, 1083K bss, 56376K reserved, 262144K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 43242 entries in 169 pages
[    0.000000] ftrace: allocated 169 pages with 4 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU event tracing is enabled.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000]  Trampoline variant of Tasks RCU enabled.
[    0.000000]  Rude variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[... 561 lines cut ...]
[  676.717725] brcmfmac: brcmf_set_channel: set chanspec 0xd02e fail, reason -52
[  676.825858] brcmfmac: brcmf_set_channel: set chanspec 0xd034 fail, reason -52
[  676.826342] brcmfmac: brcmf_set_channel: set chanspec 0xd038 fail, reason -52
[  676.826818] brcmfmac: brcmf_set_channel: set chanspec 0xd03c fail, reason -52
[  676.827215] brcmfmac: brcmf_set_channel: set chanspec 0xd040 fail, reason -52
[  797.372857] brcmfmac: brcmf_set_channel: set chanspec 0xd022 fail, reason -52
[  797.484965] brcmfmac: brcmf_set_channel: set chanspec 0xd026 fail, reason -52
[  797.592795] brcmfmac: brcmf_set_channel: set chanspec 0xd02a fail, reason -52
[  797.700787] brcmfmac: brcmf_set_channel: set chanspec 0xd02e fail, reason -52
[  797.808787] brcmfmac: brcmf_set_channel: set chanspec 0xd034 fail, reason -52
[  797.809278] brcmfmac: brcmf_set_channel: set chanspec 0xd038 fail, reason -52
[  797.809729] brcmfmac: brcmf_set_channel: set chanspec 0xd03c fail, reason -52
[  797.810160] brcmfmac: brcmf_set_channel: set chanspec 0xd040 fail, reason -52
[  918.355663] brcmfmac: brcmf_set_channel: set chanspec 0xd022 fail, reason -52
[  918.462969] brcmfmac: brcmf_set_channel: set chanspec 0xd026 fail, reason -52
[  918.568231] brcmfmac: brcmf_set_channel: set chanspec 0xd02a fail, reason -52
[  918.674975] brcmfmac: brcmf_set_channel: set chanspec 0xd02e fail, reason -52
[  918.783199] brcmfmac: brcmf_set_channel: set chanspec 0xd034 fail, reason -52
[  918.783549] brcmfmac: brcmf_set_channel: set chanspec 0xd038 fail, reason -52
[  918.783860] brcmfmac: brcmf_set_channel: set chanspec 0xd03c fail, reason -52
[  918.784165] brcmfmac: brcmf_set_channel: set chanspec 0xd040 fail, reason -52
[ 1014.510804] dwc_otg: DEVICE:006 : update_urb_state_xfer_comp:750:trimming xfer length
[ 1014.590447] usb 1-1.3: reset high-speed USB device number 6 using dwc_otg
[ 1039.383283] brcmfmac: brcmf_set_channel: set chanspec 0xd022 fail, reason -52
```

pelwell, 2024-11-09:
1. How is your Pi powered?
2. How are the attached devices powered?

ss8688, 2024-11-09:
> 1. How is your Pi powered?
> 2. How are the attached devices powered?

Pi is powered by 5v3a psu, usb ssd and usb wifi adapter is powerd by raspberry pi and printer controller is powered by another psu. So I should consider it is a power supply issue？