raspberrypi/linux #6462: hci0: BCM: Reading local name failed (-110)

manio, 2024-11-12:
### Describe the bug

Hi,
For about 1 of 10 of boots results in hci0 not visible.
I collected the following errors:
```
Jan  1 00:00:06 buildroot kern.err kernel: [    6.473578] Bluetooth: hci0: command 0x0c14 tx timeout
Jan  1 00:00:14 buildroot kern.err kernel: [   14.553559] Bluetooth: hci0: BCM: Reading local name failed (-110)
```
or
```
[    4.406253] Bluetooth: hci0: BCM: features 0x2e
[    4.419622] Bluetooth: hci0: BCM43436 37.4MHz Class 1.5 RaspBerry Pi Zero2 [Version: 1017.1042]
[    4.422736] Bluetooth: hci0: BCM43430A1 (001.002.009) build 1042
[    4.425642] Bluetooth: hci0: BCM: Using default device address (43:43:a1:12:1f:ac)
[    6.553570] Bluetooth: hci0: command 0x201c tx timeout
[    6.556460] Bluetooth: hci0: Opcode 0x201c failed: -110
```
or
```
Jan  1 01:10:55 buildroot kern.err kernel: [ 4255.673572] Bluetooth: hci0: command 0x0c1a tx timeout
Jan  1 01:10:55 buildroot kern.err kernel: [ 4255.673572] Bluetooth: hci0: Opcode 0x c1a failed: -110
```
This is leading to problems with bluetoothd which is not reporting any BT adapters, so I cannot use bluetooth.

I was trying to:
```
modprobe -r hci_uart
modprobe -r rfcomm
modprobe -r btbcm
modprobe -r bluetooth
```
And then after modprobing it again I don't have any initialization lines in `dmesg`.
Is this normal? How can I modprobe it again to have it fully initialized like on clean boot. Is it somehow related with devicetree?

I have to admit that this kind of workaround would be usable for me. I just need to reset/reinit this whole Bluetooth stack/kernel modules (btbcm?) without rebooting the whole Pi.

### Steps to reproduce the behaviour

Boot Pi and on some reboot it will occur. Next reboot most of the time fixes the problem.

### Device (s)

Raspberry Pi Zero 2 W

### System

Linux version 6.1.61-v7

### Logs

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 6.1.61-v7 (runner@fv-az735-94) (arm-buildroot-linux-gnueabihf-gcc.br_real (Buildroot -g7a59c3a4) 12.3.0, GNU ld (GNU Binutils) 2.40) #1 SMP Sat Mar 30 17:55:00 UTC 2024
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi Zero 2 W Rev 1.0
[    0.000000] random: crng init done
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x15800000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x0000000019bfffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000019bfffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000019bfffff]
[    0.000000] percpu: Embedded 12 pages/cpu s18004 r8192 d22956 u49152
[    0.000000] pcpu-alloc: s18004 r8192 d22956 u49152 alloc=12*4096
[... 178 lines cut ...]
[    1.383990] Bluetooth: HCI UART driver ver 2.3
[    1.386709] Bluetooth: HCI UART protocol H4 registered
[    1.389369] Bluetooth: HCI UART protocol Three-wire (H5) registered
[    1.392104] Bluetooth: HCI UART protocol Broadcom registered
[    1.392292] hci_uart_bcm serial0-0: supply vbat not found, using dummy regulator
[    1.397354] hci_uart_bcm serial0-0: supply vddio not found, using dummy regulator
[    1.775624] Bluetooth: hci0: BCM: chip id 94
[    1.779280] Bluetooth: hci0: BCM: features 0x2e
[    1.793200] Bluetooth: hci0: BCM43430A1
[    1.795585] Bluetooth: hci0: BCM43430A1 (001.002.009) build 0000
[    1.801079] Bluetooth: hci0: BCM43430A1 'brcm/BCM43430A1.raspberrypi,model-zero-2-w.hcd' Patch
[    2.225426] dwc2 3f980000.usb: supply vusb_d not found, using dummy regulator
[    2.227866] dwc2 3f980000.usb: supply vusb_a not found, using dummy regulator
[    2.333797] dwc2 3f980000.usb: EPs: 8, dedicated fifos, 4080 entries in SPRAM
[    3.112583] NET: Registered PF_INET6 protocol family
[    3.180252] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[    3.182830] Segment Routing with IPv6
[    3.187066] In-situ OAM (IOAM) with IPv6
[    3.390499] file system registered
[    3.439430] read descriptors
[    3.441607] read strings
[    4.456249] Bluetooth: hci0: BCM: features 0x2e
[    4.554131] uart-pl011 3f201000.serial: no DMA platform data
[    6.473577] Bluetooth: hci0: command 0x0c14 tx timeout
[   14.553562] Bluetooth: hci0: BCM: Reading local name failed (-110)

pelwell, 2024-11-12:
This looks like a symptom of undervoltage.
1. What power supply are you using?
2. What other hardware is attached?
3. What does `vcgencmd get_config int | grep volt` report?
4. How about `vcgencmd measure_volts`?

You could try adding `over_voltage=2` to config.txt, to see if that helps. If you aren't using WiFi, try with `dtoverlay=disable-wifi` in config.txt.

manio, 2024-11-12:
> This looks like a symptom of undervoltage.
> What power supply are you using?

The Pi is powered from the car headunit's USB - the project is DIY AndroidAuto dongle.

> What other hardware is attached?

LoRa device - but unused during this tests.

> What does vcgencmd get_config int | grep volt report?
> How about vcgencmd measure_volts?

I'll try that later, thanks

> You could try adding over_voltage=2 to config.txt, to see if that helps. If you aren't using WiFi, try with dtoverlay=disable-wifi in config.txt.

I am using WiFi and BT (and serial for LoRa), so I'll start with that `over_voltage` first.

manio, 2024-11-12:
# vcgencmd get_config int | grep volt
over_voltage_avs=6250
# vcgencmd measure_volts
volt=1.2563V

Is it ok?

pelwell, 2024-11-12:
It's fine - I just wanted to see if your Z2W had any voltage headroom, and it does. The limit is 1.4V, so if over_voltage=2 doesn't show any improvement, you can can go higher. Higher voltages won't do any damage in the short term, but they can theoretically shorten the life of the device, so use the lowest that works, and remove it if it doesn't help at all.

manio, 2024-11-13:
Some small update on this:
I started with over_voltage=1
```
# vcgencmd get_config int | grep volt
over_voltage=1
over_voltage_avs=6250
# vcgencmd measure_volts
volt=1.2813V
```
I had a feeling that it helped but suddenly I encounter the same problem.
So now I am testing it with level 2:
```
# vcgencmd get_config int | grep volt
over_voltage=2
over_voltage_avs=6250
# vcgencmd measure_volts
volt=1.3062V
```

I forgot to mention one important detail:
As I need a full featured serial to the external LoRa device and I also need bluetooth, so I set this:
`dtoverlay=pi3-miniuart-bt` (but without core_freq=250 as I don't want to degrade speed, I believe it is 400 IIRC)

and I also have this:
`dtparam=krnbt=on`
Which means that no hciattach is needed (in fact in this custom build the binary is missing).

I also found this: https://raspberrypi.stackexchange.com/questions/47284/pi3-and-mini-uart-bt
Where there seems to be identical problem which I currently have!
Even despite the proper `core_freq` set.

Overal... if the higher voltage can't fix this, I think I will try with removing `krnbt=on` and manually running:
```/usr/bin/hciattach /dev/ttyS0 bcm43xx 921600 noflow -```

And I believe it will force to reload the firmware (module unload is doing nothing as I stated above).

manio, 2024-11-19:
Today's update:
Hah... I was pretty sure that `over_voltage=2` helped!
It was working really well - I tested it heavily making a plenty of reboots.
But I decided to test it longer... and unfortunately today the problem appear again:
`[    6.473608] Bluetooth: hci0: Opcode 0x204b failed: -110`
And the same problem... no bt adapter.

This time I decided to not test with higher overvoltage but increase the boot_delay instead:
`boot_delay=3`

Testing...

manio, 2024-11-19:
Update:
Same problem:
```
[    6.553570] Bluetooth: hci0: command 0x0c25 tx timeout
[    6.556333] Bluetooth: hci0: Opcode 0x c25 failed: -110
```
I now disabled the `boot_delay` and `over_voltage=2` and added:
`core_freq=250`
Testing ...

@pelwell 
I also temporarily disabled the line `dtparam=krnbt=on` with manual hciattach`ing in mind - but to my surprise - I didn't notice any change: the bluetooth started automatically. Is this desired behavior? How can I force to use hciattach by me?

pelwell, 2024-11-19:
>  I didn't notice any change

That's because, after several years, we changed the default to `on`. You'll need `dtparam=krnbt=off` to revert to using hciattach, but that is not guaranteed to work forever.

pelwell, 2024-11-19:
```
pi@raspberrypi:~ $ dtparam -h krnbt
krnbt                   Set to "off" to disable autoprobing of Bluetooth
                        driver without need of hciattach/btattach
                        (default "on")
```

pelwell, 2024-11-19:
There is an experiment you could try:
1. Add `dtparam=krnbt=off` to config.txt.
2. `sudo systemctl disable hciuart`.
3. Reboot.
4. When you get to a command line: `sudo /usr/bin/btuart`
5. Check with `hcitool dev` and `dmesg -l err,warn§`.

To revert, remove the dtparam from config.txt, `sudo systemctl enable hciuart` and reboot.

manio, 2024-11-28:
@pelwell 
I was trying to do `dtparam=krnbt=off` and then playing with what's inside `btuart` script but it seems the device was still loaded by kernel and not with my `hciattach` util, BUT: this is not currently important, because:

I hope I finally found the root cause of my problems:
It was indeed the missing `core_freq=250` !!!
I know that documentation was saying that. But I was confused because 90% of the time it was working fine, so why downgrade pi performance? No! It seems this was indeed the missing puzzle !

I don't want to declare success prematurely but it is over a week (plenty of reboots) and I have 100% success rate with starting bluetooth, so I want to believe that this was it! :)

Thank you very much for your help and giving me the tips.

closed 2024-11-28 by manio

pelwell, 2024-11-28:
I'm glad you solved it. FYI, in recent firmwares there is a new option - `core_freq_fixed`. If you set it to 1, prevents the core clock from changing without the need to know what the platform's core frequency range is. It's intended for users of I2C and SPI, because the firmware doesn't automatically fix the core clock when they have been enabled.