raspberrypi/linux #6473: Pi 5 low 3V7_WL_SW_V when high Temp

llllllllqq, 2024-11-16:
### Describe the bug

I find an issue that the 3V7_WL_SW_V value (got by vcgencmd pmic_read_adc) will be lower if the CPU tempature (got by vcgencmd measure_temp) goes higher.

e.g. When 40'C, 3V7=~3.69v
When 50'C, 3V7=~3.67v
When 60'C, 3V7=~3.64v
...
When 70'C, 3V7=3.60v
And When 70'C, if I use an ice pack to cool down the Pi to 40'C, the 3V7 turns from 3.60v to 3.69v quickly.
All other Voltage (like PD 5V) values keep steady during all my tests. I use a 25W PD adapter (USB meter shows it can output 5A without big voltage decrease).
This issue seems have nothing to do with FAN power drain (I unplugged it and re-test).

I wonder whether it's only an accuracy bug about the built-in Voltage meter or the worse -- a real bug in the PMIC 3V7 converter. I cannot find the test pad of 3V7 so I cannot prove it with an external voltage meter.

### Steps to reproduce the behaviour

Simply stress/cooldown the CPU and watch vcgencmd pmic_read_adc and vcgencmd measure_temp.

### Device (s)

Raspberry Pi 5

### System

Raspberry Pi reference 2024-10-22
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, ecda2579ebbc125b321984c571e3128f6fd959d4, stage4

2024/09/23 14:02:56
Copyright (c) 2012 Broadcom
version 26826259 (release) (embedded)

Linux rpi5 6.6.51+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux

### Logs

(a full view when ~58'C)
```
 3V7_WL_SW_A current(0)=0.00390372A
   3V3_SYS_A current(1)=0.18542670A
   1V8_SYS_A current(2)=0.26642890A
  DDR_VDD2_A current(3)=0.02147046A
  DDR_VDDQ_A current(4)=0.00098040A
   1V1_SYS_A current(5)=0.46356680A
    0V8_SW_A current(6)=0.51529100A
  VDD_CORE_A current(7)=2.67900000A
   3V3_DAC_A current(17)=0.00000000A
   3V3_ADC_A current(18)=0.00042735A
   0V8_AON_A current(16)=0.00616605A
      HDMI_A current(22)=0.01282050A
 3V7_WL_SW_V volt(8)=3.64317100V
   3V3_SYS_V volt(9)=3.31652900V
   1V8_SYS_V volt(10)=1.80121900V
  DDR_VDD2_V volt(11)=1.12380800V
  DDR_VDDQ_V volt(12)=0.60476130V
   1V1_SYS_V volt(13)=1.11355200V
    0V8_SW_V volt(14)=0.80183080V
  VDD_CORE_V volt(15)=0.85152540V
   3V3_DAC_V volt(20)=3.31409900V
   3V3_ADC_V volt(21)=3.31409900V
   0V8_AON_V volt(19)=0.79794790V
      HDMI_V volt(23)=5.12416000V
     EXT5V_V volt(24)=5.12282000V
      BATT_V volt(25)=2.99743300V
```

popcornmix, 2024-11-27:
3.7V reading as 3.643V is about 1.5% out. 
That is within the expected tolerances of the components that measure the voltage.
It it likely the actual voltage (if probed with more accurate equipment) would be closer to 3.7V.

So I don't believe there's any issue here.

llllllllqq, 2024-11-27:
> 3.7V reading as 3.643V is about 1.5% out. That is within the expected tolerances of the components that measure the voltage. It it likely the actual voltage (if probed with more accurate equipment) would be closer to 3.7V.
> 
> So I don't believe there's any issue here.

Thanks for your reply. 
What is the tolerance range? It will be 3.60v or even 3.5*v when above 70℃.

dardardario, 2025-01-04:
There's also a possibility (although not typically on a 3V rail) that the decrease is intentional. This is typically done on CPU core rails because less voltage is needed when running at higher temps (Vgs goes down by leakage increases). This is typically only a few mV of change. 

I am not aware of what this rail does (or the specifics of the RPi's power delivery system) but my guess would be that this shouldn't be concerning as it is not likely a rail where 50mV of difference matters. The places where 50mV can be significant is typically on the 0.4-1.2V rails which power logic.

llllllllqq, 2025-04-04:
update: I find a 3.7v point and use a multimeter to measure. Then I find the 3V7_WL_SW_V value is the REAL value, not an ADC deviation.

I also find some other people posting their vcgencmd pmic_read_adc, and their 3V7_WL_SW_V quite different (3.65~3.75v), but their other volt rails quite close (like 3v3 rail, often 3.29~3.31v).

However, with wifi and bt off (not need), my pi running fine for 3 months with 3V7_WL_SW_A always 0. It seems this rail has only sth to do with wifi&bt, so I give up searching for more info about this issue.

closed 2025-04-04 by llllllllqq