raspberrypi/linux #6489: Can't compile DTS overlay from source

DangerD1024, 2024-11-23:
### Describe the bug

Can't compile dts overlay

### Steps to reproduce the behaviour

git clone --depth=1 https://github.com/raspberrypi/linux
cd linux/arch/arm/boot/dts/overlays
dtc -I dts -O dtb -o imx290-overlay.dtbo imx290-overlay.dts
Error: imx290-overlay.dts:6.1-9 syntax error
FATAL ERROR: Unable to parse input tree

### Device (s)

Raspberry Pi CM4

### System

Ubuntu 6.6 

### Additional context

```
dtc --version
Version: DTC 1.6.1
```

pelwell, 2024-11-23:
That's not a bug - the overlays use header files from the kernel tree, and you are supposed to use the kernel make system to build them. However, there is an alternative: https://forums.raspberrypi.com/viewtopic.php?t=377084

macmpi, 2024-11-23:
@pelwell that write-up would be nice hint within repo readme too.

pelwell, 2024-11-23:
See https://github.com/raspberrypi/utils/pull/104.