raspberrypi/linux #6490: Incorrect 5ghz wifi channels for India

nitinkmr333, 2024-11-24:
### Describe the bug

I am from India and I have been facing few issues on Raspberry Pi 5-
1. No 5ghz connections visible. I can only see 2.4ghz connections.
2. If I change `cfg80211.ieee80211_regdom=` to some other country (like Hong Kong) then I can see 5ghz network (although still very unreliable and randomly cause error- `kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd022 fail, reason -52`). 

I believe above issues are because of incorrect wifi channels set for India on Raspberry Pi OS. I have a separate `NixOS` system which works perfectly fine with both 2.4ghz and 5ghz access points. I noticed both show support for different wifi channels.

Raspberry Pi 5 (**Raspberry Pi OS** - Debian 12)-
```
❯ sudo iwlist wlan0 frequency
wlan0     32 channels in total; available frequencies :
          Channel 01 : 2.412 GHz
          Channel 02 : 2.417 GHz
          Channel 03 : 2.422 GHz
          Channel 04 : 2.427 GHz
          Channel 05 : 2.432 GHz
          Channel 06 : 2.437 GHz
          Channel 07 : 2.442 GHz
          Channel 08 : 2.447 GHz
          Channel 09 : 2.452 GHz
          Channel 10 : 2.457 GHz
          Channel 11 : 2.462 GHz
          Channel 12 : 2.467 GHz
          Channel 13 : 2.472 GHz
          Channel 34 : 5.17 GHz
          Channel 36 : 5.18 GHz
          Channel 38 : 5.19 GHz
          Channel 40 : 5.2 GHz
          Channel 42 : 5.21 GHz
          Channel 44 : 5.22 GHz
          Channel 46 : 5.23 GHz
          Channel 48 : 5.24 GHz
          Channel 52 : 5.26 GHz
          Channel 56 : 5.28 GHz
          Channel 60 : 5.3 GHz
          Channel 64 : 5.32 GHz
          Channel 100 : 5.5 GHz
          Channel 104 : 5.52 GHz
          Channel 108 : 5.54 GHz
          Channel 112 : 5.56 GHz
          Channel 116 : 5.58 GHz
          Channel 120 : 5.6 GHz
          Channel 124 : 5.62 GHz
          Current Frequency:2.462 GHz (Channel 11)
```

x86 laptop (**NixOS**)-
```
❯ sudo iwlist wlo1 frequency
wlo1      32 channels in total; available frequencies :
          Channel 01 : 2.412 GHz
          Channel 02 : 2.417 GHz
          Channel 03 : 2.422 GHz
          Channel 04 : 2.427 GHz
          Channel 05 : 2.432 GHz
          Channel 06 : 2.437 GHz
          Channel 07 : 2.442 GHz
          Channel 08 : 2.447 GHz
          Channel 09 : 2.452 GHz
          Channel 10 : 2.457 GHz
          Channel 11 : 2.462 GHz
          Channel 12 : 2.467 GHz
          Channel 13 : 2.472 GHz
          Channel 36 : 5.18 GHz
          Channel 40 : 5.2 GHz
          Channel 44 : 5.22 GHz
          Channel 48 : 5.24 GHz
          Channel 52 : 5.26 GHz
          Channel 56 : 5.28 GHz
[... 196 lines cut ...]
Nov 06 18:06:14 rpl NetworkManager[724]: <info>  [1730896574.8331] manager: NetworkManager state is now CONNECTED_SITE
Nov 06 18:06:14 rpl NetworkManager[724]: <info>  [1730896574.8380] device (wlan0): Activation: successful, device activated.
Nov 06 18:06:14 rpl NetworkManager[724]: <info>  [1730896574.8385] manager: NetworkManager state is now CONNECTED_GLOBAL
Nov 06 18:06:16 rpl NetworkManager[724]: <info>  [1730896576.6957] dhcp6 (wlan0): activation: beginning transaction (timeout in 45 seconds)
Nov 06 18:06:16 rpl NetworkManager[724]: <info>  [1730896576.6963] policy: set 'Nitinkumar-4G' (wlan0) as default for IPv6 routing and DNS
Nov 06 18:06:16 rpl avahi-daemon[627]: Leaving mDNS multicast group on interface wlan0.IPv6 with address fe80::e818:d87:40b0:90b7.
Nov 06 18:06:16 rpl NetworkManager[724]: <info>  [1730896576.7055] dhcp6 (wlan0): state changed new lease
Nov 06 18:06:16 rpl avahi-daemon[627]: Joining mDNS multicast group on interface wlan0.IPv6 with address 2405:201:681a:8842:4e32:4625:df8d:9945.
Nov 06 18:06:16 rpl avahi-daemon[627]: Registering new address record for 2405:201:681a:8842:4e32:4625:df8d:9945 on wlan0.*.
Nov 06 18:06:16 rpl avahi-daemon[627]: Withdrawing address record for fe80::e818:d87:40b0:90b7 on wlan0.
Nov 06 18:06:22 rpl sudo[2551]: pam_unix(sudo:session): session closed for user root
Nov 06 18:06:26 rpl systemd[1]: NetworkManager-dispatcher.service: Deactivated successfully.
Nov 06 18:06:30 rpl sudo[2649]:    nitin : TTY=pts/0 ; PWD=/home/nitin ; USER=root ; COMMAND=/usr/sbin/poweroff
Nov 06 18:06:30 rpl sudo[2649]: pam_unix(sudo:session): session opened for user root(uid=0) by nitin(uid=1000)
Nov 06 18:06:30 rpl systemd-logind[641]: The system will power off now!
Nov 06 18:06:30 rpl systemd-logind[641]: System is powering down.
```

### Additional context

Issue is likely caused by not Raspberry Pi OS not having support for these 5ghz frequencies for India-
Channel 128 : 5.64 GHz
Channel 132 : 5.66 GHz
Channel 136 : 5.68 GHz
Channel 140 : 5.7 GHz

desiFish, 2025-02-20:
I too have the same problem for both RP5 and RP4. Can anyone please look into this? Setting the country as IN instantly vanishes the 5Ghz band. Only 2.4Ghz is visible.

varifocal, 2026-09-26:
same problem here too the specific channel 149 i am unable to connect to