raspberrypi/linux #6491: SC16IS7xx: SPI to (full)UART: long messages are truncated?

danergo, 2024-11-25:
### Describe the bug

With RPi3 we have an SC16IS752 board (actually two) attached to SPI of the RPi to provide full-featured UART ports (RX, TX, RTS, CTS).

We have also a Bluetooth co-processor, which is attached to RPi via HCI over this full-featured UART port of the SC16IS752.

Baudrate is 921600, and full-hardware flow-control is required (requirement from Bluetooth co-processor).

We then use the Bluetooth massively, lots of radio traffic, hence lots of data flow over the SC16IS752.

Every now-and-then (after about 12hrs of constant running), we experience truncated packets coming out from SC16IS7xx:

In case we send short message from RPi, it works as intended:

[image]

However anything longer than 64 bytes are truncated to exactly 64 bytes:

[image]

In case we restart BlueZ on RPi (or RPi itself), transmission gets back to normal, but long data transmits are still looking strange:

[image]

In case a bit longer:

[image]

Bluetooth co-processor is well-capable of handling these dataflows. Are my last two signal-traces correct? Is it okay to "pause" after every 64bytes of data?

### Steps to reproduce the behaviour

Please see above.

### Device (s)

Raspberry Pi CM3+

### System

```
Debian GNU/Linux 12 \n \l
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start_cd)
Linux rpi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
```

danergo, 2024-11-25:
I tried it with `gatttool` to verify, hence the test-data of 0x00000000..ffffffff.

RPi runs `btattach` to create a hci interface, and BlueZ (bluetoothd) is using that hci0 interface for communication.

Underlying hardware is the SC16IS752 over SPI, it provides the required UARTs to the OS:

```
/usr/bin/btattach -S 921600 -B /dev/ttySC0
```

ttySC0 (from dmesg: it only shows the second UART. ttySC0 is on IRQ 184):

```
[    6.822856] spi0.1: ttySC1 at I/O 0x1 (irq = 184, base_baud = 921600) is a SC16IS752
[    6.827204] spi0.0: ttySC3 at I/O 0x1 (irq = 185, base_baud = 921600) is a SC16IS752
```
This is the DTS driving SC16IS752. (Please note we do have two of these, providing 4 UARTs total)

```
/dts-v1/;                                                                                                                                                                                                          
/plugin/;                                                                                                                                                                                                          
                                                                                                                                                                                                                   
/{                                                                                                                                                                                                                 
    compatible = "brcm,bcm2835";                                                                                                                                                                                   
                                                                                                                                                                                                 
    /* Disable SPI0 device with CE0 - from now on, we will use it exclusively for UARTs */                                                                                                                         
    fragment@0 {                                                                                                                                                                                                   
        target = <&spidev0>;                                                                                                                                                                                       
        __overlay__ {                                                                                                                                                                                              
            status = "disabled";                                                                                                                                                                                   
        };                                                                                                                                                                                                         
    };        

    /* Disable SPI0 device with CE1 - from now on, we will use it exclusively for UARTs */                                                                                                                         
    fragment@1 {                                                                                                                                                                                                   
        target = <&spidev1>;                                                                                                                                                                                       
        __overlay__ {                                                                                                                                                                                              
            status = "disabled";                                                                                                                                                                                   
        };                                                                                                                                                                                                         
    };      

    /* SPI2UART1                                                                                                                                                             
     *  Provides /dev/SC2 and /dev/SC3                                                                                                                                                                         
     *                                                                                                                                                                                                             
     *  Used GPIOs:                                                                                                                                                                                                
[... cut ...]
        __overlay__ {
            sc16is752_1_clk: sc16is752_spi0_1_clk {
                compatible = "fixed-clock";
                #clock-cells = <0>;
                clock-frequency = <14745600>;
            };
        };
    };

    __overrides__ {
        int0_pin = <&sc16is752_0>,"interrupts:0";
        xtal0 = <&sc16is752_0_clk>,"clock-frequency:0";

        int1_pin = <&sc16is752_1>,"interrupts:0";
        xtal1 = <&sc16is752_1_clk>,"clock-frequency:0";
    };
};
```

What is your opinion?

danergo, 2024-11-25:
@pelwell: 

> I note that you are using an old kernel - it might be worth trying at least 6.6.

Were there any commits between 6.1.21-v8+ and 6.6 which "touch" this area?

danergo, 2024-11-25:
We are using all 4 UARTs, and 1 of them is for BT HCI. Others are running much slower bauds (115200, 9600 and between these).

HCI needs massive data flow, so it requires 921600baud.

But I have to emphasize: it works correctly for hours (12) and sometimes days, but definitely not more than 2days.

During correct working, it sends longer data correctly (although it stops after every 64bytes).

However once it fails, it will only send the first 64bytes.

In case I restart the btattach procedure, it reinits the serial port (and hci), and it will go for another couple of hours or days correctly then will eventually fail again.

Among the commits for the SC16IS7xx, there are quite a few which might be the root cause of this, but I wish to make this 100% sure. I'm investigating this issue since September, but I blamed the Bluetooth Controller first (still can be problematic or can cause some unlucky misconfiguration with this old kernel's driver).

So now I don't want to update (yet) until I know better which commit shall solve this :)

Thank you!

pelwell, 2024-11-25:
You have still failed to answer my first question:

What application is sending the non-Bluetooth traffic? And which UART/serial library does it use?

pelwell, 2024-11-25:
64 bytes is the size of the FIFOs in the SC16IS752 - sending more than 64 bytes at a time is likely to cause data loss.

And yes, there has been some activity on that driver since 6.1:
```
2241ab53cbb5c Linux 6.2-rc5
01d93875dafda serial: sc16is7xx: Convert to i2c's .probe_new()
eeac8ede17557 Linux 6.3-rc2
0d07703be74fe serial: Switch i2c drivers back to use .probe()
9561de3a55bed Linux 6.4-rc5
fd2b55f86b8b2 serial: drivers: switch ch and flag to u8
77a82cebf0eb0 serial: sc16is7xx: Put IOControl register into regmap_volatile
5d0c230f1de8c Linux 6.5-rc4
2861ed4d6e6d1 serial: sc16is7xx: fix broken port 0 uart init
dabc54a45711f serial: sc16is7xx: remove obsolete out_thread label
0499942928341 serial: sc16is7xx: fix regression with GPIO configuration
9baeea723c0fb serial: sc16is7xx: fix bug when first setting GPIO direction
b4a778303ea0f serial: sc16is7xx: add missing support for rs485 devicetree properties
2dde18cd1d8fa Linux 6.5
98b0ca904c804 serial: sc16is7xx: address RX timeout interrupt errata
ac25535242acb Linux 6.6.7
9d528a81df2fc serial: sc16is7xx: add check for unsupported SPI modes during probe
7ce7425537898 serial: sc16is7xx: set safe default SPI clock frequency
98817289bcec2 Linux 6.6.14
452ed2b218b1d serial: sc16is7xx: improve regmap debugfs by using one regmap per port
f769407d01948 serial: sc16is7xx: remove wasteful static buffer in sc16is7xx_regmap_name()
fc3de570cb30d serial: sc16is7xx: remove global regmap from struct sc16is7xx_port
2f6ae16a5874e serial: sc16is7xx: remove unused line structure member
9879e1bec3c0f serial: sc16is7xx: change EFR lock to operate on each channels
084c24e788d9c serial: sc16is7xx: convert from _raw_ to _noinc_ regmap functions for FIFO
fbce37f616ba7 serial: sc16is7xx: fix invalid sc16is7xx_lines bitfield in case of probe error
8ed85bdd1f2f9 serial: sc16is7xx: remove obsolete loop in sc16is7xx_port_irq()
5c0471daa8c7b serial: sc16is7xx: improve do/while loop in sc16is7xx_irq()
a38e80d02eba2 serial: sc16is7xx: Use port lock wrappers
4739a8a989db3 serial: sc16is7xx: fix unconditional activation of THRI interrupt
51f354b815c44 Linux 6.6.15
f85f63647920c serial: sc16is7xx: Don't spin if no data received
3a49598d7be0e serial: sc16is7xx: Read modem line state at startup
62e5ae5007ef1 Linux 6.6.21
3431d265eab2c serial: sc16is7xx: add proper sched.h include for sched_set_fifo()
f0eb53ffd8d96 serial: sc16is7xx: replace hardcoded divisor value with BIT() macro
11244a4328504 serial: sc16is7xx: fix bug in sc16is7xx_set_baud() when using prescaler
140cf97204b54 Linux 6.6.33
```

danergo, 2024-11-26:
> You have still failed to answer my first question:
> 
> What application is sending the non-Bluetooth traffic? And which UART/serial library does it use?

Sorry, for this bit of misunderstanding here:

```
/dev/ttySC0 -> Bluetooth HCI, used by BlueZ.
/dev/ttySC1 -> Bluetooth debug, coprocessor's debug messages, minicom is reading
/dev/ttySC2 -> 3rd party communication by our application: native C serial: open, close, write, read (unrelated to this issue)
/dev/ttySC3 -> 3rd party communication by our application: native C serial: open, close, write, read (unrelated to this issue)
```

> And yes, there has been some activity on that driver since 6.1

Perfect, thank you.

I also got some insights from hvilleneuvedoo:
> I recently fixed an issue with this driver, make sure you apply all the fixes in the latest tree for it to your 6.1 tree, especially these two commits: serial: sc16is7xx: **fix invalid FIFO access with special** serial: sc16is7xx: **fix TX fifo corruption**

However I don't see these 2 important commits in your list. Are they newer, and not included in 6.6.33?

pelwell, 2024-11-26:
Those changes were introduced in Linux 6.11, but there were another 30+ commits to that driver between 6.6 and 6.11 - that's a lot to backport, but it could be done.

We are likely to be transitioning to the 6.12 kernel in the new year. As part of that process, there is an early 6.12 build available to install on our `next` branch: `sudo rpi-update next`. After backing up your data or cloning your card, I suggest trying 6.12 to see if the problem goes away.

danergo, 2024-11-27:
Thank you @pelwell, (also @hvilleneuvedoo).

6.12 seems promising: not a single lost byte since 18hrs.

With 6.1, it usually reached to only 4-10hrs, and once or twice to max 30hrs, but never more.

Will report back after a couple of days, and if all goes well, we can then close this ticket.

Thank you!

danergo, 2024-12-02:
@pelwell, @hvilleneuvedoo, thank you so much for your fixes in this kernel driver.

We can confirm, our issue has been completely gone after updated to 6.12 version. Running since more than 5 days without any single lost byte of data.

Thank you.

hvilleneuvedoo, 2024-12-02:
On 12/2/24 01:54, danergo wrote:
> @pelwell <https://github.com/pelwell>, @hvilleneuvedoo <https:// 
> github.com/hvilleneuvedoo>, thank you so much for your fixes in this 
> kernel driver.
> 
> We can confirm, our issue has been completely gone after updated to 6.12 
> version. Running since more than 5 days without any single lost byte of 
> data.

Hi,
glad to hear this.

Hugo.