raspberrypi/linux #6492: diskcorruption on 6.12.y

folkertvanheusden, 2024-11-25:
### Describe the bug

Revision 521f2baed818c04981fd61b275c996a8ef03b833 of branch rpi-6.12.y gives massive disk-corruption when realtime-kernel is enabled.

### Steps to reproduce the behaviour

Configure kernel for real-time.

### Device (s)

Raspberry Pi Zero 2 W, Raspberry Pi 3 Mod. B

### System

raspbian 64bit lite

### Logs

Not directly available. If this problem is of interest, I can see if I can reproduce it.
Errors I saw were related to directory-nodes being full.

pelwell, 2024-11-25:
Realtime support is new to 6.12:
1. Have you add a non-corrupting 6.12 realtime kernel before the indicated commit?
2. Does a non-realtime build of 6.12 work for you?
3. Have you add a non-corrupting 6.11 realtime (with patches) kernel?
4. Have you tried realtime kernels before?

popcornmix, 2024-11-27:
> Both the realtime and non-realtime 6.12 kernel fail:

I think you'll need to back off to a known good point, confirm that is reliable then just update the kernel.

Start with a clean install of RPiOS Bookworm 64-bit lite.
Update it (`sudo apt update && sudo apt full-upgrade`).

Confirm sdcard is reliable with no complaints in dmesg when running your normal workloads.

Now update to *our* build of the 6.12 kernel.
```sudo rpi-update next```
reboot and report if any sdcard corruption issues.

folkertvanheusden, 2024-11-27:
That build of the kernel works fine.
What would be the procedure for getting a kernel with CONFIG_PREEMPT_RT instead?

popcornmix, 2024-11-27:
You said
> Both the realtime and non-realtime 6.12 kernel fail:

So I think the first issue to resolve is if the default 6.12 (non-realtime) kernel has an issue with corrupting the sdcard.
Once we have an answer to that, we can consider enabling RT. But you don't want to be changing too many things at once.

folkertvanheusden, 2024-11-27:
> You said
>
>
> So I think the first issue to resolve is if the default 6.12 (non-realtime) kernel has an issue with corrupting the sdcard. Once we have an answer to that, we can consider enabling RT. [...]

Yes, but *not* the one from `rpi-update next`.
I did a diff and see quite a few possible reasons for "my" version to fail.
So 'm going to try only changing the scheduling parameters and see if that helps.

nbuchwitz, 2025-01-03:
@folkertvanheusden were you able to reproduce the errors with a more recent 6.12 version? Haven't seen such errors on PREEMPT RT enabled 6.12 builds yet.

Tested our tree (with some vendor patches, mostly targeting dt for our hardware platform) on CM5 with eMMC, pi500 with the stock rpi sd card and rpi2w. So far none of these has reported any ext4 warnings - even under high (io) load tests.  Are there any other warnings / errors in kernel log? There are still some quirks with rt on rpi (dwc_otg for example needs patching or switch to dwc2)

folkertvanheusden, 2025-01-03:
Hi,

I haven't tried it further, maybe later.