raspberrypi/linux #6497: Feature request: Include iwlwifi kernel module as part of standard kernel build

evilmonkeyslayer, 2024-11-27:
### Describe the bug

Feature request:

With the addition of pcie adapters this allows pcie devices to be plugged into the Pi 5 that were previously not usable.

This is a request to include in the standard kernel build the iwlwifi module and related modules so that devices such as the Intel AX210 which is a very common wifi chipset to be usable on the Pi 5.

This requires the following in the kernel .config file:

> CONFIG_IWLWIFI=m
> CONFIG_IWLWIFI_LEDS=y
> CONFIG_IWLDVM=m
> CONFIG_IWLMVM=m
> CONFIG_IWLWIFI_OPMODE_MODULAR=y
> \#
> \# Debugging Options
> \#
> \# CONFIG_IWLWIFI_DEBUG is not set
> CONFIG_IWLWIFI_DEVICE_TRACING=y
> \# end of Debugging Options

### Steps to reproduce the behaviour

N/A

### Device (s)

Raspberry Pi 5

### System

N/A

rowingdude, 2024-12-09:
This is not something likely to be included. It's not a RPi core piece of hardware and therefore has no place in the mainstream kernel.

pelwell, 2024-12-09:
That's nonsense (or reverse psychology) - the majority of the hundreds of modules are for non-Raspberry Pi hardware.

evilmonkeyslayer, 2024-12-09:
> That's nonsense (or reverse psychology) - the majority of the hundreds of modules are for non-Raspberry Pi hardware.

I was going to post a reply along these lines, there are a significant number of nic and wlan drivers set to compile as modules by default with the standard pi kernel. Including the iwlwifi modules doesn't use much space at all.

For example, from the modules dir:

```
184K    iwlwifi/mvm
80K     iwlwifi/dvm
400K    iwlwifi/
```

pelwell, 2024-12-09:
I'll add them next time I'm near a keyboard.

pelwell, 2024-12-09:
Pull request #6526 adds the required modules. To install a trial build, wait about 40 minutes then run `sudo rpi-update pulls/6526` (after backing up any important data).

closed 2024-12-10 by pelwell

timwb, 2025-02-17:
Great to have this - however it seems to have disappeared again from the next branch / 6.12 kernel branch. Could this module be included there as well? Thanks!

geerlingguy, 2025-03-03:
And just like that, another reason I _don't_ have to recompile the kernel so much...