raspberrypi/linux #6500: MT7921 wifi  tx urb failed

ss8688, 2024-11-29:
### Describe the bug

mt7921u 1-1.3:1.0: tx urb failed: -71

### Steps to reproduce the behaviour

using mt7921.

### Device (s)

Raspberry Pi 3 Mod. B+

### System

 6.6.51+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux
Raspberry Pi reference 2024-03-15

### Logs

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 6.6.51+rpt-rpi-v8 (serge@raspberrypi.com) (gcc-12 (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08)
[    0.000000] KASLR enabled
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001e400000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x000000001e400000..0x000000002e3fffff (262144 KiB) map reusable linux,cma
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x0000000037ffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000037ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000037ffffff]
[    0.000000] percpu: Embedded 30 pages/cpu s85672 r8192 d29016 u122880
[    0.000000] pcpu-alloc: s85672 r8192 d29016 u122880 alloc=30*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: ARM erratum 843419
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_hdmi=0  vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=fb33757d-02 rootfstype=ext4 fsck.repair=yes rootwait cfg80211.ieee80211_regdom=CN
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 225792
[    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
[    0.000000] Memory: 598916K/917504K available (13440K kernel code, 2212K rwdata, 4268K rodata, 4864K init, 1083K bss, 56444K reserved, 262144K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 43242 entries in 169 pages
[    0.000000] ftrace: allocated 169 pages with 4 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU event tracing is enabled.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000]  Trampoline variant of Tasks RCU enabled.
[    0.000000]  Rude variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: bcm2836_arm_irqchip_handle_irq
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000001] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[... 625 lines cut ...]
[  278.751812] mt7921u 1-1.3:1.0: timed out waiting for pending tx
[  279.039866] usb 1-1.3: new high-speed USB device number 8 using dwc_otg
[  279.141227] usb 1-1.3: New USB device found, idVendor=0e8d, idProduct=7961, bcdDevice= 1.00
[  279.141277] usb 1-1.3: New USB device strings: Mfr=2, Product=3, SerialNumber=4
[  279.141285] usb 1-1.3: Product: Wireless_Device
[  279.141291] usb 1-1.3: Manufacturer: MediaTek Inc.
[  279.141297] usb 1-1.3: SerialNumber: 000000000
[  279.223780] usb 1-1.3: reset high-speed USB device number 8 using dwc_otg
[  279.354699] mt7921u 1-1.3:1.0: HW/SW Version: 0x8a108a10, Build Time: 20230526130917a

[  279.619957] mt7921u 1-1.3:1.0: WM Firmware Version: ____010000, Build Time: 20230526130958
[  288.320809] wlan1: authenticate with 8c:de:f9:22:58:0d
[  288.850880] wlan1: send auth to 8c:de:f9:22:58:0d (try 1/3)
[  288.909973] wlan1: authenticate with 8c:de:f9:22:58:0d
[  288.918260] wlan1: send auth to 8c:de:f9:22:58:0d (try 1/3)
[  288.950015] wlan1: authenticated
[  288.951807] wlan1: associate with 8c:de:f9:22:58:0d (try 1/3)
[  288.969771] wlan1: RX AssocResp from 8c:de:f9:22:58:0d (capab=0x1511 status=0 aid=2)
[  289.007103] wlan1: associated
[  289.168245] wlan1: Limiting TX power to 27 (30 - 3) dBm as advertised by 8c:de:f9:22:58:0d
```

### Additional context

According to what is said here https://github.com/morrownr/USB-WiFi/issues/410 , it is a kernel driver bug?

pelwell, 2024-11-29:
What else is attached to your 3B+?
How is the Pi powered?

The linked thread is very long - please give a link to the bit that makes you think that the two issues are connected.

P33M, 2024-11-29:
Spontaneous transaction errors followed by disconnect/reconnect are usually a symptom of the device exceeding the available downstream port power. What happens when you plug the dongle into a powered USB hub?

ss8688, 2024-11-29:
After the discussion in https://github.com/raspberrypi/linux/issues/6460, I connected an external power source to my USB WiFi adapter, and since then, there have been no power supply issues.However, after switching to a new USB WiFi adapter, the issue gone on. This time, it should be a kernel-related problem. Additionally, according to https://github.com/morrownr/USB-WiFi/issues/410, this problem also affects x86 devices.

ss8688, 2024-11-29:
also see https://github.com/morrownr/USB-WiFi/issues/477

ss8688, 2024-12-08:
ok, it is a issue of bad usb cable

pelwell, 2024-12-08:
Nobody should be surprised - we've seen many shockingly bad USB cables over the years.

ss8688, 2024-12-11:
No,It not a usbcabel issue.This problem has appear again.I feel like it's related to the signal strength

ss8688, 2024-12-11:
But when this error occurs and the network is disconnected, there is still a link rate of 300mbps.

TwoBottomBuns, 2025-03-17:
I've also been stuck with similar problems and have spent a fair amount of time trying to debug it. I posted about it in the Raspberry Pi Forums last week [here](https://forums.raspberrypi.com/viewtopic.php?p=2302799) but have yet to see any responses to it.  I'm convinced that there are just issues with the 6.6 Raspberry Pi kernel and the USB host controller being able to deal with Mediatek wifi cards appropriately.

I'm not going to repost my whole forum post here, but when I use an Alfa Networks AWUS036ACM WiFi adapter and just repeatedly run Nmap scans on a /24 network, I can get the USB device to just completely lock up. It seems like there comes a point that the ieee80211 queues just aren't being processed anymore, so the queues completely fill up and it can't do anything else. It can be fixed by rebooting the pi or re-plugging the USB device. Most people report issues when using Mediatek devices in AP mode, but I think the problem is more broad than that, as I'm getting similar issues without that.

I'll note that I'm not using a powered USB hub, no USB cable to blame, and I get the problem to occur in both USB2 and USB3 ports.

I know the pi kernel is getting close to releasing 6.12, and I have had a lot of luck using that kernel. My device has not locked up at all since upgrading with `rpi-update`. It might be worth a try for others if they are still stuck, but it would be nice to find what the root cause of these lockups are at some point. I'm not super familiar with the kernel source code, but am willing to help track something down.