raspberrypi/linux #6503: build fails with rpi-6.12.y bcmrpi_defconfig (armv6)

ncopa, 2024-11-29:
### Describe the bug

building the 6.12.y kernel with `bcmrpi_defconfig` for armv6 fails with:

```
  CC      kernel/debug/kdb/gen-kdb_cmds.o
kernel/debug/kdb/gen-kdb_cmds.c:3:60: error: expected ',' or ';' before 'Common'
    3 | static __initdata char kdb_cmd0[] = "defcmd dumpcommon "" "Common kdb debugging"\n";
      |                                                            ^~~~~~
kernel/debug/kdb/gen-kdb_cmds.c:12:57: error: expected ',' or ';' before 'First'
   12 | static __initdata char kdb_cmd9[] = "defcmd dumpall "" "First line debugging"\n";
      |                                                         ^~~~~
kernel/debug/kdb/gen-kdb_cmds.c:17:58: error: expected ',' or ';' before 'Same'
   17 | static __initdata char kdb_cmd14[] = "defcmd dumpcpu "" "Same as dumpall but only tasks on cpus"\n";
      |                                                          ^~~~
make[5]: *** [/home/ncopa/aports/main/linux-rpi/src/linux-6.12/scripts/Makefile.build:244: kernel/debug/kdb/gen-kdb_cmds.o] Error 1
make[4]: *** [/home/ncopa/aports/main/linux-rpi/src/linux-6.12/scripts/Makefile.build:485: kernel/debug/kdb] Error 2
make[3]: *** [/home/ncopa/aports/main/linux-rpi/src/linux-6.12/scripts/Makefile.build:485: kernel/debug] Error 2
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [/home/ncopa/aports/main/linux-rpi/src/linux-6.12/scripts/Makefile.build:485: kernel] Error 2
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [/home/ncopa/aports/main/linux-rpi/src/linux-6.12/Makefile:1936: .] Error 2
make: *** [/home/ncopa/aports/main/linux-rpi/src/linux-6.12/Makefile:224: __sub-make] Error 2
```

### Steps to reproduce the behaviour

make  bcmrpi_defconfig && make

### Device (s)

Raspberry Pi 1 Mod. B+

### System

alpine linux edge

### Additional context

Building the 6.12 RPi kernel for Alpine Linux.

disabling `CONFIG_KGDB_KDB` makes it build.

ncopa, 2024-11-29:
I also got this error:

```
  INSTALL /home/ncopa/aports/main/linux-rpi/pkg/linux-rpi/boot/bcm2711-rpi-cm4s.dtb                                                                                                 
  INSTALL /home/ncopa/aports/main/linux-rpi/pkg/linux-rpi/boot/overlays/overlay_map.dtb                                                                                             
make[2]: *** No rule to make target '/home/ncopa/aports/main/linux-rpi/pkg/linux-rpi/boot//home/ncopa/aports/main/linux-rpi/src/linux-6.12/arch/arm/boot/dts/overlays/README', neede
d by '__dtbs_install'.  Stop.                                                                                                                                                       
make[2]: *** Waiting for unfinished jobs....                                                                                                                                        
  INSTALL /home/ncopa/aports/main/linux-rpi/pkg/linux-rpi/boot/overlays/hat_map.dtb                                                                                                 
make[1]: *** [/home/ncopa/aports/main/linux-rpi/src/linux-6.12/Makefile:1424: dtbs_install] Error 2                          
make[1]: *** Waiting for unfinished jobs....      
```

popcornmix, 2024-11-29:
I've just tried and it built for me with bcmrpi_defconfig.
Are you able to build the 6.6 kernel?

My line of that file looks like (after building):
static __initdata char kdb_cmd0[] = "defcmd dumpcommon \"\" \"Common kdb debugging\"\n";

Note the extra escape `\` characters. I believe this file gets autogenerated by make scripts.
Possibly one of the tools you are using is incompatible.

pelwell, 2024-11-29:
Does `arch/arm/boot/dts/overlays/README` exist?

ncopa, 2024-11-29:
> Are you able to build the 6.6 kernel?

I assume I do. I usually do (Haven't tried since 6.6.50  though)

> Note the extra escape \ characters. I believe this file gets autogenerated by make scripts.
> Possibly one of the tools you are using is incompatible.

This is very helpful. I think this may be a buggy busybox awk. Will investigate.

> Does arch/arm/boot/dts/overlays/README exist?

Yes. huh..
```
ncopa-edge-armhf [~/aports/main/linux-rpi/src/linux-6.12]$ ls -la arch/arm/boot/dts/overlays/README 
-rw-r--r--    1 ncopa    ncopa       252599 Nov 29 13:50 arch/arm/boot/dts/overlays/README
```

popcornmix, 2024-11-29:
> Possibly one of the tools you are using is incompatible.

see `kernel/debug/kdb/Makefile`. Looks like awk is used to generate that file.

ncopa, 2024-11-29:
> see `kernel/debug/kdb/Makefile`. Looks like awk is used to generate that file.

Yes. this is definitively a bug in busybox awk. With `AWK=mawk` it builds.

> make[2]: *** No rule to make target '/home/ncopa/aports/main/linux-rpi/pkg/linux-rpi/boot//home/ncopa/aports/main/linux-rpi/src/linux-6.12/arch/arm/boot/dts/overlays/README', needed by '__dtbs_install'.  Stop.  

This one is a bug though. I'm installing with [something that corresponds to](https://gitlab.alpinelinux.org/alpine/aports/-/blob/96a2d1c6dbbdf3eae1b12e93a3e90587383d7f1b/main/linux-rpi/APKBUILD#L189):
```sh
_outdir=/home/ncopa/aports/main/linux-rpi/pkg/linux-rpi/boot
make modules_install zinstall dtbs_install \
INSTALL_MOD_PATH="$_outdir" \
INSTALL_PATH="$_outdir"/boot \
INSTALL_DTBS_PATH="$_outdir/boot"
```

Looks like make rule prefix the source for the README with the target dir.

sanderhollaar, 2025-03-07:
The build error related to README is fixed in https://github.com/raspberrypi/linux/commit/f9d8731cc63800cd85dfa653099bdc6c8147f474

Not sure if `rpi-6.12.y` is still relevant or if this issue can be closed?

closed 2025-03-07 by pelwell