raspberrypi/linux #6507: 6.12.y is missing arm-pmu device tree node on Pi5

mmozeiko, 2024-12-03:
### Describe the bug

`perf` utility is not able to use hardware PMU counters on 6.12 kernel on Raspberry Pi 5

It seems 6.12.y tree is missing pmu node in device tree:
In 6.6.y - https://github.com/raspberrypi/linux/blob/rpi-6.6.y/arch/arm64/boot/dts/broadcom/bcm2712.dtsi#L725-L732
In 6.12.y - https://github.com/raspberrypi/linux/blob/rpi-6.12.y/arch/arm64/boot/dts/broadcom/bcm2712.dtsi

### Steps to reproduce the behaviour

Run `dmesg | grep perfevents` - it does not show anything.
Or check `/sys/bus/event_source/devices/armv8_cortex_a76` folder - it does not exist.

On 6.6 `dmesg | grep perfevents` shows:
```
[ 2.317516] hw perfevents: enabled with armv8_cortex_a76 PMU driver, 7 counters available
```
and `/sys/bus/event_source/devices/armv8_cortex_a76/events/` exists and contains bunch of counter events.

I've patched my device tree file `/boot/bcm2712-rpi-5-b.dtb` by adding this section:
```
	arm-pmu {
		compatible = "arm,cortex-a76-pmu";
		interrupts = <0x00 0x10 0x04 0x00 0x11 0x04 0x00 0x12 0x04 0x00 0x13 0x04>;
		interrupt-affinity = <0x7c 0x7d 0x7e 0x7f>;
	};
```
I don't know if these values are 100% correct, I tried to guess them - I used 6.6 dts as example, and only changed interrupt-affinity to matching values for 6.12 dts file.

Then `dmesg | grep perfevents` shows:
```
[    2.338034] hw perfevents: enabled with armv8_cortex_a76 PMU driver, 7 (0,8000003f) counters available
```
and `/sys/bus/event_source/devices/armv8_cortex_a76/events/` folder contains bunch of counter events.

After this patch the `perf` works, at least on events I've tried - cpu cycles & instructions.

### Device (s)

Raspberry Pi 5

### System

I am running ArchLinux ARM with `linux-rpi-16k 6.12.1-1` package for kernel.

```
$ uname -a
Linux pi5 6.12.1-1-rpi-16k #1 SMP PREEMPT Mon Nov 25 13:04:10 MST 2024 aarch64 GNU/Linux
```

pelwell, 2024-12-03:
Yes, that got dropped in the change to using the upstream bcm2712.dtsi. The pre-release squashing that happens to the tree means I can't see where we lost it, but was probably just never added.

https://github.com/raspberrypi/linux/pull/6508 includes the obvious fix.

mmozeiko, 2024-12-10:
I've upgraded arch package that includes this fix, and `perf` now works fine. Thanks!

closed 2024-12-10 by pelwell