raspberrypi/linux #6516: PiZeroW / 6.12.1:  no txcap_blob available (err=-2)

macmpi, 2024-12-04:
### Describe the bug

Unexpected warning on dmesg at boot, while all firmware files are available.
```
dmesg | grep brcmfmac
[   20.377896] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[   20.384907] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   20.442196] usbcore: registered new interface driver brcmfmac
[   20.942412] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
[   20.943381] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Jul 19 2021 03:24:18 version 7.45.98 (TOB) (56df937 CY) FWID 01-8e14b897
[   28.965289] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
```

### Steps to reproduce the behaviour

just boot

### Device (s)

Raspberry Pi Zero W / WH

### System

`Linux alpine-headless 6.12.1-0-rpi #1-Alpine Tue Dec  3 10:15:04 UTC 2024 armv6l Linux`
kernel compiled with zst firmware compression option

### Logs

```
-rwxr-xr-x    1 root     root        256298 Nov 21 08:14 /lib/firmware/brcm/brcmfmac43430-sdio.raspberrypi,model-zero-w.bin.zst
-rwxr-xr-x    1 root     root          1812 Nov 21 08:14 /lib/firmware/brcm/brcmfmac43430-sdio.raspberrypi,model-zero-w.clm_blob.zst
-rwxr-xr-x    1 root     root           712 Nov 21 08:14 /lib/firmware/brcm/brcmfmac43430-sdio.raspberrypi,model-zero-w.txt.zst
```

pelwell, 2024-12-05:
Yes, they all do that - this is not a bug. The standard brcmfmac driver doesn't know that the devices don't require a txcap_blob, and it complains when it doesn't find one.

closed 2024-12-05 by pelwell

macmpi, 2024-12-05:
>  it complains when it doesn't find one.
That's the point: it complains whereas file is indeed there (see logs), or am I missing something?

pelwell, 2024-12-05:
> am I missing something?

Yes - `txcap_blob` != `clm_blob`.

macmpi, 2024-12-05:
Thanks: was confused by naming & thought it related to `.txt` file. Got it.