raspberrypi/linux #6517: gpu_usage and gpu_pid_usage are missing in debugfs.

xiaohuirong, 2024-12-05:
### Describe the bug

I am using Arch Linux ARM aarch64. After upgrading the `linux-rpi` to `6.1.x`, I found that the `gpu_usage` and `gpu_pid_usage` files were no longer present in `debugfs`, so I can no longer obtain GPU usage information from there.

### Steps to reproduce the behaviour

I navigated to the `/sys/kernel/debug/dri/0/` directory, but I couldn't find these two files there.

### Device (s)

Raspberry Pi 4 Mod. B

### System

vcgencmd version:
```
Nov 11 2024 15:48:13 
Copyright (c) 2012 Broadcom
version 903570ba72a9e117f92e5499de439f59dd96e417 (clean) (release) (start)
```

uname -a:
```
Linux netpi 6.12.1-3-rpi #1 SMP PREEMPT Sat Nov 30 12:09:57 MST 2024 aarch64 GNU/Linux
```

6by9, 2024-12-05:
Thanks for the report.

It looks like patch f0b0156b38d07a45e1b5309181efc645e6fe8393 (on rpi-6.6.y) hasn't been forwarded ported due to a number of conflicts. Original PR was #5456.

@txenoo I'm happy to look at fixing it up (it doesn't look too onerous), but is that patch likely to get submitted upstream? The fact it works with nvtop implies that the mechanism has been adopted by upstream.

lategoodbye, 2024-12-05:
Oh dear,
it seems @xiaohuirong upstreamed this feature to nvtop here https://github.com/Syllo/nvtop/pull/315 .

Personally I consider this problematic for two reason:
1. the v3d commit isn't mainlined yet
2. debugfs is not consider as stable ABI

popcornmix, 2024-12-05:
The stats are present in 6.12, but they have moved from:
`/sys/kernel/debug/dri/0/gpu_usage`
to
`/sys/devices/platform/axi/1002000000.v3d/gpu_stats`
(and have a slightly different format). Been like this since 6.8 kernel.

I believe our gpu stats plugin handles both.

txenoo, 2024-12-05:
> @txenoo I'm happy to look at fixing it up (it doesn't look too onerous), but is that patch likely to get submitted upstream? The fact it works with nvtop implies that the mechanism has been adopted by upstream.

The gpustats support has already landed upstream, but it is now using the standarized fdinfo for perf process stats interfaces like Intel driver is using, so each process has information about the gpu usage by each of the queues.  

And to have the access to the global_gpu status now the information is available at /sysfs/ without the debugs interface as it is supposed to be a more stable interface.

And there is already support for GPU memory usage stats since [drm/v3d: Expose memory stats through fdinfo](0b91c6dacae24a1393f99f8154037c56454c277c)

txenoo, 2024-12-05:
@mairacanal has been on top of all the upstreaming of the feature and can provide more insight of the changes.

xiaohuirong, 2024-12-05:
Thanks all. I will try to migrate `nvtop` to the new `fdinfo` interface.