raspberrypi/linux #6518: CONFIG_MPTCP_IPV6 should be enabled

bjacke, 2024-12-06:
### Describe the bug

ip mptcp endpoint can't add IPv6 addresses.

### Steps to reproduce the behaviour

The reason is that CONFIG_MPTCP_IPV6 isn't enabled for the kernel compile configuration.

### Device (s)

Raspberry Pi 5

### System

Kernel: 6.6.62+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.62-1+rpt1 (2024-11-25) aarch64 GNU/Linux

pelwell, 2024-12-06:
This is effectively a duplicate of https://github.com/raspberrypi/linux/pull/5487, which was rejected. The fundamental facts haven't changed in the last 18 months - `MPTCP_IPV6` still requires that IPV6 is built-in (`CONFIG_IPV6=y`). Last time I checked, this caused a kernel size increase of ~500kB, which is quite a punishment for the majority of people who don't use IPV6 - it's worst on small-memory platforms like Pi Zero 2.

matttbe, 2025-03-05:
Hi @pelwell 

Thank you for your reply!

@bjacke just pinged me about this ticket.

> The fundamental facts haven't changed in the last 18 months - MPTCP_IPV6 still requires that IPV6 is built-in (CONFIG_IPV6=y). Last time I checked, this caused a kernel size increase of ~500kB, [...]

Being able to support MPTCP as a module means quite a lot of development, for a very low benefit. Not being able to use MPTCP with IPv6 is quite blocking for some users. As a compromise, would it be eventually possible to have `IPV6` (+ `MPTCP_IPV6`) built-in on all platforms except the RPi Zero and the RPI 1? So not on BCM2835 and BCM2710?

Please note that having `IPV6` as a module instead of built-in also has a small extra cost for all deployments with a network connection, including the ones with just a loopback interface having assigned the `::1/128` IPv6 address. Likely most deployments? (Or at least the majority of the ones still getting updates :) )