raspberrypi/linux #6520: Kernel 6.13.0-rc1-v8-16k bcm2712-rpi-5-b.dtb issue

Dark-Sky, 2024-12-06:
### Describe the bug

Will not boot into lightdn display manager to log in with kernel 6.13.0-rc1-v8-16k's bcm2712-rpi-5-b.dtb installed on my pi5. Boots ok using v6.12 bcm2712-rpi-5-b.dtb. I also noticed that having kernel8.img set in config.txt  does the same thing trying to boot that image with the pi5.

### Steps to reproduce the behaviour

Install 6.13.0-rc1-v8-16k  kernel and it's generated bcm2712-rpi-5-b.dtb and reboot.

### Device (s)

Raspberry Pi 5

### System

Manjaro XFCE DE

[ray@jellyfin ~]$ sudo vcgencmd version
2024/11/12 16:10:44 
Copyright (c) 2012 Broadcom
version 4b019946 (release) (embedded)

Linux jellyfin 6.13.0-rc1-v8-16k #1 SMP PREEMPT Fri Dec  6 05:09:49 CST 2024 aarch64 GNU/Linux

### Logs

Dec 06 05:36:30 jellyfin systemd[1]: lightdm.service: Main process exited, code=exited, status=1/FAILURE
Dec 06 05:36:30 jellyfin systemd[1]: lightdm.service: Failed with result 'exit-code'.
Dec 06 05:36:30 jellyfin systemd[1]: lightdm.service: Scheduled restart job, restart counter is at 1.
Dec 06 05:36:31 jellyfin systemd[1]: lightdm.service: Main process exited, code=exited, status=1/FAILURE
Dec 06 05:36:31 jellyfin systemd[1]: lightdm.service: Failed with result 'exit-code'.
Dec 06 05:36:31 jellyfin systemd[1]: lightdm.service: Scheduled restart job, restart counter is at 2.
Dec 06 05:36:31 jellyfin systemd[1]: lightdm.service: Main process exited, code=exited, status=1/FAILURE
Dec 06 05:36:31 jellyfin systemd[1]: lightdm.service: Failed with result 'exit-code'.
Dec 06 05:36:31 jellyfin systemd[1]: lightdm.service: Scheduled restart job, restart counter is at 3.
Dec 06 05:36:32 jellyfin systemd[1]: lightdm.service: Main process exited, code=exited, status=1/FAILURE
Dec 06 05:36:32 jellyfin systemd[1]: lightdm.service: Failed with result 'exit-code'.
Dec 06 05:36:32 jellyfin systemd[1]: lightdm.service: Scheduled restart job, restart counter is at 4.

### Additional context

I have only tested with the pi5 device. Do not know if the same issue exists on the other pi devices and their .dtb's with this kernel.

pelwell, 2024-12-09:
Thanks for the report. It is working for me with X.Org on 6.130-rc1 with the latest dtb and either the 2711 or 2712 defconfig, which points to there being something different about your configuration.

The fbdev failures suggests the kernel driver has failed - what does `dmesg -l err,warn` show?

6by9, 2024-12-09:
Not using fbdev is expected as it should be modesetting (and appears to be doing so and finding a load of modes).

Dark-Sky, 2024-12-09:
> Thanks for the report. It is working for me with X.Org on 6.130-rc1 with the latest dtb
 
I had a feeling you would say that. lol This has happened in the past at times and I believe it had to do with my VIZIO tv then. Hooking up another monitor and it worked.

> The fbdev failures suggests the kernel driver has failed - what does `dmesg -l err,warn` show?

That warning has always been there. I always thought it would try fbdev 1st then move on. I did try using fbdev and fkms and I was able to log in and boot to desktop but there was a ton of artifacts and my system would lock up.

dmesg with new .dtb:

```
[    2.696013] genirq: irq_chip rp1_irq_chip did not update eff. affinity mask of irq 99
[    8.135554] rpivid_hevc: module is from the staging directory, the quality is unknown, you have been warned.
[    8.136944] platform regulatory.0: loading /lib/firmware/6.13.0-rc1-1-MANJARO-RPI5/regulatory.db failed with error -20
[    8.158678] platform regulatory.0: loading /lib/firmware/6.13.0-rc1-1-MANJARO-RPI5/regulatory.db.p7s failed with error -20
[    8.249097] pispbe 1000880000.pisp_be: Runtime PM usage count underflow!
[    8.345785] hci_uart_bcm serial0-0: supply vbat not found, using dummy regulator
[    8.354092] hci_uart_bcm serial0-0: supply vddio not found, using dummy regulator
[    8.486061] FAT-fs (sda1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
```

dmesg with older .dtb that boots ok.

```
[    2.680614] genirq: irq_chip rp1_irq_chip did not update eff. affinity mask of irq 95
[    7.939245] rpivid_hevc: module is from the staging directory, the quality is unknown, you have been warned.
[    8.019375] platform regulatory.0: loading /lib/firmware/6.13.0-rc1-1-MANJARO-RPI5/regulatory.db failed with error -20
[    8.031205] platform regulatory.0: loading /lib/firmware/6.13.0-rc1-1-MANJARO-RPI5/regulatory.db.p7s failed with error -20
[    8.166815] pispbe 1000880000.pisp_be: Runtime PM usage count underflow!
[    8.556989] hci_uart_bcm serial0-0: supply vbat not found, using dummy regulator
[    8.625340] hci_uart_bcm serial0-0: supply vddio not found, using dummy regulator
[    8.701881] FAT-fs (sda1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[   14.825761] brcmfmac: brcmf_set_channel: set chanspec 0xd022 fail, reason -52
[   14.949727] brcmfmac: brcmf_set_channel: set chanspec 0xd026 fail, reason -52
[   15.061733] brcmfmac: brcmf_set_channel: set chanspec 0xd02a fail, reason -52
[   15.169716] brcmfmac: brcmf_set_channel: set chanspec 0xd02e fail, reason -52
```

Dark-Sky, 2024-12-09:
I just built 6.13-rc2 and it has the same issue here. With older .dtb it works ok.

Dark-Sky, 2024-12-13:
@6by9 I found the problem. I was using the overlays from raspberrpi-firmware git. I did not know evidently something changed in the overlays in kernel's 7.12/6.13 also to go along with their new pi5 dtb's. I used to provide the overlays from the kernel being compiled until the pi5 came out and had to add the second 16k kernel to our images as pacman will not allow the same files in 2 different packages.

popcornmix, 2024-12-13:
The kernel, the modules, the dtbs and dtbos must all be built from the same kernel tree.
If any of these files is stale after an update, there is no guarantee it will even boot.

closed 2024-12-13 by Dark-Sky