raspberrypi/linux #6521: missing TAG in github-repo for newest installed kernel

berndbenner, 2024-12-08:
### Describe the bug

if I install a new RPI5 with raspberry pi imager and I update all packages a kernel  6.6.62+rpt-rpi-2712 is installed.
I am missing a TAG in the kernel git-hub repo to download or customize this kernel. The newest TAGED kernel is 6.6.51. 

### Steps to reproduce the behaviour

Install a new RPI5

### Device (s)

Raspberry Pi 5

### System

there is only a git-hash reported 
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 891df1e21ed2b6099a2e6a13e26c91dea44b34d4, stage4

### Logs

*

### Additional context

*

XECDesign, 2024-12-09:
What do you expect the tag name to be? What should it correspond to?

ragazenta, 2024-12-10:
I saw https://github.com/RPi-Distro/linux-packaging/commit/6c7657c5742509e52ecd1465ee7aef5353cd45a6

So the tag name should be `stable_20241120`? Because this commit dd2394360860d15146c96635796a75b05bb32b61 was dated on Nov 20 2024.