raspberrypi/linux #6523: probe with driver rp1-pio failed with error -110 in 6.12.y kernel

graysky2, 2024-12-09:
### Describe the bug

I have a new dmesg error when booting into a kernel built from the rpi-6.12.y branch:
```
rp1-pio 1f00178000.pio: probe with driver rp1-pio failed with error -110
```

This is not present on a recent kernel built from rpi-6.6.y

### Steps to reproduce the behaviour

On Arch ARM, boot RPi5B into https://github.com/archlinuxarm/PKGBUILDs/commit/7ff60d41d0853434321551e38ad6788d59149116 then check `dmesg`

### Device (s)

Raspberry Pi 5

### System

* OS = Arch ARM
* firmware = `2024/09/10 14:40:30 
Copyright (c) 2012 Broadcom
version 5be4f304 (release) (embedded)
`
* Kernel = `Linux submyth 6.6.62-1-rpi #1 SMP PREEMPT Mon Dec  9 09:29:55 EST 2024 aarch64 GNU/Linux`

### Logs

Full [dmesg](https://gist.github.com/graysky2/c7623583a237065ab88241149de58cf3) output

graysky2, 2024-12-09:
Ah!  this was fix when updating my firmware to 
```
2024/11/12 16:10:44 
Copyright (c) 2012 Broadcom
version 4b019946 (release) (embedded)
```

closed 2024-12-09 by graysky2

pelwell, 2024-12-09:
You beat me to it!