raspberrypi/linux #6538: Waveshare 5inch Capacitive DSI Touchscreen - Touch not always working on boot

mattie47, 2024-12-16:
### Describe the bug

Hi,

Firstly, I'm aware this may not be the most appropriate place to raise this issue, however feel it has the right coverage of individuals to assist in troubleshooting and narrowing down what might be going wrong. I have also sent this thread to Waveshare support as currently I've been told from them to only "not touch the screen during bootup" which is not my issue. 

I have about 20x RPI 4s using the https://www.waveshare.com/5inch-dsi-lcd-b.htm screen

I've found since updating a number of these Pis from Buster to Bookworm that I'm frequently getting the touch driver failing to register any touch input after booting up the Pi. 

Typically a restart or complete disconnect/reconnect of RPI power result in the touch working on the next boot, however this has become a bit of a nightmare for reliability with end-users of our internal tool.

I'm not sure what the last kernel was that didn't appear to have issues, however I've had to update to `6.6.62+rpt-rpi-v8` due to a CAN Bus issue on recent previous kernels. 

I am using the `dtoverlay=vc4-kms-v3d` overlay which is what @waveshare say to use on their wiki, however from other posts on the forum I understand that waveshare ideally should be using their own dtoverlay rather than the same one as the official RPI screens. 

I strongly suspect I'm hitting the same problem or similar issue as https://github.com/raspberrypi/linux/issues/6298 

I also have attempted to run `evtest` (with results below), however in the same case as @kyngs there are no logs/debug when touching the screen. 

<details>
<summary>evtest logs</summary>

```
 sudo evtest
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:      soc:shutdown_button@12
/dev/input/event1:      vc4-hdmi-0
/dev/input/event2:      vc4-hdmi-0 HDMI Jack
/dev/input/event3:      vc4-hdmi-1
/dev/input/event4:      vc4-hdmi-1 HDMI Jack
/dev/input/event5:      10-0038 generic ft5x06 (79)
Select the device event number [0-5]: 5
Input driver version is 1.0.1
Input device ID: bus 0x18 vendor 0x0 product 0x0 version 0x0
Input device name: "10-0038 generic ft5x06 (79)"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 330 (BTN_TOUCH)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value      0
      Min        0
      Max      799
    Event code 1 (ABS_Y)
      Value      0
      Min        0
      Max      479
    Event code 47 (ABS_MT_SLOT)
      Value      0
      Min        0
      Max        9
    Event code 53 (ABS_MT_POSITION_X)
      Value      0
      Min        0
      Max      799
    Event code 54 (ABS_MT_POSITION_Y)
      Value      0
      Min        0
      Max      479
    Event code 57 (ABS_MT_TRACKING_ID)
      Value      0
      Min        0
      Max    65535
Properties:
  Property type 1 (INPUT_PROP_DIRECT)
Testing ... (interrupt to exit)

[... 747 lines cut ...]
[    6.903252] systemd-journald[513]: Received client request to flush runtime journal.
[    6.903392] systemd-journald[513]: File /var/log/journal/9a5616e8b9f34b1d8311a4b6d6151e82/system.journal corrupted or uncleanly shut down, renaming and replacing.
[    7.534769] Adding 524284k swap on /var/swap.  Priority:-2 extents:7 across:5849088k SS
[    7.690501] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    7.690516] Bluetooth: BNEP filters: protocol multicast
[    7.690528] Bluetooth: BNEP socket layer initialized
[    7.697226] Bluetooth: MGMT ver 1.22
[    7.715079] NET: Registered PF_ALG protocol family
[    8.508481] hwmon hwmon1: Voltage normalised
[    8.690313] bcmgenet fd580000.ethernet: configuring instance for external RGMII (RX delay)
[    8.693592] bcmgenet fd580000.ethernet eth0: Link is Down
[    8.725521] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[    9.763178] tun: Universal TUN/TAP device driver, 1.6
[   12.796815] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   14.565358] systemd-journald[513]: File /var/log/journal/9a5616e8b9f34b1d8311a4b6d6151e82/user-1000.journal corrupted or uncleanly shut down, renaming and replacing.
[   15.561198] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
[   16.141788] Bluetooth: RFCOMM TTY layer initialized
[   16.141818] Bluetooth: RFCOMM socket layer initialized
[   16.141843] Bluetooth: RFCOMM ver 1.11
[   19.143884] warning: `iwgetid' uses wireless extensions which will stop working for Wi-Fi 7 hardware; use nl80211
[   19.512785] can: controller area network core
[   19.512921] NET: Registered PF_CAN protocol family
[   19.521629] can: raw protocol
```
</details>

6by9, 2024-12-16:
Buster will have been running with the firmware based drivers by default.
Bookworm has switched to kernel drivers for almost everything, including displays and touch controllers.

The kernel will be talking to the touch controller over I2C assuming it is an EDT FT5406 or 100% compatible. I don't know what Waveshare have used on their display. The fact the product page says 5 points of touch implies it isn't an FT5406, as those generally support 10 points.

I do happen to have one of those displays, so will have a quick look.
If waveshare fancy giving any details on what they do actually do with the touch controller, then we may be able to work out a solution.

pelwell, 2024-12-16:
```
[    8.508481] hwmon hwmon1: Voltage normalised
```
It may not be the cause of the problem, but this message suggests that the power supply is inadequate for the load during boot.

6by9, 2024-12-16:
Reproduced.
The Pi is still polling the touch controller, and it's responding but saying no touch events. Nothing the Pi can do about that one.

If I were to hazard an educated guess, then I'd suspect that the touch controller is actually a GT911. On that chip the I2C address is set by the state of the IRQ line at boot, and then it changes from being an input to an output in order to function as the IRQ signal.
If there was an interrupt active just before reboot, then the GT911 will have been pulling it high. Capacitance can mean that the line is still read as high when the I2C address is determined, and suddenly it isn't on the expected address. (Logic states may be inverted, but the theory stands).
We initially had exactly the same issue on the Touch Display 2.

mattie47, 2024-12-16:
Thank you @6by9 for your analysis and feedback!

I have passed this on to Waveshare support, as well as your ability to reproduce. I'm hoping they engage in this thread soon.

Their support had asked me if I was willing to replace/resolder the chips on impacted boards however I have these in remote locations with some quite frustrated users.

I personally do not see this as a solution as I'm not the only user facing this problem, and it needs a real fix by them to address.

_If there was an interrupt active just before reboot, then the GT911 will have been pulling it high_

It's worth noting I often see this on first boot (i.e. power has also just been connected). 

**In terms of the main chips:**

_Note, only screens 1 & 2 have been tested below, but listing three showing the variation of chips used_

Screen 1 - QC17 Sticker

ICN6211 A56456 2126

WSDIA2 2220
WSTPA0 2215

Screen 2 - QC17 Sticker

ICN6211 D00090 2130

WSDIA2 2220
WSTPA0 2247

Screen 3 - QC01 Sticker

ICN6211 A56456 2126

WSDIA2 2208
WSTPA0 2142

Thanks,

Matt

mattie47, 2024-12-27:
Just to update Waveshare's latest response:

```It is a software problem. Normally, we will update the software response of the touch chip to fix this problem.```

I will update the thread with any more info.

Unfortunately Waveshare are suggesting either replacing the chips or screen replacement, with unfortunately neither choice an option for me.

pelwell, 2024-12-28:
I wonder if this PR is for the replacement? https://github.com/raspberrypi/linux/pull/6566

kyngs, 2025-01-01:
> ### Describe the bug
> 
> Hi,
> 
> Firstly, I'm aware this may not be the most appropriate place to raise this issue, however feel it has the right coverage of individuals to assist in troubleshooting and narrowing down what might be going wrong. I have also sent this thread to Waveshare support as currently I've been told from them to only "not touch the screen during bootup" which is not my issue.
> 
> I have about 20x RPI 4s using the https://www.waveshare.com/5inch-dsi-lcd-b.htm screen
> 
> I've found since updating a number of these Pis from Buster to Bookworm that I'm frequently getting the touch driver failing to register any touch input after booting up the Pi.
> 
> Typically a restart or complete disconnect/reconnect of RPI power result in the touch working on the next boot, however this has become a bit of a nightmare for reliability with end-users of our internal tool.
> 
> I'm not sure what the last kernel was that didn't appear to have issues, however I've had to update to `6.6.62+rpt-rpi-v8` due to a CAN Bus issue on recent previous kernels.
> 
> I am using the `dtoverlay=vc4-kms-v3d` overlay which is what @waveshare say to use on their wiki, however from other posts on the forum I understand that waveshare ideally should be using their own dtoverlay rather than the same one as the official RPI screens.
> 
> I strongly suspect I'm hitting the same problem or similar issue as #6298
> 
> I also have attempted to run `evtest` (with results below), however in the same case as @kyngs there are no logs/debug when touching the screen.
> evtest logs
> 
> Is there anything further I can do for troubleshooting or assisting with testing towards a potential fix?
> 
> I believe this is a software issue, and not hardware due to many of my screens/rpi4 combos not having touch issues prior to the bookworm update.
> 
> Thanks,
> 
> Matt
> ### Steps to reproduce the behaviour
> 
>     * Boot up RPI 4 with https://www.waveshare.com/5inch-dsi-lcd-b.htm connected
> 
>     * Observe touching the screen after bootup does nothing.
> 
> 
> ### Device (s)
> 
> Raspberry Pi 4 Mod. B
> ### System
> 
> ```
> cat /etc/rpi-issue
> Raspberry Pi reference 2024-11-19
> Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, b15527e15ba4d9e65126a454bfec2040b909dc1c, stage4
> ```
> 
> ```
> vcgencmd version
> Nov 26 2024 12:52:28
> Copyright (c) 2012 Broadcom
> version 2ae30f53898ae2f1ba77ff570a92991bedfb0398 (clean) (release) (start)
> ```
> 
> ```
> uname -a
> Linux VHS-20 6.6.62+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.62-1+rpt1 (2024-11-25) aarch64 GNU/Linux
> ```
> 
> /boot/firmware/config.txt
> 
> ### Logs
> dmesg
> 
> ### Additional context
> 
> _No response_

Hi, why can't you use the `dtoverlay=rpi-ft5406` overlay? I don't want to use it, because it forces me to use the legacy raspicam, but do you have a reason not to?

kyngs, 2025-01-01:
> I wonder if this PR is for the replacement? #6566

So this PR won't solve this issue?

mattie47, 2025-01-01:
>Hi, why can't you use the dtoverlay=rpi-ft5406 overlay? I don't want to use it, because it forces me to use the legacy raspicam, but do you have a reason not to?

My main reasoning is:
* it's not what waveshare officially recommend. I have not had time to test long term reliability and that it works OK without other areas of impact. 
* These devices I have are in remote locations globally and if anything breaks further, it's cost intensive to make changes
* whilst I'm not opposed to that overlay change, I haven't had enough time to understand any implications of using it. I know there are the raspi cam issues, but these pis also have a number of i2c, spi, and general gpio peripherals connected that I wasn't sure if there might be other issues. 
* given the amount of items I've purchased (not just screens) from waveshare, I would expect after sales service to officially ensure the product continues to work reliably. 

Regarding the PR, yeah I'm hoping that's a fix, but I haven't anything back from their support team to suggest that yet. 

Thanks,
Matt

kyngs, 2025-01-01:
> My main reasoning is:
>
>     * it's not what waveshare officially recommend. I have not had time to test long term reliability and that it works OK without other areas of impact.
>
>     * These devices [...]

My basic understanding is (tho this may be completely wrong):
the rpi-ft5406 is a raspberry pi touchscreen driver
the edt_ft5x06 (which I believe you're using at the moment), is the linux kernel touchscreen driver and apparently does not work with waveshare panels.

From my findings, the rpi-ft5406 driver works perfectly with the touchscreen. However, my original issue is that it cannot be used with the modern camera system (in my case, with the imx477 driver) but only with the legacy camera. 
While on the other hand, the edt_ft5x06 driver can work with the imx477 driver.

Do you have a testing device on hand where you could test whether the rpi-ft5406 works for you?

>  I would expect after sales service to officially ensure the product continues to work reliably.

Yeah I wish, tho the panels are quite old. Hopefully the PR they sent in will fix this. 
Tho this from the PR worries me: "This is the driver for the DSI-TOUCH series of panels **to be released** by Waveshare." This implicates that the PR is actually for an upcoming series of displays.

6by9, 2025-01-01:
#6566 is unrelated - that is for their new screens, not amending support for their old ones.

rpi-ft5406 uses mailbox calls to the firmware to poll the touchscreen for when the firmware is also using the I2C interface (eg the legacy camera stack).
edt-ft5x06 is the kernel directly talking I2C to the touch panel.

If the Waveshare panel were a genuine EDT FT5x06 touch controller, then either will work quite happily. As I commented in https://github.com/raspberrypi/linux/issues/6538#issuecomment-2545769584, I suspect that they are emulating it through a microcontroller, and that emulation is flawed. The same flaw would normally affect both drivers, so there is possibly something to investigate there.

The two drivers do use slightly different approaches to communicating with the touch controller. [There was a thread back in 2021](https://lore.kernel.org/all/CAJ+vNU0qfYGekS2+sCXFM6Xc+dsu_pAUm3E7AhoWhPSrRy523A@mail.gmail.com/) where DFRobot panels also tried to do the same, but implemented even less of the FT5x06 protocol to the point that it only worked with the firmware driver. Patches at https://patchwork.kernel.org/project/linux-input/patch/20211216233041.1220-4-tharvey@gateworks.com/ were suggested as a workaround there.

If I were to investigate this (which I'm not overly planning on doing as it's not a Raspberry Pi product for me to support) then I'd probe the connections on the smaller 6 pin connector going into the panel (on the left in the photos above), as that is going to be the comms for touch. I'd expect to have power, ground, SDA, SCL, reset, and interrupt.  Almost any oscilloscope these days will decode the I2C for you, so you can compare to the FT5x06 protocol. The main tell-tale I'm expecting is that when it fails you get no ACKs to any I2C read. That would match with my guess that it is a GT911 which has changed I2C address due to incorrect handling of the interrupt line.

Once there is an understanding of exactly what the problem is within the display and why it's not giving correct touch data, then there can be discussion about possible workarounds in the kernel to mitigate the limitations (delays, amending I2C calls, or something else).

kyngs, 2025-01-02:
> #6566 is unrelated - that is for their new screens, not amending support for their old ones.
> 
> rpi-ft5406 uses mailbox calls to the firmware to poll the touchscreen for when the firmware is also using the I2C interface (eg the legacy camera stack). edt-ft5x06 is the kernel directly talking I2C to the touch panel.
> 
> If the Waveshare panel were a genuine EDT FT5x06 touch controller, then either will work quite happily. As I commented in [#6538 (comment)](https://github.com/raspberrypi/linux/issues/6538#issuecomment-2545769584), I suspect that they are emulating it through a microcontroller, and that emulation is flawed. The same flaw would normally affect both drivers, so there is possibly something to investigate there.
> 
> The two drivers do use slightly different approaches to communicating with the touch controller. [There was a thread back in 2021](https://lore.kernel.org/all/CAJ+vNU0qfYGekS2+sCXFM6Xc+dsu_pAUm3E7AhoWhPSrRy523A@mail.gmail.com/) where DFRobot panels also tried to do the same, but implemented even less of the FT5x06 protocol to the point that it only worked with the firmware driver. Patches at https://patchwork.kernel.org/project/linux-input/patch/20211216233041.1220-4-tharvey@gateworks.com/ were suggested as a workaround there.
> 
> If I were to investigate this (which I'm not overly planning on doing as it's not a Raspberry Pi product for me to support) then I'd probe the connections on the smaller 6 pin connector going into the panel (on the left in the photos above), as that is going to be the comms for touch. I'd expect to have power, ground, SDA, SCL, reset, and interrupt. Almost any oscilloscope these days will decode the I2C for you, so you can compare to the FT5x06 protocol. The main tell-tale I'm expecting is that when it fails you get no ACKs to any I2C read. That would match with my guess that it is a GT911 which has changed I2C address due to incorrect handling of the interrupt line.
> 
> Once there is an understanding of exactly what the problem is within the display and why it's not giving correct touch data, then there can be discussion about possible workarounds in the kernel to mitigate the limitations (delays, amending I2C calls, or something else).

Hi, thanks for the analysis. I personally possess no skills, or equipment to fix this issue, so I guess I'm out of luck. I would love to use the official display, but afaik it's only made in 7", and I need 5" max (my current 4.3" is ideal).
A bit offtopic: do you have an idea if I can somehow get the imx477 and rp-ft5406 drivers working together?

mattie47, 2025-01-02:
>I'd expect to have power, ground, SDA, SCL, reset, and interrupt.

I have one of those cheap usb logic analyzers so might try capturing a
"good" vs "bad" case and attach to this issue.

Do I just need the sda scl and Gnd? Or also the interrupt line?

Thanks

Eng027, 2025-01-02:
> #6566 is unrelated - that is for their new screens, not amending support for their old ones.
>
> rpi-ft5406 uses mailbox calls to the firmware to poll the touchscreen for when the firmware is also [...]

Yes, as you mentioned earlier, if the touch signals are active before the Bookworm system fully boots, the GT911 levels may behave abnormally, leading to occasional errors. 

We have adjusted the MCU program to handle this anomaly and are conducting comprehensive testing. We expect the results soon. Once the testing is complete and successful, we will do our utmost to provide appropriate after-sales support for affected customers.

mattie47, 2025-01-03:
Thanks for the update @eng027!

Much appreciated. 

> We have adjusted the MCU program to handle this anomaly and are conducting comprehensive testing. We expect the results soon.

Do you expect the fix to be purely in the MCU/GT911? In other words any impacted device would require a hardware modification fix?

Alternatively perhaps a way of upgrading the MCU's FW?

Or is there likely a "software workaround" in the kernel? 

Thanks,

Matt

riban-bw, 2025-02-16:
I found that touching the panel (and hence triggering some comms) before the device driver (edt_ft5x06) is loaded will trigger this issue. I guess the touch controller fails to receive ack or similar condition and then refuses to behave. Waveshare support have been looking at this since I reported it to them almost a year ago and have recently said this can only be fixed with a hardware change, replacing the controller chip which requires desoldering the old and soldering the new - quite drastic. It is a shame that the chip can't be reset by the driver.

In case anyone is struggling to reproduce the issue, here is my workflow to trigger the fault:

- Remove the driver, e.g. `rmmod edt_ft5x06`.
- Touch and drag on the screen.
- Install the driver, e.g. `modprobe edt_ft5x06`.

What diagnostics could I perform to assist the driver maintainer to possibly implement a software workaround?

dsamarin1, 2025-02-27:
I have this same issue with the same LCD on Pi 4B, and am in a similar situation as the OP in terms of number of units in distant places. I can faithfully recreate this issue on my touchscreen. According to dmesg, I am using the ft5x06 driver. I am in contact with Waveshare support but they have not provided any details to me yet, only asking clarifying questions.

Commenting here to keep track and hopefully shine more light on this issue. Based on my mechanical configuration, many users inadvertently touch the touchscreen while plugging in power. I can ask them not to, but that is very poor product design.

6by9, 2025-02-27:
> Commenting here to keep track and hopefully shine more light on this issue. Based on my mechanical configuration, many users inadvertently touch the touchscreen while plugging in power. I can ask them not to, but that is very poor product design.

I'll stress that this is not an issue that is under Raspberry Pi's control, and is an issue in the Waveshare panel.
We're happy to work with 3rd parties to work around these issues where they are understood, but it's not our design to fix. I'm suspecting that there is a failure mode with the legacy drivers too, but the timing is different and so isn't hit as easily.

@riban-bw 
> This only occurs on RPi5. RPi4 does not suffer this issue.

Please be aware that editing comments does not notify participants on an issue. I'd therefore read your original post which didn't have this comment, and only saw it due to dsamarin1's update today.

Others are reporting that they have the issue on Pi4. Are you sure you can't reproduce on Pi4?

dsamarin1, 2025-02-27:
Hi @6by9, I totally appreciate this isn't a Raspberry Pi org problem, it is a 3rd party vendor problem. There doesn't seem to be any open discussion happening elsewhere online for us to latch on to, since Waveshare doesn't have user forums, and their official support communications (wiki, email) are lacking. Would be great if those of us having this issue could continue to collaborate here!

Latest email reply to me from Waveshare:
> Hi,
> Sorry for the late reply.
> We have arranged engineers to conduct tests, and it will take some time to see if the problem you mentioned can be reproduced.
> Thank you for your understanding. 
> We will inform you as soon as we have any news.

riban-bw, 2025-02-27:
I have validated that this also affects RPi4 but the boot process / timing on my OS differs so it is less likely to occur during boot. If I unload driver, touch screen then load driver, the fault occurs. This is true with legacy driver.

Waveshare have informed me that this is due to an error in their interface chip (probably their EDT FT5x06 clone) and a fix involves replacing the chip. They are sure this cannot be resolved with software.

kyngs, 2025-02-27:
For me, it occurs on Pi 4B, at all times. No matter the timing.
I've already received the allegedly fixed MCU from Waveshare. However, it's way tinier than I expected (4x6mm!), and I can't tell when, or if will I be able to solder it.

dsamarin1, 2025-02-28:
I have the ability to de/solder such a chip. I'll ask them if they can send me one or two to test.

If the chip looks the same, presumably this means they solved it with a firmware update perhaps? I wonder if this means we could create a jig to reflash the chip in-situ.

mattie47, 2025-02-28:
I had an update in January (which I only skim read at the time unfortunately) and was advised that they may be able to send a programmer with firmware update to address the issue, rather than resolding chips. 

I'm awaiting a response to the email. 

I unfortunately recently was made redundant due to the company I did this work for leaving the New Zealand market, so please excuse me for being less intentional on this issue than previously.

kyngs, 2025-03-01:
> I have the ability to de/solder such a chip. I'll ask them if they can send me one or two to test.
> 
> If the chip looks the same, presumably this means they solved it with a firmware update perhaps? I wonder if this means we could create a jig to reflash the chip in-situ.

Yeah I assume that they only changed the firmware. To be fair, they did ask me upfront whether I could solder the chip myself, and I told them yes. (I kinda expected it would be smth like atmega328 size, not a few mm)

Reprogramming the chip in place would be best, but I don't posses a programmer to do so.

mattie47, 2025-03-02:
Hi all,

I've spent some time today trying to understand the WSTPA0 (Waveshare Touch Panel A0) chip. 

I thought I'd share my findings here.

**WSTPA0** 
Footprint - TSSOP-20

**Pin out information:**

[image]

FFC = 6 Pin Flat Flex Connector

**1 - FFC5 - INT**
**2 - FFC1 - SCL**
**3 - FFC2 - SDA**
4 - x
5 - x
6 - x
7 - x
**8 - FFC4 - RST**
9 - x
10 - x

11 - x
12 - x
13 - x
14 - TP4 - ???
15 - TP2 - ???
16 - x
17 - x
18 - TP3 - GND
19 - TP1 - PWR
20 - TP1 - PWR

TP (Test Points):

4  - 14 - ???
3 - GND
2 - 15 - ???
1 - 19/20 - PWR

I then soldered on several wires and connected up my cheap logic analyzer.

[image]

As per the GT911 [Datasheet](https://www.fortec-integrated.de/fileadmin/pdf/produkte/Touchcontroller/DDGroup/GT911_Datasheet.pdf) and [Programming Guide](https://www.crystalfontz.com/controllers/datasheet-viewer.php?id=478), we see it use Write Address 0xBA, and Read Address 0xBB:

I then recorded a good case VS bad case i2c capture **(Note, bad case is following the figure touch during boot up process, not necessarily the same "bad case" I first reported in this issue)**

[tests.zip](https://github.com/user-attachments/files/19041937/tests.zip)

You can use [PulseView](https://sigrok.org/wiki/PulseView) to open them.

**What I don't understand:**
* What this chip actually talks to elsewhere on the board. I didn't take it off to see where the rest of the comms it does goes. 
* Is this chip (whilst marked as WSTPA0) actually a GT911 in disguise (That's what I've understood)
  *  Therefore, as a chip is it then emulating the [FT5x06's](https://www.displayfuture.com/Display/datasheet/controller/FT5x06.pdf) protocol, or is this likely taking place by another chip on the LCD's board? 
  * I ask because we see i2c addresses 
    * 0xBA and 0xBB on the FFC connector
    * 0x38 on the RPI for the 

```
sudo i2cdetect 10
I will probe file /dev/i2c-10.
I will probe address range 0x08-0x77.
Continue? [Y/n]
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- --
40: -- -- -- -- -- UU -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

Thanks,

Matt

6by9, 2025-03-03:
My expectation is that the touch controller on the actual panel is a GT911. The Pi is expecting to see an FT5406.
The little MCU is doing protocol conversion to match the 2. It's also likely to be generating the PWM for the backlight, and probably the GPIO for panel and touch resets.

I haven't looked at your traces, but will do in due course.

Three options come to mind:
- Fit a pull down resistor between INT (FFC pin 5) and ground. 
- Rework the MCU firmware so that it pulls FFC pin 5 to ground as it asserts RESET (FFC pin 4).
- Link the GT911 I2C to the main I2C from the Pi, and creating a new DT overlay for these devices. It'll break compatibility with the firmware drivers, but potentially avoid the dodgy I2C addressing. If you look at [Hyperpixel 4 overlay](https://github.com/raspberrypi/linux/blob/rpi-6.6.y/arch/arm/boot/dts/overlays/vc4-kms-dpi-hyperpixel4-overlay.dts#L21-L44) you'll see that they've done a similar fudge in creating devices at both 0x14 and 0x5d (7bit I2C addressing) because they obviously hit this same issue.

delphimon, 2025-03-05:
Just found this - I loaded Bookworm onto my RP4 using Waveshare 4.3" DSI 800x480 touchscreen. I've been debugging for the last two days. Was working previously so I assume it is the same issue - everything looks like it detects but no events ever fire for touchscreen. Like @kyngs, this happens all the time; it is not dependent on timing. I will see if the hack above of using the older rpi-ft5406 driver works - I'm not using camera, this is a network management box for me.

Tiger0215, 2025-03-07:
> Quick side note for those curious and need a solution fast: [DFRobot's 5-inch touch display](https://www.dfrobot.com/product-2791.html) does not exhibit this issue, at least according to one short evening of testing in my setup. It was plug-and-play replacement for the Waveshare display for me.

Compared to the $50 800×480 DSI display you recommended, I chose Waveshare's latest generation 5inch DSI LCD (C). It comes at a lower price, offers a higher resolution of 1024×600, and is also plug-and-play. 
So far, I haven't encountered any issues: https://www.waveshare.com/5inch-dsi-lcd-c.htm.

dsamarin1, 2025-03-11:
> My expectation is that the touch controller on the actual panel is a GT911.

Confirmed. I recently got a returned device with a smashed touchscreen, pulled it apart to check the touch controller. See image.

[image]

riban-bw, 2025-05-24:
Waveshare sent me a suface mount chip. I was able to use a reflow tool to remove the old chip and solder the new one and it fixed the issue with touch failing if the screen is touched whilst the kernel driver is not loaded, e.g. during boot. Not a very accessible fix for many but, if you have the tools and time, you could request the chip from Waveshare support.

riban-bw, 2025-05-25:
My old chip was 2412. The new chip is 2451.

6by9, 2025-08-04:
> From Chapter 6 of the GT911 specification about I2C: "It is strongly recommended that transmission rate be kept at or below 400Kbps." For my system I added to the device tree, where that I2C bus is declared: clock-frequency = <400000>; Without specifying a limit, in case of a rk3588, the speed would be much higher and the GT911 cannot keep up. Perhaps an I2C frequency limit declaration in the Rpi' device tree will do the trick.

The Pi default I2C bus rate is 100kHz, so that is not an issue here.

I don't have the Waveshare schematics, but based on our own use of GT911 I believe that the issue is not controlling the INT line correctly during boot. As detailed in section 6.1 of the GT911 datasheet, selecting between I2C addresses 0xBA/0xBB and 0x28/0x29 (0x5d and 0x14 in 7bit addressing) is done by holding the INT line either high (for 0x14) or low (for 0x51) for 5ms around when the reset line is asserted. If there isn't explicit control of the INT line, then it can end up in either state, and the I2C address isn't the one expected so comms fails.
Either driving the int line as an output whilst releasing reset, or having a pull resistor to define the state, solves the issue.