raspberrypi/linux #6548: Allo Piano DAC 2.1 missing alsa controls

Sunfish-2, 2024-12-18:
### Describe the bug

Running a rpi4 with Allo Piano 2.1 dac and Allo Kali reclocker.
Streamer is MoOde audio version 9.x.x. kernel 6.6.5 With this kernel the Rpi is locking, no response what so ever.
I bumped the kernel to include the "Fix volume limit locking" patch (RPiOS: 12.8 Bookworm 64-bit | Linux: 40 64-bit)
[https://github.com/raspberrypi/linux/com...7156a1a4b4](https://github.com/raspberrypi/linux/commit/3f85231cf3f332be7f016e5efd3d6d7156a1a4b4)
Now the RPi is responding and producing sound.
Alsamixer command reports : "This sound device does not have any controls."

### Steps to reproduce the behaviour

Running a rpi4 with Allo Piano 2.1 dac and Allo Kali reclocker.
Streamer is MoOde audio version 9.x.x. kernel 6.6.5
Bump the kernel to include the "Fix volume limit locking" patch (RPiOS: 12.8 Bookworm 64-bit | Linux: 40 64-bit)
[https://github.com/raspberrypi/linux/com...7156a1a4b4](https://github.com/raspberrypi/linux/commit/3f85231cf3f332be7f016e5efd3d6d7156a1a4b4)
Alsamixer command fails

### Device (s)

Raspberry Pi 4 Mod. B

### System

cat /etc/rpi-issue:
Raspberry Pi reference 2024-11-26
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 6fb311e7fb57b4e7980ae3270c25b569b19dfe8f, stage3

vcgencmd version:
Dec  5 2024 11:45:37
Copyright (c) 2012 Broadcom
version 03554ca336a03ace164f36755144e0d8c060062d (clean) (release) (start)

uname -a:
Linux raspberrypi 6.6.66-v8+ #1830 SMP PREEMPT Mon Dec 16 14:40:26 GMT 2024 aarch64 GNU/Linux

### Additional context

With kernel 6.6.5 after selecting dtoverlay=allo-piano-dac-plus-pcm512x-audio for Allo Piano 2.1 the system comes in a locked status during boot. (Deadlock)
After bumping the kernel the boot process completes leaving a working system

pelwell, 2024-12-18:
It would be helpful if you could resurrect an old image and make a list of the controls which used to be available (`amixer scontrols`?).

@foonerd Does this chime with your experience?

moodeaudio, 2024-12-19:
I had to step into the Wayback Machine but here's some information from my notes on the vintage Piano 2.1.

__Piano 2.1 DSP firmware files__
I don't recall whether these files are required to expose the additional scontrols but IIRC they are needed to set the Subwoofer params.
```
wget https://github.com/allocom/piano-firmware/archive/master.zip
sudo unzip -q master.zip
sudo cp -r ./piano-firmware-master/lib/firmware/allo /lib/firmware
```
__Amixer output__
```
Simple mixer control ‘Master’,0
Capabilities: pvolume pswitch
Playback channels: Front Left - Front Right
Limits: Playback 0 - 207
Mono:
Front Left: Playback 170 [82%] [-18.50dB] [off]
Front Right: Playback 170 [82%] [-18.50dB] [off]
Simple mixer control ‘DSP Program’,0
Capabilities: enum
Items: ‘FIR interpolation with de-emphasis’ ‘Low latency IIR with de-emphasis’ ‘High attenuation with de-emphasis’ ‘Fixed process flow’ ‘Ringing-less low latency FIR’
Item0: ‘FIR interpolation with de-emphasis’
Simple mixer control ‘Analogue’,0
Capabilities: pvolume
Playback channels: Front Left - Front Right
Limits: Playback 0 - 1
Mono:
Front Left: Playback 1 [100%] [0.00dB]
Front Right: Playback 1 [100%] [0.00dB]
Simple mixer control ‘Analogue Playback Boost’,0
Capabilities: volume
Playback channels: Front Left - Front Right
Capture channels: Front Left - Front Right
Limits: 0 - 1
Front Left: 0 [0%] [0.00dB]
Front Right: 0 [0%] [0.00dB]
Simple mixer control ‘Auto Mute’,0
Capabilities: pswitch
Playback channels: Front Left - Front Right
Mono:
Front Left: Playback [on]
Front Right: Playback [on]
Simple mixer control ‘Auto Mute Mono’,0
Capabilities: pswitch pswitch-joined
Playback channels: Mono
Mono: Playback [on]
Simple mixer control ‘Auto Mute Time Left’,0
Capabilities: enum
Items: ‘21ms’ ‘106ms’ ‘213ms’ ‘533ms’ ‘1.07s’ ‘2.13s’ ‘5.33s’ ‘10.66s’
Item0: ‘21ms’
Simple mixer control ‘Auto Mute Time Right’,0
Capabilities: enum
Items: ‘21ms’ ‘106ms’ ‘213ms’ ‘533ms’ ‘1.07s’ ‘2.13s’ ‘5.33s’ ‘10.66s’
Item0: ‘21ms’
Simple mixer control ‘Clock Missing Period’,0
Capabilities: enum
Items: ‘1s’ ‘2s’ ‘3s’ ‘4s’ ‘5s’ ‘6s’ ‘7s’ ‘8s’
Item0: ‘1s’
Simple mixer control ‘Deemphasis’,0
Capabilities: pswitch pswitch-joined
Playback channels: Mono
Mono: Playback [on]
Simple mixer control ‘Digital’,0
Capabilities: pvolume pswitch
Playback channels: Front Left - Front Right
Limits: Playback 0 - 207
Mono:
Front Left: Playback 170 [82%] [-18.50dB] [on]
Front Right: Playback 170 [82%] [-18.50dB] [on]

# Piano 2.1 stuff
Simple mixer control ‘Dual Mode’,0
Capabilities: enum
Items: ‘None’ ‘Dual-Mono’ ‘Dual-Stereo’
Item0: ‘None’
Simple mixer control ‘Lowpass’,0
Capabilities: enum
Items: ‘60’ ‘70’ ‘80’ ‘90’ ‘100’ ‘110’ ‘120’ ‘130’ ‘140’ ‘150’ ‘160’ ‘170’ ‘180’ ‘190’ ‘200’
Item0: ‘90’
#

Simple mixer control ‘Max Overclock DAC’,0
Capabilities: volume volume-joined
Playback channels: Mono
Capture channels: Mono
Limits: 0 - 40
Mono: 0 [0%]
Simple mixer control ‘Max Overclock DSP’,0
Capabilities: volume volume-joined
Playback channels: Mono
Capture channels: Mono
Limits: 0 - 40
Mono: 0 [0%]
Simple mixer control ‘Max Overclock PLL’,0
Capabilities: volume volume-joined
Playback channels: Mono
Capture channels: Mono
Limits: 0 - 20
Mono: 0 [0%]

# Piano 2.1 stuff
Simple mixer control ‘Subwoofer’,0
Capabilities: pvolume pswitch
Playback channels: Front Left - Front Right
Limits: Playback 0 - 207
Mono:
Front Left: Playback 170 [82%] [-18.50dB] [off]
Front Right: Playback 170 [82%] [-18.50dB] [off]
Simple mixer control ‘Subwoofer mode’,0
Capabilities: enum
Items: ‘None’ ‘2.0’ ‘2.1’ ‘2.2’
Item0: ‘2.2’
#
Simple mixer control ‘Volume Ramp Down Emergency Rate’,0
Capabilities: enum
Items: ‘1 sample/update’ ‘2 samples/update’ ‘4 samples/update’ ‘Immediate’
Item0: ‘1 sample/update’
Simple mixer control ‘Volume Ramp Down Emergency Step’,0
Capabilities: enum
Items: ‘4dB/step’ ‘2dB/step’ ‘1dB/step’ ‘0.5dB/step’
Item0: ‘4dB/step’
Simple mixer control ‘Volume Ramp Down Rate’,0
Capabilities: enum
Items: ‘1 sample/update’ ‘2 samples/update’ ‘4 samples/update’ ‘Immediate’
Item0: ‘1 sample/update’
Simple mixer control ‘Volume Ramp Down Step’,0
Capabilities: enum
Items: ‘4dB/step’ ‘2dB/step’ ‘1dB/step’ ‘0.5dB/step’
Item0: ‘1dB/step’
Simple mixer control ‘Volume Ramp Up Rate’,0
Capabilities: enum
Items: ‘1 sample/update’ ‘2 samples/update’ ‘4 samples/update’ ‘Immediate’
Item0: ‘1 sample/update’
Simple mixer control ‘Volume Ramp Up Step’,0
Capabilities: enum
Items: ‘4dB/step’ ‘2dB/step’ ‘1dB/step’ ‘0.5dB/step’
Item0: ‘1dB/step’
```

foonerd, 2024-12-19:
> It would be helpful if you could resurrect an old image and make a list of the controls which used to be available (`amixer scontrols`?).
> 
> @foonerd Does this chime with your experience?
Unfortunately I'm away till weekend. If you can outline what details are needed, or tests - happy to oblige.
The kernel path I was on was 5.5.y -> 6.1.y -> 6.6.y and can get the details from these if it helps.

Sunfish-2, 2024-12-19:
FYI
I have an older version running.
RPiOS: 11.9 Bullseye 64-bit | Linux: 6.1.21 64-bit

Output of amixer scontrols
Simple mixer control 'Master',0
Simple mixer control 'Dual Mode',0
Simple mixer control 'Lowpass',0
Simple mixer control 'Main Analogue',0
Simple mixer control 'Main Analogue Playback Boost',0
Simple mixer control 'Main DSP Program',0
Simple mixer control 'Main Deemphasis',0
Simple mixer control 'Sub Analogue',0
Simple mixer control 'Sub Analogue Playback Boost',0
Simple mixer control 'Sub DSP Program',0
Simple mixer control 'Sub Deemphasis',0
Simple mixer control 'Subwoofer',0
Simple mixer control 'Subwoofer mode',0

Note that the control 'Master' in the early days of the Piano 2.1 driver 
was called 'Digital'.

--------------SN3PcCOOlDFO8fsRWNm0vbVp
Content-Type: text/html; charset=UTF-8
Content-Transfer-Encoding: 8bit

<!DOCTYPE html><html><head>
<meta http-equiv="Content-Type" content="text/html; charset=utf-8">
  </head>
  <body>
    <font face="Tahoma">FYI<br>
      I have an older version running.<br>
      RPiOS: 11.9 Bullseye 64-bit | Linux: 6.1.21 64-bit<br>
      <br>
      <br>
      Output of amixer scontrols<br>
      Simple mixer control 'Master',0<br>
      Simple mixer control 'Dual Mode',0<br>
      Simple mixer control 'Lowpass',0<br>
      Simple mixer control 'Main Analogue',0<br>
      Simple mixer control 'Main Analogue Playback Boost',0<br>
      Simple mixer control 'Main DSP Program',0<br>
      Simple mixer control 'Main Deemphasis',0<br>
      Simple mixer control 'Sub Analogue',0<br>
      Simple mixer control 'Sub Analogue Playback Boost',0<br>
      Simple mixer control 'Sub DSP Program',0<br>
      Simple mixer control 'Sub Deemphasis',0<br>
      Simple mixer control 'Subwoofer',0<br>
      Simple mixer control 'Subwoofer mode',0<br>
      <br>
      Note that the control 'Master' in the early days of the Piano 2.1
      driver was called 'Digital'.<br>
    </font><br>
    <br>
    <br>
    <div class="moz-cite-prefix">Op 19-12-2024 om 05:58 schreef Just a
      nerd:<br>
    </div>
    <blockquote type="cite" ***@***.***">
      
      <blockquote>
        <p dir="auto">It would be helpful if you could resurrect an old
          image and make a list of the controls which used to be
          available (<code class="notranslate">amixer scontrols</code>?).</p>
        <p dir="auto"><a class="user-mention notranslate" data-hovercard-type="user" data-hovercard-url="/users/foonerd/hovercard" data-octo-click="hovercard-link-click" data-octo-dimensions="link_type:self" href="https://github.com/foonerd" ***@***.***</a>
          Does this chime with your experience?</p>
      </blockquote>
      <p dir="auto">Unfortunately I'm away till weekend. If you can
        outline what details are needed, or tests - happy to oblige.<br>
        The kernel path I was on was 5.5.y -&gt; 6.1.y -&gt; 6.6.y and
        can get the details from these if it helps.</p>
      <p style="font-size:small;-webkit-text-size-adjust:none;color:#666;">—<br>
          ID: <span>&lt;raspberrypi/linux/issues/6548/2552787877</span><span>@</span><span>github</span><span>.</span><span>com&gt;</span></span></p>
      <script type="application/ld+json">[
{
"target": "https://github.com/raspberrypi/linux/issues/6548#issuecomment-2552787877",
"url": "https://github.com/raspberrypi/linux/issues/6548#issuecomment-2552787877",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]</script>
    </blockquote>
    <br>
  </body>
</html>

--------------SN3PcCOOlDFO8fsRWNm0vbVp--

pelwell, 2024-12-19:
I hope to get a chance to look at this tomorrow.

pelwell, 2024-12-20:
With a one-line change to allo-piano-dac-plus.c:
```diff
diff --git a/sound/soc/bcm/allo-piano-dac-plus.c b/sound/soc/bcm/allo-piano-dac-plus.c
index f999133dddb20..aa6f01069cdd2 100644
--- a/sound/soc/bcm/allo-piano-dac-plus.c
+++ b/sound/soc/bcm/allo-piano-dac-plus.c
@@ -916,7 +916,7 @@ static struct snd_soc_dai_link_component allo_piano_2_1_codecs[] = {
 SND_SOC_DAILINK_DEFS(allo_piano_dai_plus,
        DAILINK_COMP_ARRAY(COMP_EMPTY()),
        DAILINK_COMP_ARRAY(COMP_CODEC("pcm512x.1-004c", "pcm512x-hifi"),
-                          COMP_CODEC("pcm512x.1-004d", "pcm512x-hifi")),
+                          COMP_CODEC("pcm512x.1-004c", "pcm512x-hifi")),
        DAILINK_COMP_ARRAY(COMP_EMPTY()));
 
 static struct snd_soc_dai_link snd_allo_piano_dac_dai[] = {
```
I am able to use the `allo-piano-dac-plus-pcm512x-audio` with an IQaudIO Pi-DAC+, with a reasonable looking list of controls:
```
pi@raspberrypi:~$ amixer -c 2 scontrols
Simple mixer control 'Master',0
Simple mixer control 'Dual Mode',0
Simple mixer control 'Lowpass',0
Simple mixer control 'Main Analogue',0
Simple mixer control 'Main Analogue Playback Boost',0
Simple mixer control 'Main DSP Program',0
Simple mixer control 'Main Deemphasis',0
Simple mixer control 'Subwoofer',0
Simple mixer control 'Subwoofer mode',0
```
which is the same as your list on 6.1.

In the kernel (`dmesg -l err`) it looks as though all is not well:
```
[    5.781690] snd-allo-piano-dac-plus soc:sound: Failed to set master volume limit: -22
[    5.789594] snd-allo-piano-dac-plus soc:sound: Failed to set subwoofer volume limit: -22
[    5.789598] snd-allo-piano-dac-plus soc:sound: Failed to set Sub Digital Playback Volume volume limit: -22
[    5.789638] snd-allo-piano-dac-plus soc:sound: Control Sub Digital Playback Volume not found
[    5.789639] snd-allo-piano-dac-plus soc:sound: Control Sub Digital Playback Switch not found
[    5.789640] snd-allo-piano-dac-plus soc:sound: Control Sub Auto Mute Mono Switch not found
[    5.789642] snd-allo-piano-dac-plus soc:sound: Control Sub Auto Mute Switch not found
[    5.789643] snd-allo-piano-dac-plus soc:sound: Control Sub Auto Mute Time Left not found
[    5.789644] snd-allo-piano-dac-plus soc:sound: Control Sub Auto Mute Time Right not found
[    5.789645] snd-allo-piano-dac-plus soc:sound: Control Sub Clock Missing Period not found
[    5.789647] snd-allo-piano-dac-plus soc:sound: Control Sub Max Overclock DAC not found
[    5.789648] snd-allo-piano-dac-plus soc:sound: Control Sub Max Overclock DSP not found
[    5.789649] snd-allo-piano-dac-plus soc:sound: Control Sub Max Overclock PLL not found
[    5.789650] snd-allo-piano-dac-plus soc:sound: Control Sub Volume Ramp Down Emergency Rate not found
[    5.789652] snd-allo-piano-dac-plus soc:sound: Control Sub Volume Ramp Down Emergency Step not found
[    5.789653] snd-allo-piano-dac-plus soc:sound: Control Sub Volume Ramp Up Rate not found
[    5.789654] snd-allo-piano-dac-plus soc:sound: Control Sub Volume Ramp Down Rate not found
[    5.789656] snd-allo-piano-dac-plus soc:sound: Control Sub Volume Ramp Up Step not found
[    5.789657] snd-allo-piano-dac-plus soc:sound: Control Sub Volume Ramp Down Step not found
```
However, these messages come from the code that removes the codec's controls, and only for the Sub codec (which doesn't exist on an IQaudIO Pi-DAC+).

I also get a working volume control etc. using alsamixer.

Running the current 6.6 kernel (`sudo rpi-update`), what do `dmesg -l err` and `amixer scontrols` report? And what appears in alsamixer?

Sunfish-2, 2024-12-25:
Kernel (uname): Linux raspberrypi 6.6.67-v8+ #1833 SMP PREEMPT Fri Dec 20 14:14:15 GMT 2024 aarch64 GNU/Linux
dmesg -l err: gives no output.
amixer scontrols: gives no output.
alsamixer:  "This sound device does not have any controls."

moodeaudio, 2024-12-27:
amixer and alsamixer default to card 0 unless -c N is specified

Might be worth checking `aplay -l` to confirm the piano card assignment.

Sunfish-2, 2024-12-27:
Output of aplay -l:
**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi0 [vc4-hdmi-0], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: PianoDACPlus [PianoDACPlus], device 0: PianoDACPlus multicodec-0 [PianoDACPlus multicodec-0]
  Subdevices: 0/1
  Subdevice #0: subdevice #0

moodeaudio, 2024-12-27:
Then add -c2
```
amixer -c2 scontrols
alsamixer -c2
```

pelwell, 2024-12-27:
Alternatively, if you don't want to use the HDMI audio, add the `noaudio` parameter to the `vc4-kms-v3d` overlay usage in your config.txt:
```
dtoverlay=vc4-kms-v3d,noaudio
```

Sunfish-2, 2024-12-27:
output of amixer -c2 scontrols:
Simple mixer control 'Master',0
Simple mixer control 'Dual Mode',0
Simple mixer control 'Lowpass',0
Simple mixer control 'Main Analogue',0
Simple mixer control 'Main Analogue Playback Boost',0
Simple mixer control 'Main DSP Program',0
Simple mixer control 'Main Deemphasis',0
Simple mixer control 'Sub Analogue',0
Simple mixer control 'Sub Analogue Playback Boost',0
Simple mixer control 'Sub DSP Program',0
Simple mixer control 'Sub Deemphasis',0
Simple mixer control 'Subwoofer',0
Simple mixer control 'Subwoofer mode',0

alsamixer -c2 command gives the graphical interface of the controls

Nice.

Sunfish-2, 2024-12-27:
Tried to add the noaudio option and rebooted the system now the result of aplay -l is:
aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi0 [vc4-hdmi-0], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: PianoDACPlus [PianoDACPlus], device 0: PianoDACPlus multicodec-0 [PianoDACPlus multicodec-0]
  Subdevices: 0/1
  Subdevice #0: subdevice #0

The Piana 2.1 dac card number now is 1.

pelwell, 2024-12-27:
> alsamixer -c2 command gives the graphical interface of the controls

Good catch, @moodeaudio.