raspberrypi/linux #6554: h264_v4l2m2m / bcm2835_v4l2 hardware accelerated decoding of h264 on kernel 6.6.63

t3therdev, 2024-12-20:
### Describe the bug

We are having an issue with h.264 input decoding/transcoding when migrating a working pi4/cm4 system from kernel 5.15 / ffmpeg 5.1.x where hardware decoding worked perfectly, to kernel 6.6.63 / (Tested ffmpeg 5.1.x/6/7). Tested on mainline and rpi-ffmpeg branches.  On the newer 6.x kernel, hardware decoding in ffmpeg using h264_v4l2m2m, specifically does not work and just hangs. If we remove the hardware decoding in the command, everything works perfectly. Taking the same hardware back to kernel 5.15 also works. 

v4l2-ctl returns the bcm encoder and decoder properly on /dev/video10 and /dev/video11

boot/config/.txt we have gpu_mem=128 and have tried other settings as well with no success. falling back to the same exact configuration on the older kernel works perfectly. 

### Steps to reproduce the behaviour

Using h264 input source (for example a usb camera with h264 video, h264 file , or any rtsp stream with h264 video input and trying to decode/encode at the same time.  Lowering resolutions, fps doesn't help.

Example commands - NOT working
> ffmpeg -f video4linux2 **-c:v h264_v4l2m2m** -input_format h264 -video_size 1280x720 -framerate 30 -i /dev/video0 -c:v h264_v4l2m2m -an -pix_fmt yuv420p -b:v 2048k -r 30 test.mp4

With -c:v h264_v4l2m2m removed) its working. 

> ffmpeg -f video4linux2 -input_format h264 -video_size 1280x720 -framerate 30 -i /dev/video0 -c:v h264_v4l2m2m -an -pix_fmt yuv420p -b:v 2048k -r 30 test.mp4

### Device (s)

Raspberry Pi Zero 2 W, Raspberry Pi 4 Mod. B, Raspberry Pi CM4

### System

Openwrt 24.10 , Kernel 6.6
Linux 6.6.63 #0 SMP Tue Dec 10 00:22:05 2024 aarch64 GNU/Linux
version ddfba3e3c234500025b545512b4b214f28e453e9 (clean) (release) (start)

### Logs

Running ffmpeg in debug mode:
`Splitting the commandline.
Reading option '-loglevel' ... matched as option 'loglevel' (set logging level) with argument 'debug'.
Reading option '-f' ... matched as option 'f' (force container format (auto-detected otherwise)) with argument 'video4linux2'.
Reading option '-c:v' ... matched as option 'c' (select encoder/decoder ('copy' to copy stream without reencoding)) with argument 'h264_v4l2m2m'.
Reading option '-input_format' ... matched as AVOption 'input_format' with argument 'h264'.
Reading option '-video_size' ... matched as AVOption 'video_size' with argument '1280x720'.
Reading option '-framerate' ... matched as AVOption 'framerate' with argument '30'.
Reading option '-i' ... matched as input url with argument '/dev/video0'.
Reading option '-fflags' ... matched as AVOption 'fflags' with argument 'nobuffer'.
Reading option '-c:v' ... matched as option 'c' (select encoder/decoder ('copy' to copy stream without reencoding)) with argument 'h264_v4l2m2m'.
Reading option '-an' ... matched as option 'an' (disable audio) with argument '1'.
Reading option '-pix_fmt' ... matched as option 'pix_fmt' (set pixel format) with argument 'yuv420p'.
Reading option '-b:v' ... matched as option 'b' (video bitrate (please use -b:v)) with argument '2048k'.
Reading option '-r' ... matched as option 'r' (override input framerate/convert to given output framerate (Hz value, fraction or abbreviation)) with argument '30'.
Reading option '-f' ... matched as option 'f' (force container format (auto-detected otherwise)) with argument 'rtsp'.
Reading option 'rtsp://stream:stream@localhost:554/device1' ... matched as output url.
Finished splitting the commandline.
Parsing a group of options: global .
Applying option loglevel (set logging level) with argument debug.
Successfully parsed a group of options.
Parsing a group of options: input url /dev/video0.
Applying option f (force container format (auto-detected otherwise)) with argument video4linux2.
Applying option c:v (select encoder/decoder ('copy' to copy stream without reencoding)) with argument h264_v4l2m2m.
Successfully parsed a group of options.
Opening an input file: /dev/video0.
[video4linux2,v4l2 @ 0x7fb5c3c0d0] fd:3 capabilities:84a00001
[video4linux2,v4l2 @ 0x7fb5c3c0d0] Current input_channel: 0, input_name: Camera 1, input_std: 0
[video4linux2,v4l2 @ 0x7fb5c3c0d0] Setting time per frame to 1/30
[h264 @ 0x450320] non-existing PPS 0 referenced
[extract_extradata @ 0x7fb3dd9db0] nal_unit_type: 1(Coded slice of a non-IDR picture), nal_ref_idc: 2
[h264 @ 0x450320] nal_unit_type: 1(Coded slice of a non-IDR picture), nal_ref_idc: 2
[h264 @ 0x450320] non-existing PPS 0 referenced
[h264 @ 0x450320] decode_slice_header error
[h264 @ 0x450320] no frame!
[h264 @ 0x450320] non-existing PPS 0 referenced
[extract_extradata @ 0x7fb3dd9db0] nal_unit_type: 1(Coded slice of a non-IDR picture), nal_ref_idc: 2
[h264 @ 0x450320] nal_unit_type: 1(Coded slice of a non-IDR picture), nal_ref_idc: 2
[h264 @ 0x450320] non-existing PPS 0 referenced
[... 61 lines cut ...]
[h264_v4l2m2m @ 0x7fb3eb4c00] v4l2 output format not supported
[h264_v4l2m2m @ 0x7fb3eb4c00] probing device /dev/video20
[h264_v4l2m2m @ 0x7fb3eb4c00] driver 'bcm2835-isp' on card 'bcm2835-isp' in splane mode
[h264_v4l2m2m @ 0x7fb3eb4c00] v4l2 output format not supported
[h264_v4l2m2m @ 0x7fb3eb4c00] probing device /dev/video10
[h264_v4l2m2m @ 0x7fb3eb4c00] driver 'bcm2835-codec' on card 'bcm2835-codec-decode' in mplane mode
[h264_v4l2m2m @ 0x7fb3eb4c00] Using device /dev/video10
[h264_v4l2m2m @ 0x7fb3eb4c00] driver 'bcm2835-codec' on card 'bcm2835-codec-decode' in mplane mode
[h264_v4l2m2m @ 0x7fb3eb4c00] requesting formats: output=H264/none capture=YU12/yuvj420p
[h264_v4l2m2m @ 0x7fb3eb4c00] output: H264 16 buffers initialized: 1280x0720, sizeimage 00691328, bytesperline 00000000
Successfully opened the file.
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (h264_v4l2m2m) -> h264 (h264_v4l2m2m))
[vost#0:0/h264_v4l2m2m @ 0x7fb38c6820] Starting thread...
[vf#0:0 @ 0x7fb5af5650] Starting thread...
[vist#0:0/h264 @ 0x7fb8991340] [dec:h264_v4l2m2m @ 0x7fb3eb42a0] Starting thread...
[in#0/video4linux2,v4l2 @ 0x7fb5cbf420] Starting thread...
Press [q] to stop, [?] for help
[h264_v4l2m2m @ 0x7fb3eb4c00] capture: YU12 20 buffers initialized: 1280x0720, sizeimage 01382400, bytesperline 00001280
[h264 @ 0x7fb38c6080] Decoding VUI
`

### Additional context

Testing on multiple Pi devices, this seems to be specific to the newer kernel and h264_v4l2m2m decoding of h264.

popcornmix, 2024-12-20:
Duplicate of https://github.com/jc-kynesim/rpi-ffmpeg/issues/97

6by9, 2024-12-20:
Closing here. Please use https://github.com/jc-kynesim/rpi-ffmpeg/issues/97