raspberrypi/linux #6557: PCI: Warn if no host bridge NUMA node info still in kernel 6.13

by, 2024-12-21:
### Describe the bug

Kernel 6.13 still reports `pci_bus 0000:0n: Unknown NUMA node; performance will be reduced`, only 6.6 seems to be patched.

### Steps to reproduce the behaviour

Could you please port https://github.com/raspberrypi/linux/pull/6531#issue-2730631601 to kernel 6.13, too, thank you!

### Device (s)

Raspberry Pi 5

### System

Linux 6.13.0-rc3-v8-16k-NTP+ #5 SMP Mon Dec 16 18:27:29 CET 2024 aarch64 GNU/Linux

pelwell, 2025-01-01:
I initially assumed this missing revert was due to an error in the forward-porting script, but I now think it just got dropped because the revert is not a clean one. It should be fixed now.

by, 2025-01-01:
Thanks and yes, it's gone.

closed 2025-01-01 by by